avr-rust/rust-legacy-fork #8: Set `avr-none` as the default target

dylanmckay, 2015-04-01:
Set the default target to `avr-none` so that by default we attempt to build for AVR.

This should also fix #4.

alexking, 2015-05-13:
So I've been trying to look into how the default target is set (I'm trying to get the core library building – #4). It seems like the default `--target` comes from [`host_triple()`](https://github.com/dylanmckay/avr-rust/blob/avr-support/src/librustc/session/config.rs#L199), which gets it from `CFG_COMPILER_HOST_TRIPLE`. Any idea if this is what we're looking to change?

dylanmckay, 2015-05-13:
If I recall correctly, if we pass `--target=avr-none`, LLVM also gets configured so that `avr-none` is the default target.

The problem with this is that `compiler-rt` does not support AVR, so Makefile generation fails.

dylanmckay, 2015-08-09:
We don't want to modify the Makefiles to ignore the host and generate AVR bindings - we want to add support to Rust so that we can call `./configure --target=avr-atmel-none` and we get a compiler which targets AVR by default (not passing the target triple to `rustc`) which has a `libcore` compiled for AVR as well.

Currently I have a patch which fixes AVR support for compiler-rt, and adds support for 16-bit pointers to `libcore`. Will push when ready.

dylanmckay, 2015-08-17:
This is currently blocked by AVR-LLVM issue [#149](https://github.com/avr-llvm/llvm/issues/149)

briansmith, 2015-10-23:
> we want to add support to Rust so that we can call ./configure --target=avr-atmel-none and we get a compiler which targets AVR by default (not passing the target triple to rustc)

I would prefer that, instead, the default target is the host, and that --target=avr-atmel-none is passed to cargo and rustc to build for AVR. Many crates need the host target in order to run their `build.rs` build script. Also, I would like to see AVR target support included in all releases of rustc for all platforms, and that can only happen if it is supported as a cross-compiler.

dylanmckay, 2015-10-23:
I have good news :)

I am currently in the (long,slow) process of merging AVR-LLVM directly into the official LLVM repository. Once that is completed, Rust will simply have to update LLVM and it will be able to be used.

With regards to this issue, the only reason we can't use `avr` as the default target is because AVR-LLVM crashes while compiling `libcore`. This means that there's nothing physically stopping us from using AVR as the default target, except a bug that would've stopped us from compile `libcore` anyway.

The AVR-LLVM migration process will take a while, and so I plan on working on fixing #149 once that is completed.

mrdeveloperdude, 2018-08-23:
@dylanmckay Hi! I was randomly looking for llvm avr/arduino support when I came here. Taht is awesome! Any status update as of today (2018-08-23)?

shepmaster, 2019-01-20:
> AVR can be made the default target.

@dylanmckay I'm not sure that there's much of a point to this. Rust defaults to the host system, and I'm 99.9999% sure that rustc will never run on AVR. Since our goal is to merge into upstream Rust, we'd never want to have such a behavior.

dylanmckay, 2019-01-20:
Good point, I hadn't though of that - changing the default target to AVR could only ever be relevant in a rustc fork.