avr-rust/rust-legacy-fork #13: Unable to compile dependency-free bare-bones example - unsupported instruction MCInst 157

shepmaster, 2016-01-02:
Forgive me if the code below makes no sense; I'm kind of wildly experimenting to get something that doesn't work. Since libcore doesn't compile, I'm trying this example file:

```rust
#![feature(no_std)]
#![feature(no_core)]
#![feature(lang_items)]
#![feature(fundamental)]

#![no_std]
#![no_core]
#![no_main]

const PORTB: *mut u8 = 0x25 as *mut u8;

#[no_mangle]
pub extern fn main(_argc: i32, _argv: *const *const u8) -> i32 {
    loop {
        unsafe {
            *PORTB |= 0x1;
        }
    }
}

#[lang = "sized"]
#[fundamental]
pub trait Sized {}

#[lang = "copy"]
pub trait Copy : Clone {}

pub trait Clone : Sized {
    fn clone(&self) -> Self;

    #[inline(always)]
    fn clone_from(&mut self, source: &Self) {
        *self = source.clone()
    }
}
```

Compiled with `rustc --target avr-atmel-none hello.rs` yields:

```
'generic' is not a recognized processor for this target (ignoring processor)
LLVM ERROR: Not supported instr: <MCInst 157 <MCOperand Reg:51> <MCOperand Imm:11> <MCOperand Reg:49>>
```

The intermediate LLVM IR is

```llvm
; ModuleID = 'hello.0.rs'
target datalayout = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8"
target triple = "avr-atmel-none"

@const141 = internal unnamed_addr constant i8* inttoptr (i16 37 to i8*)
@const142 = internal unnamed_addr constant i8 1

define i32 @main(i32, i8**) unnamed_addr {
"the block":
  %2 = alloca i32
  store i32 %0, i32* %2
  %3 = load i32, i32* %2
  %4 = alloca i8**
  store i8** %1, i8*** %4
  %5 = load i8**, i8*** %4
  %6 = call i32 @_ZN4main10__rust_abiE(i32 %3, i8** %5)
  ret i32 %6
}

; Function Attrs: uwtable
define internal i32 @_ZN4main10__rust_abiE(i32, i8**) unnamed_addr #0 {
entry-block:
  %sret_slot = alloca i32
  %dropflag_hint_11 = alloca i8
  %dropflag_hint_14 = alloca i8
  %_argc = alloca i32
  %_argv = alloca i8**
  store i8 61, i8* %dropflag_hint_11
  store i8 61, i8* %dropflag_hint_14
  store i32 %0, i32* %_argc, align 1
  store i8** %1, i8*** %_argv, align 1
  br label %loop_body

loop_exit:                                        ; No predecessors!
  unreachable

loop_body:                                        ; preds = %loop_body, %entry-block
  %2 = load i8, i8* inttoptr (i16 37 to i8*), align 1
  %3 = or i8 %2, 1
  store i8 %3, i8* inttoptr (i16 37 to i8*), align 1
  br label %loop_body
}

attributes #0 = { uwtable }
```

Version information

```
rustc 1.4.0-dev (f841b21fa 2015-09-06)
binary: rustc
commit-hash: f841b21fa4578dc379adad79f1af66b201154b5e
commit-date: 2015-09-06
host: x86_64-apple-darwin
release: 1.4.0-dev
```

dylanmckay, 2016-01-07:
For the "generic CPU not found" error, try passing `-C target-cpu=atmega328p`. This should fix the second error as well.

shepmaster, 2016-01-14:
I actually took a different tack, and updated to the newest versions of avr-llvm and Rust proper. Now I've got two interesting errors. The first is that `compiler-rt` isn't being built:

```
error: linking with `cc` failed: exit code: 1
note: "cc" "-L" "/Users/shep/Projects/avr-rust/build-no-core/x86_64-apple-darwin/stage2/lib/rustlib/avr-atmel-none/lib" "hello.0.o" "-o" "hello" "-Wl,-dead_strip" "-pie" "-nodefaultlibs" "-L" "/Users/shep/Projects/avr-rust/build-no-core/x86_64-apple-darwin/stage2/lib/rustlib/avr-atmel-none/lib" "-l" "compiler-rt"
note: clang: warning: argument unused during compilation: '-pie'
ld: warning: directory not found for option '-L/Users/shep/Projects/avr-rust/build-no-core/x86_64-apple-darwin/stage2/lib/rustlib/avr-atmel-none/lib'
ld: warning: directory not found for option '-L/Users/shep/Projects/avr-rust/build-no-core/x86_64-apple-darwin/stage2/lib/rustlib/avr-atmel-none/lib'
ld: library not found for -lcompiler-rt
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

So now I'm digging into why that wouldn't be built.

The other error is an LLVM assertion when building the compiler, but only in non-debug mode. That's always exciting.

dylanmckay, 2016-09-25:
This has been fixed in AVR-LLVM.

I have also double checked by compiling the IR of the program @shepmaster linked above. It worked fine.

closed 2016-09-25 by dylanmckay