avr-rust/rust-legacy-fork #17: Missing compiler-rt commit

tbelaire, 2016-08-01:
It looks like there's a commit in compiler-rt which isn't published?  I can't fetch the submodule.
```
error: no such remote ref 91f0fe0e367edd404ec5576bab070abb92333db9
Fetched in submodule path 'src/compiler-rt', but it did not contain 91f0fe0e367edd404ec5576bab070abb92333db9. Direct fetching of that commit failed.
```

dylanmckay, 2016-08-02:
@shepmaster do you know about this?

@tbelaire could you do a `git remote -v` inside the `src/compiler-rt` directory?

shepmaster, 2016-08-02:
Ugh, yeah, that's my fault. And unfortunately, it looks like I pushed it to `avr-llvm/compiler-rt` instead of `avr-rust/compiler-rt`... too many forks. Is that going to screw up the normal AVR work, @dylanmckay ?

shepmaster, 2016-08-02:
Overall, the whole story around compiler-rt is less-than-ideal. Rust upstream is going to try to avoid the whole mess around building it, for example - rust-lang/rust#34400.

dylanmckay, 2016-08-02:
It won't mess anything up, I haven't even been checking it out recently lol

dylanmckay, 2016-09-26:
No longer an issue as far as I can tell.

closed 2016-09-26 by dylanmckay