avr-rust/rust-legacy-fork #18: ResumeInst needs to be in a function with a personality

shepmaster, 2016-09-26:
Trying to update to avr-llvm/llvm@3c0d2adc4d75f34bbeb3639172197b837409bedf, I've made some [good progress and some nasty hacks](https://github.com/shepmaster/rust/tree/avr-support-4.0), but now I'm stuck at this intriguing LLVM error

```
ResumeInst needs to be in a function with a personality.
  resume { i8*, i32 } %3
LLVM ERROR: Broken function found, compilation aborted!
```

Unfortunately, I have not been able to generate an LLVM IR dump of the invalid input before the failure occurs, and I can't get useful debugging symbols in LLVM **or** add debugging prints to the LLVM codebase. Frustrating!

dylanmckay, 2016-09-26:
Interesting, I'm not getting that.

I've been working off the branch you pushed before

https://github.com/avr-rust/rust/commits/avr-support-4.0

I am getting this error though:

```
Assertion failed: (isa<X>(Val) && "cast<Ty>() argument of incompatible type!"), function cast, file //Users//dylanmckay//projects//avr-rust//rust//src//llvm//include/llvm/Support/Casting.h, line 237.
```

dylanmckay, 2016-09-26:
Pretty sure we duplicated a bunch of stuff

shepmaster, 2016-09-26:
> Pretty sure we duplicated a bunch of stuff

That wouldn't be too surprising, but I definitely hacked-and-slashed some stuff to get it working. There's a few places where I lost functionality. Do you have a branch of your in-progress changes? Maybe something will jump out at us if we compare the two.

dylanmckay, 2016-09-26:
The branch I linked before has my WIP stuff

shepmaster, 2016-09-26:
Oh, I didn't even notice it was in this repo ;-)

I do notice that you are building LLVM inside of the Rust directory; are you letting the Rust build process drive that? I've got LLVM built in a separate directory and am linking to it.

dylanmckay, 2016-09-26:
> are you letting the Rust build process drive that?

Yeah I am

Other differences I can see:

* You're using the beta bootstrap stuff instead of the `configure` script like myself
* You've got a bunch more inlining-related stuff on your branch, whereas I've got your initial inlining stuff in my branch

shepmaster, 2016-09-26:
> You're using the beta bootstrap stuff instead of the configure script like myself

Yeah, for a long time the `configure` script couldn't handle the CMake-based LLVM. That was addressed when they wanted to upgrade to 3.9 for Emscripten, but I'm still used to the way that I got working ;-)

> a bunch more inlining-related stuff

I don't know that there's a bunch more, but I think I changed it from the initial commit because I got missing symbol errors, which might be fixed from your inclusion of the Objective-C library...

dylanmckay, 2016-09-26:
I'm merging our two branches, and will prune stuff and see if I can get a build working

shepmaster, 2016-09-26:
> I'm merging our two branches

👍 I'll be off to bed shortly. I'll expect a magically working compiler in the morning 😈

shepmaster, 2016-09-26:
Woo, my last hail-Mary effort worked:

```llvm
; Function Attrs: inlinehint uwtable
define internal void @_ZN4core3mem6forget17h6a12bc5fd51e1250E(i64) unnamed_addr #2 !dbg !633 {
entry-block:
  %arg0 = alloca i64
  %t = alloca i64
  %personalityslot = alloca { i8*, i32 }
  store i64 %0, i64* %arg0
  call void @llvm.dbg.declare(metadata i64* %arg0, metadata !641, metadata !642), !dbg !643
  call void @llvm.dbg.declare(metadata i64* %t, metadata !644, metadata !642), !dbg !646
  br label %start

start:                                            ; preds = %entry-block
  %1 = load i64, i64* %arg0, !dbg !647
  store i64 %1, i64* %t, !dbg !647
  %2 = load i64, i64* %t, !dbg !648
  br label %bb2, !dbg !648

bb1:                                              ; No predecessors!
  %3 = load { i8*, i32 }, { i8*, i32 }* %personalityslot, !dbg !647
  resume { i8*, i32 } %3

bb2:                                              ; preds = %start
  ret void, !dbg !646
}

; Function Attrs: inlinehint uwtable
define internal void @_ZN4core3mem6forget17h6a12bc5fd51e1250E(i64) unnamed_addr #2 !dbg !5106 {
entry-block:
  %arg0 = alloca i64
  %t = alloca i64
  %personalityslot = alloca { i8*, i32 }
  store i64 %0, i64* %arg0
  call void @llvm.dbg.declare(metadata i64* %arg0, metadata !5111, metadata !642), !dbg !5112
  call void @llvm.dbg.declare(metadata i64* %t, metadata !5113, metadata !642), !dbg !5115
  br label %start

start:                                            ; preds = %entry-block
  %1 = load i64, i64* %arg0, !dbg !5116
  store i64 %1, i64* %t, !dbg !5116
  %2 = load i64, i64* %t, !dbg !5117
  br label %bb2, !dbg !5117

bb1:                                              ; No predecessors!
  %3 = load { i8*, i32 }, { i8*, i32 }* %personalityslot, !dbg !5116
  resume { i8*, i32 } %3

bb2:                                              ; preds = %start
  ret void, !dbg !5115
}
```

dylanmckay, 2016-09-26:
I've also gotten the `ResumeInst` error after merging the two branches. I'll try cleaning and rebuilding

dylanmckay, 2016-09-26:
I've rebuilt twice and am still getting that error, weird.

dylanmckay, 2016-09-26:
Looks like others have had this bug before too

https://botbot.me/mozilla/rust-internals/2016-08-11/?msg=71118645&page=1

shepmaster, 2016-09-29:
I've got a very hacky reproduction:

```rust
#![feature(lang_items)]
#![feature(start)]
#![feature(intrinsics)]
#![feature(no_core)]
#![feature(on_unimplemented)]
#![feature(fundamental)]

#![no_std]
#![no_core]

#[start]
fn start(_argc: isize, _argv: *const *const u8) -> isize {
    forget(42);
    0
}

#[lang = "eh_personality"]
extern fn eh_personality() {}

#[lang = "panic_fmt"]
extern fn panic_fmt() -> ! { loop {} }

mod intrinsics {
    extern "rust-intrinsic" {
        pub fn forget<T>(_: T) -> ();
    }
}

#[inline]
pub fn forget<T>(t: T) {
    unsafe { intrinsics::forget(t) }
}

#[lang = "sized"]
#[rustc_on_unimplemented = "`{Self}` does not have a constant size known at compile-time"]
#[fundamental]
pub trait Sized {}

#[lang = "copy"]
pub trait Copy {}
```

And this is how I reproduce

```
RUSTC="/Users/shep/Projects/avr-rust/debug/build/bootstrap/debug/rustc" \
RUSTC_BOOTSTRAP_KEY="836a4172" \
RUSTC_LIBDIR="/Users/shep/Projects/avr-rust/debug/build/x86_64-apple-darwin/stage1/lib" \
RUSTC_REAL="/Users/shep/Projects/avr-rust/debug/build/x86_64-apple-darwin/stage1/bin/rustc" \
RUSTC_SNAPSHOT="/Users/shep/Projects/avr-rust/debug/build/x86_64-apple-darwin/stage0/bin/rustc" \
RUSTC_SNAPSHOT_LIBDIR="/Users/shep/Projects/avr-rust/debug/build/x86_64-apple-darwin/stage0/lib" \
RUSTC_STAGE="1" \
RUSTC_SYSROOT="/Users/shep/Projects/avr-rust/debug/build/x86_64-apple-darwin/stage1" \
/Users/shep/Projects/avr-rust/debug/build/bootstrap/debug/rustc \
/tmp/repro.rs \
--crate-type lib \
--emit=link \
--target x86_64-apple-darwin
```

Really, I'm running that process in LLDB, but it forks the real compiler, so I need to do something like:

```
# LLDB #1
(lldb) env ...env...
(lldb) b fork
(lldb) r ...args...
# LLDB #2
(lldb) attach -w -n rustc
# LLDB #1
(lldb) c
# LLDB #2
(lldb) br set -r '.*personality.*'
(lldb) br set -r '.*Personality.*'
(lldb) c
```

Doing this, I break at the following frames:

```
rustc::middle::lang_items::{{impl}}::eh_personality at lang_items.rs:136
rustc_trans::mir::block::{{impl}}::get_personality_slot at block.rs:758
rustc_trans::mir::block::{{impl}}::get_personality_slot::{{closure}} at block.rs:764
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
llvm::Function::hasPersonalityFn() const
```

So it currently looks like we are indeed not setting the personality on the function.

dylanmckay, 2016-09-29:
Sounds like an upstream Rust bug, perhaps we should raise it there?

shepmaster, 2016-09-29:
Perhaps. I'm never quite sure how upstream wants to treat LLVM upgrades that they don't need directly. I suppose I could open an issue there later today with the appropriate information. In the worst case, they can close it.

dylanmckay, 2016-09-29:
@shepmaster: Does this occur when disabling MIR and using the old backend?

shepmaster, 2016-09-29:
I'm... not sure how to disable MIR anymore... I'm looking, but maybe you recall?

shepmaster, 2016-09-29:
Maybe it was permanently enabled? cb9b0ed91b898cbda23caa8ed1a82c37b39845de

dylanmckay, 2016-09-29:
Looks like you're right, that's tricky. I do think it's weird that the only things being done in the IR you posted before:

* Loading
* Storing
* Branching within the function
* Calling `llvm.dbg.declare` to declare debug information

I don't think any of this would trigger exceptions, it does literally nothing (as per the definition of `drop`). The bug could be that we shouldn't even have `resume`

```llvm
  call void @llvm.dbg.declare(metadata i64* %arg0, metadata !641, metadata !642), !dbg !643
  call void @llvm.dbg.declare(metadata i64* %t, metadata !644, metadata !642), !dbg !646
```

dylanmckay, 2016-10-01:
The IR validator for 'ResumeInst must be in a function with a personality' was introduced in LLVM commit [r277360](https://reviews.llvm.org/rL277360).

We could always quick-fix this by reverting the commit on our fork of LLVM.

dylanmckay, 2016-10-01:
I think I may have fixed the bug, will keep this updated.

dylanmckay, 2016-10-01:
I've opened #20, although it currently causes a segfault.

dylanmckay, 2016-10-01:
I'm pretty happy to revert out the verifier addition in LLVM that caused this breakage and not think about it until upstream Rust fixes this when they next update LLVM tbh

dylanmckay, 2016-10-01:
I've fixed the segfault and am upstreaming it in rust-lang/rust@36896

Found another segfault

shepmaster, 2016-10-02:
I think this is the most recent segfault:

```
* thread #2: tid = 0x2c556b, 0x00000001083fca78 librustc_llvm-d85452a8c927d1f5.dylib`llvm::MemoryBufferRef::getBuffer() const + 8, name = 'rustc', stop reason = EXC_BAD_ACCESS (code=1, address=0x10)
  * frame #0: 0x00000001083fca78 librustc_llvm-d85452a8c927d1f5.dylib`llvm::MemoryBufferRef::getBuffer() const + 8
    frame #1: 0x0000000108447285 librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Binary::getData() const + 21
    frame #2: 0x00000001084337b3 librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Archive::Child::Child(llvm::object::Archive const*, char const*, llvm::Error*) + 131
    frame #3: 0x000000010843528d librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Archive::Child::Child(llvm::object::Archive const*, char const*, llvm::Error*) + 45
    frame #4: 0x0000000107ca4e73 librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Archive::child_iterator::child_iterator(this=0x000000010da13bc0) + 35 at Archive.h:135
    frame #5: 0x0000000107ca4e45 librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Archive::child_iterator::child_iterator(this=0x000000010da13bc0) + 21 at Archive.h:135
    frame #6: 0x0000000107ca4e09 librustc_llvm-d85452a8c927d1f5.dylib`RustArchiveIterator::RustArchiveIterator(this=0x000000010da13bc0) + 25 at ArchiveWrapper.cpp:29
    frame #7: 0x0000000107c9a905 librustc_llvm-d85452a8c927d1f5.dylib`RustArchiveIterator::RustArchiveIterator(this=0x000000010da13bc0) + 21 at ArchiveWrapper.cpp:29
    frame #8: 0x0000000107c9a6b5 librustc_llvm-d85452a8c927d1f5.dylib`::LLVMRustArchiveIteratorNew(ra=0x000000010da128a0) + 85 at ArchiveWrapper.cpp:108
    frame #9: 0x0000000107c7c90b librustc_llvm-d85452a8c927d1f5.dylib`rustc_llvm::archive_ro::{{impl}}::iter(self=0x000070001049e8f0) + 43 at archive_ro.rs:73
    frame #10: 0x0000000102675095 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::{{impl}}::new(ar=(ptr = 0x000000010da128a0)) + 117 at loader.rs:735
    frame #11: 0x00000001026760a9 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::get_metadata_section_imp(target=0x00007000104bb8a0, flavor=Rlib, filename=0x000000010da128b0) + 1545 at loader.rs:794
    frame #12: 0x000000010267578e librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::get_metadata_section(target=0x00007000104bb8a0, flavor=Rlib, filename=0x000000010da128b0) + 142 at loader.rs:771
    frame #13: 0x0000000102671d90 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::{{impl}}::extract_one(self=0x00007000104a2ec0, m=HashMap<std::path::PathBuf, rustc::session::search_paths::PathKind, core::hash::BuildHasherDefault<rustc_data_structures::fnv::FnvHasher>> @ 0x00007000104a0640, flavor=Rlib, slot=0x00007000104a0b00) + 1616 at loader.rs:549
    frame #14: 0x00000001026747df librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::{{impl}}::find_commandline_library<collections::btree::set::Iter<collections::string::String>>(self=0x00007000104a2ec0, locs=Iter<collections::string::String> @ 0x00007000104a0cd0) + 1295 at loader.rs:713
    frame #15: 0x000000010266ff51 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::{{impl}}::find_library_crate(self=0x00007000104a2ec0) + 385 at loader.rs:404
    frame #16: 0x000000010266d97c librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::loader::{{impl}}::maybe_load_library_crate(self=0x00007000104a2ec0) + 44 at loader.rs:314
    frame #17: 0x00000001026618c8 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::creader::{{impl}}::load(self=0x00007000104b0608, loader=0x00007000104a2ec0) + 88 at creader.rs:394
    frame #18: 0x0000000102662f56 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::creader::{{impl}}::read_extension_crate(self=0x00007000104b0608, span=Span @ 0x00007000104a3050, info=0x00007000104a4098) + 2166 at creader.rs:513
    frame #19: 0x000000010266402b librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::creader::{{impl}}::read_macros(self=0x00007000104b0608, item=0x000000010da131f0) + 299 at creader.rs:559
    frame #20: 0x0000000102678e68 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::macro_import::{{impl}}::load_macros(self=0x00007000104b0600, vi=0x000000010da131f0, allows_macros=true, import=Option<std::collections::hash::map::HashMap<syntax::parse::token::InternedString, syntax_pos::Span, core::hash::BuildHasherDefault<rustc_data_structures::fnv::FnvHasher>>> @ 0x00007000104a4980, reexport=HashMap<syntax::parse::token::InternedString, syntax_pos::Span, core::hash::BuildHasherDefault<rustc_data_structures::fnv::FnvHasher>> @ 0x00007000104a4968) + 616 at macro_import.rs:115
    frame #21: 0x0000000102678276 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::macro_import::{{impl}}::load_crate(self=0x00007000104b0600, extern_crate=0x000000010da131f0, allows_macros=true) + 454 at macro_import.rs:94
    frame #22: 0x00000001026780a5 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::macro_import::load_macros(loader=0x00007000104b0600, extern_crate=0x000000010da131f0, allows_macros=true) + 85 at macro_import.rs:40
    frame #23: 0x0000000102669a75 librustc_metadata-15876f14be6b409b.dylib`rustc_metadata::creader::{{impl}}::load_macros(self=0x00007000104b0600, extern_crate=0x000000010da131f0, allows_macros=true) + 85 at creader.rs:1048
[... cut ...]
    frame #67: 0x000000010d22b9c1 libstd-dec491587576f8e3.dylib`std::sys_common::thread::start_thread(main=0x000000010dc07c30) + 97 at thread.rs:21
    frame #68: 0x000000010d255b6d libstd-dec491587576f8e3.dylib`std::sys::thread::{{impl}}::new::thread_start(main=0x000000010dc07c30) + 29 at thread.rs:71
    frame #69: 0x00007fffad46dabb libsystem_pthread.dylib`_pthread_body + 180
    frame #70: 0x00007fffad46da07 libsystem_pthread.dylib`_pthread_start + 286
    frame #71: 0x00007fffad46d231 libsystem_pthread.dylib`thread_start + 13
```

shepmaster, 2016-10-02:
I got my debugging info setup correctly:

```
frame #3: 0x00000001084335bd librustc_llvm-d85452a8c927d1f5.dylib`llvm::object::Archive::Child::Child(this=0x000000010f40b1c0, Parent=0x0000000000000000, Start=0x0000000000000000, Err=0x0000000000000000) + 45 at Archive.cpp:310
   307
   308 	Archive::Child::Child(const Archive *Parent, const char *Start, Error *Err)
   309 	    : Parent(Parent), Header(Parent, Start, Parent->getData().size() -
-> 310 	                             (Start - Parent->getData().data()), Err) {
   311 	  if (!Start)
   312 	    return;
   313```

so `Parent` and `Start` are null. I did cherry-pick your change c9ec72ab8f1ee61644d41db8ebe19757336bcffb.

dylanmckay, 2016-10-03:
I'm looking at that too. I believe this was broken in [r276686](http://llvm.org/viewvc/llvm-project?view=revision&revision=276686). There is still a default constructor for `Archive::Child` but it gives `nullptr` as the `Parent` which will always segfault.

Upstream LLVM regression :(

shepmaster, 2016-10-03:
> Upstream LLVM regression :(

So what's the right process here?

dylanmckay, 2016-10-03:
We can email `llvm-commits` and CC in the original author, or we could fix it ourselves and I could upstream it.

The fix will basically be to change this in `Archive.cpp`

```cpp
Archive::Child::Child(const Archive *Parent, const char *Start, Error *Err)
    : Parent(Parent), Header(Parent, Start, Parent->getData().size() -
                              (Start - Parent->getData().data()), Err) {
```

So that it handles `Parent` being null. One issue is that `Header` itself doesn't have a default constructor, and it is a non-nullable field on `Archive::Child` so it's not as easy as calling a different constructor.

dylanmckay, 2016-10-04:
I've just sent an email to `llvm-commits` CCing @shepmaster and the original author.

dylanmckay, 2016-10-05:
Lang Hames is currently working on a fix, we should be able to update LLVM again after that and we should be good.

dylanmckay, 2016-10-06:
This has been fixed as of r283387. I'm merging master in again now.

dylanmckay, 2016-10-06:
I needed to rebase on top of master again because Rust stopped building with Apple Clang 8, more merge conflicts, JS backend added, etc.

I think I'm on top of it now, we'll see how this build goes.

dylanmckay, 2016-10-06:
The `ResumeInst` thing has been fixed, I'm now raising #22 to look at the current error I'm getting.

closed 2016-10-06 by dylanmckay