avr-rust/rust-legacy-fork #23: Can't check out sub-repos?

mwm, 2016-10-26:
In a fresh checkout out of avr-support@8974f, ollowing the instructions in the README, the configure step fails with this error:

> Cloning into '/export/mwm/src/github/avr-rust/rust/src/rust-installer'...
> error: no such remote ref 91f0fe0e367edd404ec5576bab070abb92333db9
> Fetched in submodule path 'src/compiler-rt', but it did not contain 91f0fe0e367edd404ec5576bab070abb92333db9. Direct fetching of that commit failed.
> configure: error: git failed

I'm not sure how that's even possible unless I've got a corrupt repo somewhere. But multiple attempts with completely fresh checkouts of everything are failing.

I have no idea how to proceed from here. Any suggestions for what to do or other information I can provide gladly accepted.

Thanks

shepmaster, 2016-10-26:
Don't worry, the error is not your fault; it's (probably) completely on our side! The problem is likely that we've done some development in this repository but failed to push up some of the related repositories.

At this second, stuff is in a bit of flux; I hope to have a chance to hammer out some of the issues after [Rust Belt Rust](http://www.rust-belt-rust.com/) is over!

mwm, 2016-10-31:
Well, Rust Belt Rust is over. Any chance I can get an update to those repos? I understand they may have issues, but hopefully it'll at least build. Thanks.

dylanmckay, 2016-11-01:
I'll check if it was me who forgot to push something, but it was probably @shepmaster

dylanmckay, 2016-11-01:
Looks like @shepmaster  forgot to push `897f485ea`.

dylanmckay, 2016-11-01:
Last person to touch that directory was @shepmaster

shepmaster, 2016-11-02:
Yuck, what state *did* I leave this in...

shepmaster, 2016-11-02:
Wait, that commit **does** exist:

avr-llvm/compiler-rt@91f0fe0e367edd404ec5576bab070abb92333db9

Can you check to see what remote your compiler-rt is using?

mwm, 2016-11-02:
I think what you want is this, from .git/modules/src/compiler-rt/config:

```
[remote "origin"]
        url = https://github.com/avr-rust/compiler-rt.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "avr-rust-2015-01-08-do-not-delete"]
        remote = origin
        merge = refs/heads/avr-rust-2015-01-08-do-not-delete
```

nateozem, 2016-11-02:
For me to resolve this, following these steps should help.

**Deinit Submodule**

    $ git submodule deinit -f .

**Remove Project Git Submodules**

    $ rm -fR .git/modules/src

**Edit .gitmodules**
Open up .gitmodules and replace the url key with `https://github.com/avr-llvm/compiler-rt`  under compiler-rt section

    url = https://github.com/avr-llvm/compiler-rt

**Reinit & Update**

    $ git submodule init
    $ git submodule update

Let me know if it works.

dylanmckay, 2016-11-02:
@nateozem What does `git diff` look like after that?

If that fixes it, would you be able to open a PR :)

mwm, 2016-11-02:
Ok, I can check out the repos now. The build is failing with a missing library error. I think it's getting confused by my having multiple C compilers installed. In particular, FreeBSD used clang as the system compiler, but config says it's using the gcc install. I'll investigate, and open a new issue if I can't sort it out.

Thanks!

nateozem, 2016-11-02:
Here is the diff:
```
diff --git a/.gitmodules b/.gitmodules
index c8ed2f6..32669c7 100644
--- a/.gitmodules
+++ b/.gitmodules
@@ -4,7 +4,7 @@
        branch = master
 [submodule "src/compiler-rt"]
        path = src/compiler-rt
-       url = https://github.com/avr-rust/compiler-rt.git
+       url = https://github.com/avr-llvm/compiler-rt
 [submodule "src/rt/hoedown"]
        path = src/rt/hoedown
        url = https://github.com/rust-lang/hoedown.git
```

It changed only one file.
I can open a PR.

dylanmckay, 2016-11-03:
> I can open a PR.

That'd be really good :)

dylanmckay, 2016-11-03:
Merged!

closed 2016-11-03 by dylanmckay