avr-rust/rust-legacy-fork #25: Rebase onto rust master?

arianvp, 2016-12-01:
So,  `llvm-avr` was merged into `llvm` proper. This means this project can probably be less experimental in nature.  What changes were actually made to make it work? and would it be difficult to rebase these changes on the current rust master branch?

dylanmckay, 2016-12-01:
In general the changes to Rust are really minimal, by far the hardest bit is wrangling Rust's custom fork of LLVM to get everything working.

At this point it isn't any easier to manage with AVR-LLVM being in-tree - we still have to deal with conflicts and mismatches between LLVM versions.

Once Rust updates their fork of LLVM however, it we will get all of that for free. Upstream Rust's LLVM will have everything required.

The update is currently in progress.

rust-lang/rust#37609

shepmaster, 2016-12-01:
Duplicate of #22

closed 2016-12-01 by shepmaster

shepmaster, 2016-12-01:
> Once Rust updates their fork of LLVM however, it we will get all of that for free

And to clarify, we still wont be able to compile arbitrary Rust programs for AVR. There are still a number of code generation issues.