avr-rust/rust-legacy-fork #26: Segfault in Greedy Register Allocator

shepmaster, 2017-02-15:
Working on minimizing this further, but... 

Based on cfeab8d97d30d74ecbad027e3ac88d1aef972ce9; compiling libcore...

[bugpoint-reduced-simplified.ll.zip](https://github.com/avr-rust/rust/files/778700/bugpoint-reduced-simplified.ll.zip)

dylanmckay, 2017-02-16:
This sounds like it might be a bug mentioned to me on llvm-commits a while ago. I'll fetch more details later.

dylanmckay, 2017-02-18:
Here's the bug I was thinking of

http://llvm.org/viewvc/llvm-project?view=revision&revision=293934

> [LiveRangeEdit] Don't mess up with LiveInterval when a new vreg is created.
>
> In r283838, we added the capability of splitting unspillable register.
When doing so we had to make sure the split live-ranges were also
unspillable and we did that by marking the related live-ranges in the
delegate method that is called when a new vreg is created.
However, by accessing the live-range there, we also triggered their lazy
computation (LiveIntervalAnalysis::getInterval) which is not what we
want in general. Indeed, later code in LiveRangeEdit is going to build
the live-ranges this lazy computation may mess up that computation
resulting in assertion failures. Namely, the createEmptyIntervalFrom
method expect that the live-range is going to be empty, not computed.
>
> Thanks to Mikael Holmén <mikael.holmen@ericsson.com> for noticing and
reporting the problem.

It doesn't sound as relevant now that I remember it.

shepmaster, 2017-04-19:
I didn't re-bugpoint, but a segfault is still present even with the proposed LLVM 4.0 merge into Rust:

```
* thread #2, name = 'rustc', stop reason = EXC_BAD_ACCESS (code=1, address=0x28)
  * frame #0: 0x0000000102d97326 librustc_llvm-8af32171c42232fc.dylib`llvm::MachineInstr::addRegisterDead(unsigned int, llvm::TargetRegisterInfo const*, bool) + 230
    frame #1: 0x0000000102d3781b librustc_llvm-8af32171c42232fc.dylib`llvm::LiveIntervals::computeDeadValues(llvm::LiveInterval&, llvm::SmallVectorImpl<llvm::MachineInstr*>*) + 491
    frame #2: 0x0000000102d37e1c librustc_llvm-8af32171c42232fc.dylib`llvm::LiveIntervals::shrinkToUses(llvm::LiveInterval*, llvm::SmallVectorImpl<llvm::MachineInstr*>*) + 972
    frame #3: 0x0000000102d4d882 librustc_llvm-8af32171c42232fc.dylib`llvm::LiveRangeEdit::eliminateDeadDefs(llvm::SmallVectorImpl<llvm::MachineInstr*>&, llvm::ArrayRef<unsigned int>, llvm::AAResults*) + 514
    frame #4: 0x0000000102d19206 librustc_llvm-8af32171c42232fc.dylib`(anonymous namespace)::InlineSpiller::postOptimization() + 14502
    frame #5: 0x0000000102e28210 librustc_llvm-8af32171c42232fc.dylib`llvm::RegAllocBase::postOptimization() + 32
    frame #6: 0x0000000102e30114 librustc_llvm-8af32171c42232fc.dylib`(anonymous namespace)::RAGreedy::runOnMachineFunction(llvm::MachineFunction&) + 3412
    frame #7: 0x0000000102d8c716 librustc_llvm-8af32171c42232fc.dylib`llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 134
    frame #8: 0x0000000103493352 librustc_llvm-8af32171c42232fc.dylib`llvm::FPPassManager::runOnFunction(llvm::Function&) + 530
    frame #9: 0x0000000103493553 librustc_llvm-8af32171c42232fc.dylib`llvm::FPPassManager::runOnModule(llvm::Module&) + 51
    frame #10: 0x00000001034939ea librustc_llvm-8af32171c42232fc.dylib`llvm::legacy::PassManagerImpl::run(llvm::Module&) + 922
    frame #11: 0x00000001020bcbed librustc_llvm-8af32171c42232fc.dylib`::LLVMRustWriteOutputFile(Target=0x000000011b146800, PMR=0x0000000104d03d50, M=0x0000000107c49de0, Path=<unavailable>, RustFileType=<unavailable>) at PassWrapper.cpp:477 [opt]
```

shepmaster, 2017-04-28:
I tried cherry-picking the commit you mentioned (https://github.com/llvm-mirror/llvm/commit/5340452c64bb5008ab3f2007bd1b09563e7e5b65) but it seems like I get the same assertion.

shepmaster, 2017-04-28:
Looking in the debugger, it seems like some data got out of alignment somewhere. I think it's saying that the store came from offset 1676, but that's blank:

```
(lldb) p Def.dump()
1676r
(lldb) p Indexes->dump()
1640 %vreg89<def> = COPY %vreg88; DREGS:%vreg89,%vreg88
1648 CPWRdRr %vreg89, %vreg67, %SREG<imp-def>; DREGS:%vreg89,%vreg67
1664
1672 %vreg71<earlyclobber,def> = LDDWRdYQ <fi#4>, 0; mem:LD2[FixedStack4](align=1) DREGS:%vreg71
1676
1684 %vreg84<earlyclobber,def> = LDDWRdYQ <fi#2>, 0; mem:LD2[FixedStack2](align=1) IWREGS:%vreg84
1692 BRLOk <BB#1>, %SREG<imp-use,kill>
1696 RJMPk <BB#2>
1712
```

brainlag, 2018-02-21:
I tried to debug this quite a bit. With the latest llvm this changed a bit. Now `Def.dump()` is `1832r` which does not point to an empty machine instruction but between two machine instructions which is not valid(?). The offset between two machine instructions is always 16 but 1832 % 16 is 8 and not 0. I tried to figure out where this invalid live interval is created but these breakpoints are never hit. For me this looks like some kind of corruption but I don't have enough experience with C++ to make an educated guess.

brainlag, 2018-03-05:
So after a lot of debugging I still don't know whats going on but here are my findings.
There is no corruption just some iterator magic which leads to a machine instruction at 1832.

Here is the instruction map before 1832 gets removed:
```
1800B	  %138:dregs = COPY %136:dregs
1808B	  %139:dldregs = LDIWRdK 14
1824B	  %102:dregs = COPY %138:dregs
1832B	  early-clobber %109:dldregs = LDDWRdYQ %stack.2, 0; mem:LD2[FixedStack2](align=1)
1840B	  early-clobber %120:iwregs = LDDWRdYQ %stack.1, 0; mem:LD2[FixedStack1](align=1)
1848B	  RJMPk %bb.22
```

The function which removes it is InlineSpiller::coalesceStackAccess. So I tried to prevent that function to remove the instruction but the result is the same. The InlineSpiller remvoes the instruction somewhere in the postOptimization step.

For some reason we end up with this total broken live range which then blows up:
```
%149 [1832r,1832d:0)[2744r,2744d:1)[2760B,2760d:2)  0@1832r 1@2744r 2@2760B-phi weight:1.106830e-03`
```
All 3 indexes point to same machine instruction:

```
early-clobber %XXX:dldregs = LDDWRdYQ %stack.2, 0; mem:LD2[FixedStack2](align=1)
```

I also found this somewhat related bug: 

http://lists.llvm.org/pipermail/llvm-dev/2012-July/051741.html
https://bugs.llvm.org/show_bug.cgi?id=13375

Maybe someone has an idea what is going on?

dylanmckay, 2018-03-06:
Interestingly, the OP of the [mailing list](https://lists.llvm.org/pipermail/llvm-dev/2012-July/051741.html) post you linked is one of the three original people that worked on AVR-LLVM, so it's fairly likely it's related, if not the same!

dylanmckay, 2018-03-06:
Cherry picking this from the mailing list thread

> OK, I see what is happening. We don't expect foldMemoryOperand to turn a normal def into an early-clobber and vice versa.
>
> This is not easy to fix, could you file a PR, please?
>
> As a workaround, you can use a pseudo-instruction in loadRegFromStackSlot() that doesn't have the early-clobber flag. Then replace it with the real instruction in expandPostRAPseudo().
>
> /jakob

For what it's worth, the original llvm-avr sources had a comment on LDDWRdYQ to the effect of it being a workaround for something. It was never clear what it was a workaround for though.

dylanmckay, 2018-03-06:
We could fix this properly, or we could work around it by adding a guard condition to skip [the call to `foldMemoryOperand`](https://github.com/llvm-mirror/llvm/blob/209f04866320b6e58b78c06876e77254bd5d6102/lib/CodeGen/InlineSpiller.cpp#L988) if the slot is not an earlyclobber slot and the folded instruction has earlyclobber operands.

I suspect that it will be unnecessary to check for earlyclobber slots because if the existing slots always come from `COPY` pseudo instructions, then earlyclobber will never be present on this generic instruction.

Would be nice to fix it properly though

dylanmckay, 2018-03-06:
Good news!

The mailing list thread and bug report shown in @brainlag's investigation reminded me of a very old `FIXME` comment left in the AVR backend since 2010. At one point, I removed a small number of historic TODOs/FIXMEs, because most of them were very vague and short, no longer making sense to anybody.

At some point, the earlyclobber flag was added to LDDWRdYQ, which was originally introduced to work around this bug. albeit with a completely different way of showing itself.

I've removed the earlyclobber flag which fixes this issue. At some point it would be nice to see the register allocator esoteric handle cases like this though.

dylanmckay, 2018-03-06:
Here's the important bit

https://github.com/llvm-mirror/llvm/commit/d45c0f1adf57831f068af99ef92570a15672f1ea

dylanmckay, 2018-03-06:
Thanks for all of the hours spent @brainlag, it took a few hours to put all of the pieces together, but the debugging you did filled the last piece of the puzzle!

dylanmckay, 2018-03-06:
We should cherry-pick this, but it's probably not worth it until #90 lands

brainlag, 2018-03-06:
I think i saw a similar workaround in MIPS and/or ARM but those didn't mention the bug.

brainlag, 2018-03-18:
I still don't know what this earlyclobber flag is supposed to do but don't you think we also should remove it from LDDWRdPtrQ too? This fixes #37 for me and also #95. And then there is still LDDRdPtrQ, which also has a earlyclobber flag but didn't trigger a error yet.

shepmaster, 2018-03-18:
> This fixes #37 for me and also #95

That's super exciting!

dylanmckay, 2018-03-18:
The `@earlyclobber` flag is there to tell LLVM that the output register and input registers cannot be the same because the instruction itself "clobbers the register's value early", sometime before the actual instruction completes.

In AVR, this is used on pseudo instructions because some of the expansion patterns introduce additional instructions which overwrite some registers.

For example, we often turn 16-bit load pseudo instructions into something like this

```asm
LDD r12, X+ ; load from pointer into r12, postincrement the pointer value
LDD r13, X
```

Because the expanded instruction sequence always write to the source pointer register, that means that none of the other registers in the instruction can overlap with the X register, namely `r27:r26`.

I believe that removing the earlyclobber hint will remove this constraint from the register allocator. This would reduce register pressure, but it will also corrupt the values we store into registers. This could explain why it fixes #37 and #95.

[Here](https://lists.llvm.org/pipermail/llvm-dev/2009-December/028020.html) is the mailing list posting where it was originally introduced.

This is my understanding of earlyclobber at least.

brainlag, 2018-03-18:
So do you think LDDWRdPtrQ is different to LDDWRdYQ or are we getting away without adding earlyclobber to it? [This workaround](https://github.com/llvm-mirror/llvm/blob/d45c0f1adf57831f068af99ef92570a15672f1ea/lib/Target/AVR/AVRInstrInfo.cpp#L83) for bug [13375](https://bugs.llvm.org/show_bug.cgi?id=13375) is for both instructions.

brainlag, 2018-03-19:
Ok never mind LDDWRdYQ is already a special case of LDDWRdPtrQ.

dylanmckay, 2018-03-19:
As you've implied but I'll be concrete

Leaving earlyclobber off `LDDWRdYQ` only works because the pseudo expansion pass will directly replace it with `LDDWRdPtrQ` prior to register allocation, when the constraint actually matters.

This works around the bug in `foldMemoryOperand` that doesn't handle earlyclobber loads. With the workaround, it never handles earlyclobber loads - we essentially add earlyclobber after the fact by replacing the instruction opcode with one that does specify earlyclobber.

shepmaster, 2018-05-03:
Here's a smaller testcase:

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-3cab020.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

%"num::bignum::Big32x40" = type { [0 x i8], i16, [0 x i8], [40 x i32], [0 x i8] }

define void @"core::num::flt2dec::strategy::dragon::format_exact"() unnamed_addr {
start:
  %_72 = alloca %"num::bignum::Big32x40", align 1
  %0 = getelementptr inbounds %"num::bignum::Big32x40", %"num::bignum::Big32x40"* null, i16 0, i32 3
  %1 = bitcast [40 x i32]* %0 to i8*
  %2 = getelementptr inbounds %"num::bignum::Big32x40", %"num::bignum::Big32x40"* %_72, i16 0, i32 3
  %3 = bitcast [40 x i32]* %2 to i8*
  call void @llvm.memcpy.p0i8.p0i8.i16(i8* nonnull %3, i8* nonnull %1, i16 160, i32 1, i1 false)
  %4 = getelementptr inbounds %"num::bignum::Big32x40", %"num::bignum::Big32x40"* %_72, i16 0, i32 3, i16 0
  br label %"core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut.exit.i.i"

"core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut.exit.i.i": ; preds = %"core::num::bignum::Big32x40::div_rem_small.exit.i.bb2.i.i_crit_edge", %start
  br i1 undef, label %"core::num::bignum::Big32x40::div_rem_small.exit.i", label %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.lr.ph.i.i"

"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.lr.ph.i.i": ; preds = %"core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut.exit.i.i"
  br label %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.i.i"

"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.i.i": ; preds = %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.i.i", %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.lr.ph.i.i"
  %5 = load i32, i32* undef, align 1
  %6 = zext i32 %5 to i64
  %7 = or i64 0, %6
  %8 = udiv i64 %7, 1000000000
  %9 = trunc i64 %8 to i32
  store i32 %9, i32* undef, align 1
  %10 = icmp eq i32* undef, %4
  br i1 %10, label %"core::num::bignum::Big32x40::div_rem_small.exit.i", label %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.i.i"

"core::num::bignum::Big32x40::div_rem_small.exit.i": ; preds = %"<u32 as core::num::bignum::FullOps>::full_div_rem.exit.i.i", %"core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut.exit.i.i"
  %11 = icmp ugt i16 undef, 9
  br i1 %11, label %"core::num::bignum::Big32x40::div_rem_small.exit.i.bb2.i.i_crit_edge", label %bb7.i74

"core::num::bignum::Big32x40::div_rem_small.exit.i.bb2.i.i_crit_edge": ; preds = %"core::num::bignum::Big32x40::div_rem_small.exit.i"
  br label %"core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut.exit.i.i"

bb7.i74:                                          ; preds = %"core::num::bignum::Big32x40::div_rem_small.exit.i"
  unreachable
}

; Function Attrs: argmemonly nounwind
declare void @llvm.memcpy.p0i8.p0i8.i16(i8* nocapture writeonly, i8* nocapture readonly, i16, i32, i1) #0

attributes #0 = { argmemonly nounwind }
```