avr-rust/rust-legacy-fork #27: Assertion failed: (DstReg != SrcReg && "SrcReg and DstReg cannot be the same")

dylanmckay, 2017-04-25:
While compiling Rust libcore

I've got a (massive) testcase 
[libcore.ll](https://github.com/avr-rust/rust/files/955046/libcore.ll.txt). This is the entire Rust libcore as an LLVM IR program.

Reproduction

```bash
./bin/llc ~/libcore.ll -O0 -o /dev/null
```

```
Assertion failed: (DstReg != SrcReg && "SrcReg and DstReg cannot be the same"), function expand, file /Users/dylan/projects/llvm-project/llvm/lib/Target/AVR/AVRExpandPseudoInsts.cpp, line 590.
0  libLLVMSupport.dylib     0x0000000111c333ac llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 60
1  libLLVMSupport.dylib     0x0000000111c338f9 PrintStackTraceSignalHandler(void*) + 25
2  libLLVMSupport.dylib     0x0000000111c2fa29 llvm::sys::RunSignalHandlers() + 425
3  libLLVMSupport.dylib     0x0000000111c33c42 SignalHandler(int) + 354
4  libsystem_platform.dylib 0x00007fff925b6b3a _sigtramp + 26
5  libsystem_platform.dylib 0xffffffffffffffff _sigtramp + 1839502559
6  libsystem_c.dylib        0x00007fff9243b420 abort + 129
7  libsystem_c.dylib        0x00007fff92402893 basename_r + 0
8  libLLVMAVRCodeGen.dylib  0x000000010d4f73e5 bool (anonymous namespace)::AVRExpandPseudo::expand<184u>(llvm::MachineBasicBlock&, llvm::MachineInstrBundleIterator<llvm::MachineInstr, false>) + 277
9  libLLVMAVRCodeGen.dylib  0x000000010d4f4470 (anonymous namespace)::AVRExpandPseudo::expandMI(llvm::MachineBasicBlock&, llvm::MachineInstrBundleIterator<llvm::MachineInstr, false>) + 1184
10 libLLVMAVRCodeGen.dylib  0x000000010d4f3c7c (anonymous namespace)::AVRExpandPseudo::expandMBB(llvm::MachineBasicBlock&) + 316
11 libLLVMAVRCodeGen.dylib  0x000000010d4f36fa (anonymous namespace)::AVRExpandPseudo::runOnMachineFunction(llvm::MachineFunction&) + 298
12 libLLVMCodeGen.dylib     0x000000010ef26b51 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 449
13 libLLVMCore.dylib        0x000000010ff3679f llvm::FPPassManager::runOnFunction(llvm::Function&) + 399
14 libLLVMCore.dylib        0x000000010ff36ca5 llvm::FPPassManager::runOnModule(llvm::Module&) + 117
15 libLLVMCore.dylib        0x000000010ff37a74 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 2196
16 libLLVMCore.dylib        0x000000010ff36f66 llvm::legacy::PassManagerImpl::run(llvm::Module&) + 342
17 libLLVMCore.dylib        0x000000010ff387a1 llvm::legacy::PassManager::run(llvm::Module&) + 33
18 llc                      0x0000000108b462b2 compileModule(char**, llvm::LLVMContext&) + 21810
19 llc                      0x0000000108b4017f main + 2447
20 libdyld.dylib            0x00007fff923a7235 start + 1
Stack dump:
0.      Program arguments: ./bin/llc /Users/dylan/libcore.ll -O0 -o /dev/null
1.      Running pass 'Function Pass Manager' on module '/Users/dylan/libcore.ll'.
2.      Running pass 'AVR pseudo instruction expansion pass' on function '@_ZN4core4sync6atomic11atomic_load17h9977d1b8983da2bdE'
[1]    55133 abort      ./bin/llc ~/libcore.ll -O0 -o /dev/null
```

dylanmckay, 2017-04-25:
I have a patch which fixes this, will upstream when I've tested it further.

dylanmckay, 2017-04-26:
I've got a patch, it's been merged into LLVM trunk

* `Support the LDWRdPtr instruction with the same Src+Dst register` ([here](https://github.com/llvm-mirror/llvm/commit/0226ba23734092ef63e157196c135fc7e902d27a))

libcore still doesn't build for another problem though - will raise something else.

shepmaster, 2017-04-26:
> merged into LLVM trunk

Will targeting trunk cause troubles to try and get the patches into Rust's LLVM?

dylanmckay, 2017-04-26:
> Will targeting trunk cause troubles to try and get the patches into Rust's LLVM?

I'm not entirely sure to be honest. I'd really like it if we could just cherry-pick changes into Rust's LLVM. That should be alright because the changes will get picked up anyway when we upgrade LLVM next.

The only problem is that I'm not sure if we can do that - we're supposed to be tracking emscripten's version of LLVM so I'm not sure what the Rust team think about adding more patches into the fork. I guess there is already going to be all the Rust-specific LLVM patches in there.

shepmaster, 2017-05-01:
Cherry picked llvm-mirror/llvm@0226ba23734092ef63e157196c135fc7e902d27a to avr-rust/llvm

closed 2017-05-02 by dylanmckay