avr-rust/rust-legacy-fork #29: Merge changes into upstream Rust

dylanmckay, 2017-04-25:
This issue is for discussion on upstreaming AVR support to upstream Rust.

Things we need to do first

- [ ] Successfully compile the `libcore` library

One thing that I am concerned about is how hard it is going to be in order to add avr-llvm patches in order to fix codegen bugs - if AVR support is in Rust `master`, we will have to get AVR backend patches into the Rust LLVM fork.

shepmaster, 2017-04-26:
From #27:

>> Will targeting trunk cause troubles to try and get the patches into Rust's LLVM?
>
> I'm not entirely sure to be honest. I'd really like it if we could just cherry-pick changes into Rust's LLVM. That should be alright because the changes will get picked up anyway when we upgrade LLVM next.
>
> The only problem is that I'm not sure if we can do that - we're supposed to be tracking emscripten's version of LLVM so I'm not sure what the Rust team think about adding more patches into the fork. I guess there is already going to be all the Rust-specific LLVM patches in there.

If I understand correctly:

- Emscripten has forked off from LLVM 4.0
- Rust has forked off from Emscripten

If patches are submitted to LLVM *trunk*, then at some point, the relevant base code will have diverged and cherry-picks into 4.0 will start to accumulate conflicts. If patches are submitted to LLVM 4.0, then the same thing can happen in reverse (or the patches get lost, which would be worse). The only real solution there is to make sure that Rust / Emscripten continue to track LLVM trunk, which seems not-fully-likely.

Then there's the question of "will Rust accept AVR-specific patches to the Rust LLVM fork". I'll try and ask my contacts to see what they think.

shepmaster, 2017-04-26:
```
09:51 < acrichto> shep: we don't really have a policy around llvm upgrades right now
09:51 < acrichto> but historically
09:51 < acrichto> a) backports of merged patches are totally fine
09:52 < acrichto> b) upgrades to llvm HEAD are also ok if they pass tests
```

So it seems like we should be more-or-less good to go in that area.

dylanmckay, 2017-08-23:
Have raised rust-lang/rust#44052 to see what the Rust folks think.

dylanmckay, 2019-01-21:
A list of bugfixes to be upstreamed is at #116.