avr-rust/rust-legacy-fork #30: Rework C-ABI translation

shepmaster, 2017-04-25:
Looks like f0636b61c7f84962a609e831760db9d77f4f5e14 rewrote how all of the layout / alignment / LLVM type stuff is handled (*insert handwaving here*). I've had to completely comment it out to proceed with compilation, but I'm pretty sure any resulting compiler will be inoperable.

@dylanmckay I can push up (yet another) branch with this WIP if you'd be able to do that brain-thinking-thing you know how do to?

dylanmckay, 2017-04-25:
> @dylanmckay I can push up (yet another) branch with this WIP if you'd be able to do that brain-thinking-thing you know how do to?

Might as well give it a go :)

shepmaster, 2017-04-25:
Hurrah! https://github.com/avr-rust/rust/tree/avr-support-4.0-abi-hack

dylanmckay, 2017-04-26:
I think I got it working at #33 

The build succeeds locally, and it successfully converts a small Rust program to assembly.

closed 2017-04-26 by shepmaster

shepmaster, 2017-04-26:
Cool. I've merged, removed the commenting / revert of commenting, rebased on latest latest Rust, and am doing a fresh build. If it compiles, I'm pushing to `avr-support` and we can run from there.