avr-rust/rust-legacy-fork #32: LDDWRdYQ - Bad machine code: Using an undefined physical register

dylanmckay, 2017-04-26:
Found while compiling `libcore` with the patch from #27 applied

[libcore.ll](https://github.com/avr-rust/rust/files/955046/libcore.ll.txt)

Reproduction

```bash
./bin/llc ~/libcore.ll -O0 -o /dev/null
```

```
*** Bad machine code: Using an undefined physical register ***
- function:    _ZN76_$LT$core..char..EscapeDefault$u20$as$u20$core..iter..iterator..Iterator$GT$4next17hd387834b920a4d78E
- basic block: BB#3 bb2 (0x7fb1b59aea38)
- instruction: %R29R28<earlyclobber,def> = LDDWRdYQ
- operand 1:   %R29R28

*** Bad machine code: Using an undefined physical register ***
- function:    _ZN76_$LT$core..char..EscapeDefault$u20$as$u20$core..iter..iterator..Iterator$GT$4next17hd387834b920a4d78E
- basic block: BB#3 bb2 (0x7fb1b59aea38)
- instruction: %R31R30<earlyclobber,def> = LDDWRdYQ
- operand 1:   %R29R28

*** Bad machine code: Using an undefined physical register ***
- function:    _ZN76_$LT$core..char..EscapeDefault$u20$as$u20$core..iter..iterator..Iterator$GT$4next17hd387834b920a4d78E
- basic block: BB#3 bb2 (0x7fb1b59aea38)
- instruction: %R21R20<earlyclobber,def> = LDDWRdYQ
- operand 1:   %R29R28
LLVM ERROR: Found 3 machine code errors.
```

dylanmckay, 2017-04-26:
I've got an upstreamed fix [here](https://github.com/llvm-mirror/llvm/commit/f23578fbc3fa1c096372b0a4fa3bb38b6d156027)

shepmaster, 2017-05-01:
Cherry-picked llvm-mirror/llvm@f23578fbc3fa1c096372b0a4fa3bb38b6d156027 into avr-rust/llvm

closed 2017-05-02 by shepmaster