avr-rust/rust-legacy-fork #34: ICE: Layout::compute: unexpected type `[type error]`

dylanmckay, 2017-04-26:
I get this while compiling `libcore`.

```
error: internal compiler error: /home/dylan/projects/rust-lang/rust/src/librustc/ty/layout.rs:1485: Layout::compute: unexpected type `[type error]`

note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/blob/master/CONTRIBUTING.md#bug-reports

thread 'rustc' panicked at 'Box<Any>', /home/dylan/projects/rust-lang/rust/src/librustc_errors/lib.rs:416
note: Run with `RUST_BACKTRACE=1` for a backtrace.
```

The bug must have been broken somewhere inbetween `avr-support-4.0-timnn..master` (note that master is at 1db0bbffc37bb10536313034a6684bdc7569f957 at the time of writing). Previously, the Rust compiler managed to get all the way to codegen, where LLVM issues would stop us.

Because of this problem, `libcore` cannot be built.

This may be caused by my fix for #30, but I'm not sure.

cc @shepmaster

shepmaster, 2017-04-27:
I think this should be the changes:

```
git log 0777c757a6832dc5f8f218377f99960f5477311f..b0a4074c5e
```

shepmaster, 2017-04-27:
I'm chatting with some compiler devs and building with debug information, see if we can find anything useful. Right now, they are surprised because the `TyError` exists farther than would be expected.

shepmaster, 2017-04-27:
```
15:35 < eddyb> shep: oh no need
15:35 < eddyb> shep: I'm laughing because it's obviously wrong
15:36 < eddyb> shep: or rather, the comment is a lie now
15:36 < shep> when does a fat pointer have both halves the same value?
15:36 < eddyb> https://github.com/rust-lang/rust/commit/f1f453cf3b124711b2e20315937403d89913bcd2#diff-57cb116a80bef6bc27ee1864378f8260R538
15:36 < eddyb> shep: null slice? idk
15:36 < eddyb> shep: not that it matters
15:37 < eddyb> I think I'm responsible for both the original .err and the thing that looks at the .err
15:37 < eddyb> how only AVR hit it is a mystery to me
15:37 < eddyb> shep: do you do something custom for panics?
15:37 < shep> (Oh they aren't the same:  b38 vs c38)
15:37 < eddyb> ah heh
15:37 < eddyb> I didn't notice either
15:38 < shep> eddyb: we haven't gotten that far IIRC. This should just be compiling stock libcore.
15:41 < shep> eddyb: for my own education, what should that closure be doing?
15:42 < eddyb> shep: the closure is uninteresting
15:42 < eddyb> shep: but the code in question is splitting a fat pointer value into its two halves and passing them as two argument
15:42 < eddyb> *arguments
15:42 < shep> yup yup
15:43 < eddyb> it's going to die here https://github.com/rust-lang/rust/blob/master/src/librustc_trans/mir/block.rs#L638
15:43 < eddyb> I'm *very* confused
15:44 < eddyb> how this isn't hit more often
15:44 < eddyb> because I fixed this exact bug in like half a dozen places
15:45 < eddyb> wait
15:45 < eddyb> no it can't be a pair
15:45 < eddyb> shep: oooooh
15:45 < eddyb> shep: it must be this https://github.com/rust-lang/rust/blob/master/src/librustc_trans/mir/block.rs#L635
15:46 < eddyb> and it hits https://github.com/rust-lang/rust/blob/master/src/librustc_trans/mir/operand.rs#L303
15:46 < eddyb> shep: okay this is messed up :D
15:46 < shep> yeah: https://github.com/avr-rust/rust/blob/1db0bbffc37bb10536313034a6684bdc7569f957/src/librustc_trans/cabi_avr.rs#L20-L28
15:47 < shep> we make lots of things indirect
15:47 < eddyb> AHAHAHAHAHAHAHAHAHAHA
15:47 < eddyb> shep: you updated wrong
15:47 < eddyb> glad I caught it
15:47 < shep> (ohdear)
15:47 < eddyb> shep: you have a flipped condition
15:47 < shep> should be aggregate => indirect?
15:47 < eddyb> shep: you pass as immediate iff the argument/return is an aggregate
15:48 < eddyb> shep: yeah
15:48 < shep> >  I looked at cabi_msp430.rs and cabi_mips.rs for inspiration.
15:48 < eddyb> aggregate implies members
15:48 < eddyb> shep: hmm maybe those wre arong too? lemme check
15:48 < eddyb> if it's my fault.. then... well
15:49 < eddyb> we need more automatic testing :(
15:49 < shep> FWIW, I'm just the messenger here, I try to follow along as best I can
15:49 < eddyb> shep: https://github.com/rust-lang/rust/blob/master/src/librustc_trans/cabi_mips.rs#L16
15:49 < eddyb> note the !
15:49 < shep> But I'm not well-versed enough to do this on my own
15:49 < eddyb> shep: yeah it's just a missing ! in each function
15:49 < eddyb> it should work fine after that
15:50 < eddyb> shep: the err thing is a bug only if your ABI can't pass the two halves of a fat pointer as one immediate each which can't be a thing
15:50 < eddyb> at least not in LLVM
15:50 < eddyb> that's why it never triggered
15:50 < eddyb> shep: it will probably get cleaned up as I move the ABI out of librustc_trans fwiw
15:51 < shep> eddyb: so I was right back in the beginning: something we were doing wrong that the new code showed, so that's a +1 for the new code
15:52 < eddyb> shep: no I think this used to be correct :P
15:52 < eddyb> shep: i.e. when I changed the API, updating cabi_avr.rs went wrong
15:53 < eddyb> shep: the [type error] "bug" has been there all along, so unless you've seen it before, you weren't doing ABI wrong
```

shepmaster, 2017-04-27:
I'm building with the flipped ABI conditions now...

shepmaster, 2017-04-27:
Should be fixed with 021843646282289b305f45048fde4a5064089048

dylanmckay, 2017-04-27:
Interesting, I believe I also tried flipping the conditions but I don't recall it working.

When I added `println!` statements to the `classify_arg_ty` and `classify_ret_ty` functions, I saw that they only got called 3 times until the bug got hit. In all cases, the type was `()`.

shepmaster, 2017-04-27:
With the updated conditions, compiling libcore leads me to this, so it seemed like we were past this issue:

```
Assertion failed: (MI && "No instruction defining live value"), function computeDeadValues, file /Users/shep/Projects/avr-rust/src/llvm/lib/CodeGen/LiveIntervalAnalysis.cpp, line 489.
```

closed 2017-05-01 by shepmaster