avr-rust/rust-legacy-fork #37: LLVM ERROR: ran out of registers during register allocation

shepmaster, 2017-04-28:
```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-650d16e.bc"
target datalayout = "e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8"
target triple = "avr"

%FmtFormatter.5.12.19.26.33.40.54.68.75.82.89.192 = type { i32, [0 x i8], i32, [0 x i8], %"Option<usize>.0.7.14.21.28.35.49.63.70.77.84.187", [0 x i8], %"Option<usize>.0.7.14.21.28.35.49.63.70.77.84.187", [0 x i8], { i8*, void (i8*)** }, [0 x i8], %"Iter<fmt::ArgumentV1>.4.11.18.25.32.39.53.67.74.81.88.191", [0 x i8], { %ArgumentV1.2.9.16.23.30.37.51.65.72.79.86.189*, i16 }, [0 x i8], i8, [3 x i8] }
%"Option<usize>.0.7.14.21.28.35.49.63.70.77.84.187" = type { i16, [0 x i16], [1 x i16] }
%"Iter<fmt::ArgumentV1>.4.11.18.25.32.39.53.67.74.81.88.191" = type { %ArgumentV1.2.9.16.23.30.37.51.65.72.79.86.189*, [0 x i8], %ArgumentV1.2.9.16.23.30.37.51.65.72.79.86.189*, [0 x i8], %"PhantomData<&fmt::ArgumentV1>.3.10.17.24.31.38.52.66.73.80.87.190", [0 x i8] }
%ArgumentV1.2.9.16.23.30.37.51.65.72.79.86.189 = type { %Void.1.8.15.22.29.36.50.64.71.78.85.188*, [0 x i8], i8 (%Void.1.8.15.22.29.36.50.64.71.78.85.188*, %FmtFormatter.5.12.19.26.33.40.54.68.75.82.89.192*)*, [0 x i8] }
%Void.1.8.15.22.29.36.50.64.71.78.85.188 = type { {}, [0 x i8] }
%"PhantomData<&fmt::ArgumentV1>.3.10.17.24.31.38.52.66.73.80.87.190" = type {}
%EscapeUnicode.6.13.20.27.34.41.55.69.76.83.90.193 = type { i32, [0 x i8], i16, [0 x i8], i8, [1 x i8] }

@str.2y = external constant [7 x i8]

; Function Attrs: uwtable
define void @EscapeDefaultStateDebug(%FmtFormatter.5.12.19.26.33.40.54.68.75.82.89.192* dereferenceable(32)) unnamed_addr #0 personality i32 (...)* @rust_eh_personality {
start:
  %_47 = alloca %EscapeUnicode.6.13.20.27.34.41.55.69.76.83.90.193*, align 2
  switch i2 undef, label %bb4 [
    i2 0, label %_ZN4core3fmt8builders10DebugTuple6finish17h8f1e007edfc4d0a6E.exit
  ]

_ZN4core3fmt8builders10DebugTuple6finish17h8f1e007edfc4d0a6E.exit: ; preds = %start
  ret void

bb4:                                              ; preds = %start
  %1 = getelementptr inbounds %FmtFormatter.5.12.19.26.33.40.54.68.75.82.89.192, %FmtFormatter.5.12.19.26.33.40.54.68.75.82.89.192* %0, i16 0, i32 8, i32 1
  %2 = load void (i8*)**, void (i8*)*** %1, align 2, !noalias !0, !nonnull !9
  %3 = getelementptr inbounds void (i8*)*, void (i8*)** %2, i16 3
  %4 = bitcast void (i8*)** %3 to i8 ({}*, i8*, i16)**
  %5 = load i8 ({}*, i8*, i16)*, i8 ({}*, i8*, i16)** %4, align 2, !invariant.load !9, !noalias !0, !nonnull !9
  %6 = tail call i8 %5({}* nonnull undef, i8* noalias nonnull readonly getelementptr inbounds ([7 x i8], [7 x i8]* @str.2y, i16 0, i16 0), i16 7), !noalias !10
  unreachable
}

declare i32 @rust_eh_personality(...) unnamed_addr

attributes #0 = { uwtable }

!0 = !{!1, !3, !5, !6, !8}
!1 = distinct !{!1, !2, !"_ZN4core3fmt9Formatter9write_str17hf8083fd39f1b0c0cE: argument 0"}
!2 = distinct !{!2, !"_ZN4core3fmt9Formatter9write_str17hf8083fd39f1b0c0cE"}
!3 = distinct !{!3, !4, !"_ZN4core3fmt8builders15debug_tuple_new17ha94131b12062f03fE: argument 0"}
!4 = distinct !{!4, !"_ZN4core3fmt8builders15debug_tuple_new17ha94131b12062f03fE"}
!5 = distinct !{!5, !4, !"_ZN4core3fmt8builders15debug_tuple_new17ha94131b12062f03fE: argument 1"}
!6 = distinct !{!6, !7, !"_ZN4core3fmt9Formatter11debug_tuple17hadc8e3bda8cfb00cE: argument 0"}
!7 = distinct !{!7, !"_ZN4core3fmt9Formatter11debug_tuple17hadc8e3bda8cfb00cE"}
!8 = distinct !{!8, !7, !"_ZN4core3fmt9Formatter11debug_tuple17hadc8e3bda8cfb00cE: argument 1"}
!9 = !{}
!10 = !{!3, !6}
```

I'm not actually sure if Rust or LLVM should be handling this...

shepmaster, 2017-04-28:
[The debug log from llc](https://gist.github.com/shepmaster/dbad5a4521259e753799e14b2e9b01c4)

shepmaster, 2017-04-28:
And the machine code when we are doing register allocation

```
# Machine code for function EscapeDefaultStateDebug: NoPHIs, TracksLiveness
Frame Objects:
  fi#0: size=2, align=2, at location [SP+2]
Function Live Ins: %R25R24 in %vreg0

BB#0: derived from LLVM BB %start
    Live Ins: %R25R24
	%vreg11<def> = COPY %R25R24; DREGS:%vreg11
	%vreg1<def> = LDIRdK 0; LD8:%vreg1
	CPIRdK %vreg1, 0, %SREG<imp-def>, %SREG<imp-use,kill>; LD8:%vreg1
	BRNEk <BB#2>, %SREG<imp-use,kill>
	RJMPk <BB#1>
    Successors according to CFG: BB#1(0x7ffff800 / 0x80000000 = 100.00%) BB#2(0x00000800 / 0x80000000 = 0.00%)

BB#1: derived from LLVM BB %_ZN4core3fmt8builders10DebugTuple6finish17h8f1e007edfc4d0a6E.exit
    Predecessors according to CFG: BB#0
	RET

BB#2: derived from LLVM BB %bb4
    Predecessors according to CFG: BB#0
	%vreg13<def> = COPY %vreg11; DREGS:%vreg13,%vreg11
	%vreg14<def> = COPY %vreg13; PTRDISPREGS:%vreg14 DREGS:%vreg13
	%vreg5<earlyclobber,def> = LDDWRdPtrQ %vreg14, 18; mem:LD2[%2](noalias=!1,!3,!5,!6,!8)(dereferenceable) PTRDISPREGS:%vreg5,%vreg14
	%vreg4<earlyclobber,def> = LDDWRdPtrQ %vreg5, 6; mem:LD2[%5](noalias=!1,!3,!5,!6,!8)(invariant) DREGS:%vreg4 PTRDISPREGS:%vreg5
	ADJCALLSTACKDOWN 0, %SP<imp-def,dead>, %SREG<imp-def,dead>, %SP<imp-use>
	%vreg6<def> = LDIWRdK <ga:@str.2y>; DLDREGS:%vreg6
	%vreg7<def> = LDIWRdK 7; DLDREGS:%vreg7
	%R23R22<def> = COPY %vreg6; DLDREGS:%vreg6
	%R21R20<def> = COPY %vreg7; DLDREGS:%vreg7
	%R31R30<def> = COPY %vreg4; DREGS:%vreg4
	ICALL %R31R30, %R25R24<undef>, %R23R22, %R21R20, <regmask %R2 %R3 %R4 %R5 %R6 %R7 %R8 %R9 %R10 %R11 %R12 %R13 %R14 %R15 %R16 %R17 %R28 %R29 %R3R2 %R5R4 %R7R6 %R9R8 %R11R10 %R13R12 %R15R14 %R17R16 %R29R28>, %SP<imp-use>, %R31R30<imp-use>, %SP<imp-def>, %R24<imp-def,dead>
	ADJCALLSTACKUP 0, 0, %SP<imp-def,dead>, %SREG<imp-def,dead>, %SP<imp-use>

# End machine code for function EscapeDefaultStateDebug.
```

dylanmckay, 2017-04-28:
This is almost certainly related to avr-llvm/llvm#1. The bug was really hard to fix (I had to figure out how the greedy register allocator worked and make a change to it so that it would attempt to split unspillable live ranges.

I'm surprised to see this again. Hopefully it's easier to solve this time lol

shepmaster, 2017-04-28:
One thing I notice but don't know if it's important: When `selectOrSplit PTRDISPREGS` is called at the end of the debug log, it doesn't pick R29R28, only R31R30. That seems like it would artifically increase the contention of registers.

dylanmckay, 2017-04-28:
That _may_ be because `R29R28` is the frame pointer register, which is enabled for functions in a few cases including if there was a stack spill.

shepmaster, 2017-04-29:
One thing I haven't been able to figure out yet: *why* would something be unspillable? With 32 registers, it seems like it should be conceptually pretty hard to actually run out...

dylanmckay, 2017-04-30:
There may be 32 general purpose registers, but all of the IO instructions
only support using pointers from the X, Y, and Z registers.

If a function has at least one stack spill, the Y register becomes the
frame pointer, and that means we can only have two pointers in registers at
a time in some cases.

Pointer-heavy code will quite quickly require a lot of spilling and loading
of registers.

shepmaster, 2017-04-30:
> only support using pointers from the X, Y, and Z registers.

> only have two pointers in registers at a time in some cases.

But in this case, it looks like we are using [`PTRDISPREGS`](https://github.com/avr-rust/llvm/blob/45b94b350229ec89ddf5d8352b4e18d2a2b35f48/lib/Target/AVR/AVRRegisterInfo.td#L195-L198) which only has *2* registers, `Y` and `Z`. If `Y` is taken up, then we only have `Z`, which fits with the output. Is using this register class, instead of [`PTRREGS `](https://github.com/avr-rust/llvm/blob/45b94b350229ec89ddf5d8352b4e18d2a2b35f48/lib/Target/AVR/AVRRegisterInfo.td#L186-L191) suspicious at all?

dylanmckay, 2017-04-30:
> Is using this register class, instead of PTRREGS suspicious at all?

Nah, it looks like it's `PTRDISPREGS` because the instruction is `lddw`, which (I can't remember why) but it only supports a small set of registers, named `PTRDISPREGS` because they can only be used in thie **disp**lacement instructions.

shepmaster, 2017-05-01:
I've been trying to comment out parts of libcore to get to the next error, but it feels like this is occurring for every implementation of `core::fmt::Debug`, of which there are a lot... probably means it's an important issue.

dylanmckay, 2017-05-01:
I've got an unfinished patch locally for #38 which may reduce the number of these (reserve the frame pointer in less cases).

shepmaster, 2017-05-01:
The `alloca` definitely plays a part - the resulting value is never used when it fails, but removing that line cause the compilation to succeed. Some quick searching seems to indicate that a frame pointer is often used when `alloca` is.

shepmaster, 2017-05-01:
I'm continuing to try to drive down the example size. This is what I have now:

```llvm
source_filename = "bugpoint-output-87a64d6.bc"
target datalayout = "e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8"
target triple = "avr"

%"fmt::Formatter" = type { i32, { i8*, void (i8*)** } }

@str.1b = external constant [0 x i8]

define void @"TryFromIntError::Debug"(%"fmt::Formatter"* dereferenceable(32)) unnamed_addr #0 personality i32 (...)* @rust_eh_personality {
start:
  %builder = alloca i8, align 8
  %1 = getelementptr inbounds %"fmt::Formatter", %"fmt::Formatter"* %0, i16 0, i32 1
  %2 = bitcast { i8*, void (i8*)** }* %1 to {}**
  %3 = load {}*, {}** %2, align 2
  %4 = getelementptr inbounds %"fmt::Formatter", %"fmt::Formatter"* %0, i16 0, i32 1, i32 1
  %5 = load void (i8*)**, void (i8*)*** %4, align 2
  %6 = getelementptr inbounds void (i8*)*, void (i8*)** %5, i16 3
  %7 = bitcast void (i8*)** %6 to i8 ({}*, i8*, i16)**
  %8 = load i8 ({}*, i8*, i16)*, i8 ({}*, i8*, i16)** %7, align 2
  %9 = tail call i8 %8({}* nonnull %3, i8* noalias nonnull readonly getelementptr inbounds ([0 x i8], [0 x i8]* @str.1b, i16 0, i16 0), i16 15)
  unreachable
}

declare i32 @rust_eh_personality(...) unnamed_addr

attributes #0 = { uwtable }
```

Things I've noticed:

- Removing `alloca` - compiles. My guess is that `alloca` is triggering the frame pointer, reducing available registers.
- Removing the first member of `Formatter` - compiles. My guess is that this allows us to use the non-displacement instructions.

shepmaster, 2017-05-01:
Continuing to minimize:

```llvm
source_filename = "bugpoint-output-87a64d6.bc"
target datalayout = "e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8"
target triple = "avr"

%"fmt::Formatter" = type { i32, {}*, i8 ({}*)** }

define void @"TryFromIntError::Debug"(%"fmt::Formatter"*) {
start:
  %builder = alloca i8
  %1 = getelementptr %"fmt::Formatter", %"fmt::Formatter"* %0, i16 0, i32 1
  %2 = load {}*, {}** %1
  %3 = getelementptr %"fmt::Formatter", %"fmt::Formatter"* %0, i16 0, i32 2
  %4 = load i8 ({}*)**, i8 ({}*)*** %3
  %5 = load i8 ({}*)*, i8 ({}*)** %4
  %6 = tail call i8 %5({}* %2)
  unreachable
}
```

shepmaster, 2017-05-02:
@dylanmckay yo, why does [`LDWRdPtr` use `PTRDISPREGS`](https://github.com/avr-rust/llvm/blob/a9c81036357c6ef45a853a3b489e5f291244e153/lib/Target/AVR/AVRInstrInfo.td#L1154-L1161)? Shouldn't the `ld` instruction work with `PTRREGS`?

Changing that locally allows this example code to compile (but I'm not sure it's correct as I get a rogue `cli` in the output)

shepmaster, 2017-05-02:
Also, I'm pretty sure that even with one register, the example LLVM IR should be able to be compiled to machine code. I feel like something's not quite right with spilling / reordering algorithm. I was trying to write out a proposed register allocation when I stumbled on the `LDWRdPtr` comment.

dylanmckay, 2017-05-02:
> @dylanmckay yo, why does LDWRdPtr use PTRDISPREGS? Shouldn't the ld instruction work with PTRREGS?

I'll have a look in a few hours and get back to you.

> Also, I'm pretty sure that even with one register, the example LLVM IR should be able to be compiled to machine code. I feel like something's not quite right with spilling / reordering algorithm. I was trying to write out a proposed register allocation when I stumbled on the LDWRdPtr comment.

Yeah, it is a bug in LLVM. I'll get more details in a bit but from memory, when LLVM runs out of registers it isn't smart enough to perform a cross-class copy (i.e. copy a `PTRREGS` to a `DREGS`) temporarily. I don't think we would have this issue if `PTRREGS` == `DREGS`. I believe the original AVR-LLVM developers put something on bugzilla, I'll dig it up.

shepmaster, 2017-05-03:
> Changing that locally allows this example code to compile

Unfortunately, the real code still fails; guess that that one extra register wasn't enough.

shepmaster, 2017-05-03:
> Yeah, it is a bug in LLVM. I'll get more details in a bit but from memory, when LLVM runs out of registers it isn't smart enough to perform a cross-class copy

How scary would implementing something like that be? It feels like changing the register allocator is A Big Deal. Beyond scariness, how complicated would you guess?

jackpot51, 2017-05-03:
Guys, I have been following changes. I have removed `bignum`, and `i128/u128` and am now on this error. If you need me to hack out `Debug` implementations I can do that too - then I will post my changes on a fork

dylanmckay, 2017-05-03:
[Here's](https://reviews.llvm.org/D25274) the patch where we decided to use the generic relaxation pass

[Here's](https://bugs.llvm.org/show_bug.cgi?id=14879) the bug which discusses the register allocator problem.

dylanmckay, 2017-05-03:
> How scary would implementing something like that be? It feels like changing the register allocator is A Big Deal. Beyond scariness, how complicated would you guess?

It's not _too_ scary - there are a lot of test cases for the register allocator and so it is hard to break. I've [worked on it before](https://reviews.llvm.org/D25070), but funnily enough I also caused a small regression that was discovered a few weeks later.

> how complicated would you guess?

It depends on how much support there is for it. I'm sure there's some sort of mechanism for finding free registers in a register class and performing temporary copies to it, it shouldn't be too hard to make it work across class boundaries *famous last words*

shepmaster, 2017-05-03:
> to use the generic relaxation pass.

@dylanmckay did you want to put this on #44 ?

> the register allocator problem.

That does seem like the same root cause; too bad the earlier fix didn't solve it. 😿

shepmaster, 2017-05-03:
@jackpot51 there's no easy way to know which Rust constructs will trigger a given LLVM issue. Essentially, my process has been:

1. attempt to build libcore
1. get a failure
1. [reduce a testcase](https://github.com/avr-rust/rust/wiki)
1. report that testcase here
1. attempt to comment out the offending code
1. repeat

You are welcome to try to comment out all the `Debug` implementations in libcore and see if that gets past this issue. It's possible that other constructs generate the same style of code and will be subject to the same problem.

dylanmckay, 2017-05-03:
It looks like the register allocator, upon seeing a live variable, deduces all of the allocatable registers  (by looking at the required register class) and attempts to select one (`RegAllocGreedy.cpp:2525`).

If it can not find one, it attempts to evict a register, but it is given the exact same list of allocatable registers as before, only containing the registers from the original register class (`RegAllocGreedy.cpp:2551`).

I think it'll be doable to teach it to do this.

riaqn, 2017-12-29:
Hello, I wonder how it's going? It would be a killer app for rust to be able to run on AVR without a glitch.

brainlag, 2018-03-20:
As far as I understand the problem occurs when `LDDWRdPtrQ` uses the `ptrdispregs` register class as target register like below in line 48B. This should work, but the allocator can't deal with this for some reason. So from my testing, it seams like (and I might be totally wrong on this) the allocator reserves the Z register for the ICALL instruction and then the register class `ptrdispregs` only has 1 register left and we can't use Y for source and destination. Removing the Z register from DREGS fixes the problem but removing Y register does not. Coincidence? Maybe or ICALL does reserve the Z register.

```
40B	  %11:ptrdispregs = COPY %10:dregs
48B	  early-clobber %4:ptrdispregs = LDDWRdPtrQ %11:ptrdispregs, 6 :: (dereferenceable load 2 from %ir.4)
80B	  early-clobber %3:dregs = LDDWRdPtrQ %4:ptrdispregs, 6 :: (load 2 from %ir.7)
88B	  %12:ptrdispregs = COPY %10:dregs
96B	  early-clobber %5:dregs = LDDWRdPtrQ %12:ptrdispregs, 4 :: (dereferenceable load 2 from %ir.2)
```

Sure we could fix up the register allocator but we don't have the resources. We should just disallow machine instructions like 48B and live with the extra copy instruction here and there.

Comments?

dylanmckay, 2018-03-20:
>  the allocator reserves the Z register for the ICALL instruction

[Correct](https://github.com/llvm-mirror/llvm/blob/master/lib/Target/AVR/AVRInstrInfo.td#L798)

> and then the register class ptrdispregs only has 1 register left and we can't use Y for source and destination. Removing the Z register from DREGS fixes the problem but removing Y register does not. Coincidence?

Explained very well - I will edit the description to include this

Perhaps the `Y` register is being used as a frame pointer? All functions which have spills or allocas will always reserve `Y` to store the frame pointer ([code](https://github.com/llvm-mirror/llvm/blob/master/lib/Target/AVR/AVRFrameLowering.cpp#L231-L232)).

> We should just disallow machine instructions like 48B and live with the extra copy instruction here and there.

That sounds like a reasonable tradeoff to me.

I'm not sure what you mean by "like 48B" - what properties distinguish 48B from other wide load instructions?

I wonder how the `LDDWRdPtrQ` instruction was inserted in this case.

brainlag, 2018-03-20:
> I'm not sure what you mean by "like 48B" - what properties distinguish 48B from other wide load instructions?

The target is `ptrdispregs` register class while at 80B and 96B it is `dregs`.

brainlag, 2018-03-20:
Here my workaround for this bug and #95:

https://gist.github.com/brainlag/6f9ce4676de274fbb6c4c59a5612a429

leo60228, 2018-07-16:
Is there a reason the above couldn't be merged?

dylanmckay, 2018-07-19:
I've written a bug report [here](https://bugs.llvm.org/show_bug.cgi?id=38224). I should get around to merging this tomorrow.

leo60228, 2018-07-19:
Should I convert this to a PR? What branch should I do it on? I've only compiled master, but I don't have time to test anyway (compiling is so slow...).

TimNN, 2018-10-15:
Friendly ping :) @brainlag's patch something that could be submitted upstream or something to be kept around in `avr-rust`?

Note that after applying this patch to fix one assertion (and already having some others applied for different issues) I'm getting the following assertion when compiling stock-`libcore`:

```plain
rustc: /home/logic/avr/src/src/llvm/lib/CodeGen/MachineInstr.cpp:233: void llvm::MachineInstr::addOperand(llvm::MachineFunction&, const llvm::MachineOperand&): Assertion `(isImpReg || Op.isRegMask() || MCID->isVariadic() || OpNo < MCID->getNumOperands() || isMetaDataOp) && "Trying to add an operand to a machine instr that is already done!"' failed.
```

Could that be related to the proposed patch?

dylanmckay, 2018-11-05:
Upstreamed in [r346114](https://github.com/llvm-mirror/llvm/commit/9de46729ce2fb4bf4b2a19b7b387898f256d827f).

Great work @brainlag, I didn't even need to add tests.

Also raised [LLVM bug PR39553](https://bugs.llvm.org/show_bug.cgi?id=39553) to track the workaround and a proper fix.

closed 2018-11-05 by dylanmckay

dylanmckay, 2019-01-16:
The root cause has been fixed in upstream LLVM.

Raise #126 to track removal of the hack.