avr-rust/rust-legacy-fork #41: Other AVR changes to merge into the Rust LLVM fork

shepmaster, 2017-05-01:
This is the list of new commits we have in avr-rust/llvm compared to rust-lang/llvm now:

```
cbca85fc0e6ec823d483303f19eaaa4617a2e652 [AVR] Fix a bug so that we now emit R_AVR_16 fixups with the correct offset
f2a53d44ce8f2cdb1daa7a61e9e13f4866a5d0b0 [AVR] Implement stacksave/stackrestore by expanding (PR31342)
f2dd609afb61e46fa310feb24b920e701c950007 [AVR] Support zero-sized arguments in defined methods
103e56b3cd49479ab0048a9f5d10c98d48f77ebd [LiveRangeEdit] Don't mess up with LiveInterval when a new vreg is created.
45b94b350229ec89ddf5d8352b4e18d2a2b35f48 [AVR] Do not kill the dest register for a pseudo instruction
243e3dba20f0918adc78e9d5879948dfbf69a310 [AVR] Support the LDWRdPtr instruction with the same Src+Dst register
```

3 of those correspond to issues filed here; the other 3 "looked interesting" to me so I cherry-picked them - should we actually have those 3? Should the be upstreamed to Rust?

dylanmckay, 2017-05-01:
Those other patches sound are new enough (from memory) to not be in LLVM 4.0.

Good idea to cherry-pick them, they don't hurt to have and will let us compile a bigger variety of programs.

> Should they be upstreamed to Rust?

I'm not bothered just yet, there's going to be some time until we can upstream Rust support anyway so I'm not in a big rush.

On top of that, if we wait until this fork is more developed, we can upstream all patches at once.

shepmaster, 2017-05-01:
> On top of that, if we wait until this fork is more developed, we can upstream all patches at once.

Yeah, I figured that would be the courteous thing to do. I was mostly just wondering if they should *ever* be upstreamed, and the answer to that sounds like "yes"

dylanmckay, 2017-05-02:
Do we still want this open @shepmaster?

shepmaster, 2017-05-02:
I'll add the tag and close it to help us track changes, but can be closed.

closed 2017-05-02 by shepmaster