avr-rust/rust-legacy-fork #43: LLVM ERROR: Cannot select: t55: i8 = rotl Constant:i8<-2>, t32

gergoerdi, 2017-05-01:
For this one, I don't have a nice small test case :( Full LLVM IR is at https://gist.github.com/gergoerdi/9f41a400d7611f7577cc83decf2dbbd4
The error message is:
```
LLVM ERROR: Cannot select: t44: i8 = rotl Constant:i8<-2>, t17
  t43: i8 = Constant<-2>
  t17: i8 = and t11, Constant:i8<7>
    t11: i8 = add t8, t10
      t8: i8,ch = CopyFromReg t0, Register:i8 %vreg6
        t7: i8 = Register %vreg6
      t10: i8,ch = CopyFromReg t0, Register:i8 %vreg23
        t9: i8 = Register %vreg23
    t16: i8 = Constant<7>
In function: main
```

dylanmckay, 2017-05-01:
It looks like we've got support for constant bit-rotations but rotation by anything that is not a constant is unsupported.

First off the bat, we need to add this to `AVRISelLowering`

dylanmckay, 2017-05-01:
I've implemented n-bit rotations locally, will push up a patch soon.

dylanmckay, 2017-05-01:
Implemented in `r301794`

dylanmckay, 2017-05-01:
Backported in e7e0178c9acb5c542d7bdaf5505a16b9decbe057.

If you pull master, this should now be fixed.

closed 2017-05-01 by dylanmckay