avr-rust/rust-legacy-fork #49: Final continuation jump missing from MBB

gergoerdi, 2017-05-10:
Here's the minimized LLVM IR:

```
target triple = "avr-atmel-none"

define internal fastcc void @loopy() unnamed_addr{
start:
  br label %bb7.preheader

bb7.preheader:                                    ; preds = %bb10, %start
  %i = phi i8 [ 0, %start ], [ %j, %bb10 ]
  %j = phi i8 [ 1, %start ], [ %next, %bb10 ]
  br label %bb10

bb4:                                              ; preds = %bb10
  ret void

bb10:                                             ; preds = %bb7.preheader
  tail call fastcc void @observe(i8 %i, i8 1)
  %0 = icmp ult i8 %j, 20
  %1 = zext i1 %0 to i8
  %next = add i8 %j, %1
  br i1 %0, label %bb7.preheader, label %bb4

}

declare void @observe(i8, i8) unnamed_addr;
```

And here's the generated assembly:

```
loopy:                                  ; @loopy
; BB#0:                                 ; %start
        push    r16
        push    r17
        ldi     r17, 1
        ldi     r24, 0
LBB0_1:                                 ; %bb4
        pop     r17
        pop     r16
        ret
LBB0_2:                                 ; %bb10
        ldi     r17, 0
LBB0_3:                                 ; %bb10
        add     r17, r16
        cpi     r16, 20
        mov     r24, r16
        brlo    .+2
        rjmp    LBB0_1
; BB#4:                                 ; %bb7.preheader
        mov     r16, r17
        ldi     r17, 1
        ldi     r22, 1
        call    observe
        cpi     r16, 20
        brlo    .+2
        rjmp    LBB0_2
        rjmp    LBB0_3
```

The problem is that `BB#0` doesn't jump to its continuation, and so falls through to `LBB0_1` which should only happen at the end of the loop.

gergoerdi, 2017-05-10:
This is actually similar to https://github.com/avr-rust/rust/issues/46. The selection DAG for `loopy:start` with `-O1` doesn't have the final jump:

```
Initial selection DAG: BB#0 'loopy:start'
SelectionDAG has 8 nodes:
  t0: ch = EntryToken
    t3: ch = CopyToReg t0, Register:i8 %vreg3, Constant:i8<0>
    t6: ch = CopyToReg t0, Register:i8 %vreg4, Constant:i8<1>
  t7: ch = TokenFactor t3, t6
```

but the final jump is present with `-O0`:

```
Initial selection DAG: BB#0 'loopy:start'
SelectionDAG has 10 nodes:
  t0: ch = EntryToken
      t3: ch = CopyToReg t0, Register:i8 %vreg3, Constant:i8<0>
      t6: ch = CopyToReg t0, Register:i8 %vreg4, Constant:i8<1>
    t8: ch = TokenFactor t3, t6
  t9: ch = br t8, BasicBlock:ch<bb7.preheader 0x2f1f9a8>
```

Could it be that this is all fine & dandy, the fall-through is intentional (and so the fix from https://github.com/avr-rust/rust/issues/46 is not needed), and the real problem is that `EmitInstrWithCustomInserter` screws up the MBB ordering?

gergoerdi, 2017-05-10:
I think the problem is indeed that `EmitInstrWithCustomInserter` inserts `trueMBB` and `falseMBB` to `MF`'s start instead of maintaining `MBB`'s position.

I'll experiment with replacing

```
 MachineFunction::iterator I = MBB->getParent()->begin();
 ++I;
 MF->insert(I, trueMBB);
 MF->insert(I, falseMBB);
```

with

```
  MachineFunction::iterator I;
  for (I = MF->begin(); I != MF->end() && &(*I) != MBB; ++I);
  MF->insert(I, trueMBB);
  MF->insert(I, falseMBB);
```

gergoerdi, 2017-05-10:
The change described in https://github.com/avr-rust/rust/issues/49#issuecomment-300380048 fixes the original issue, but breaks other code spectacularly; in particular, given the following LLVM IR:

```
source_filename = "libcore_mini.cgu-0.rs"
target datalayout = "e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8"
target triple = "avr-atmel-none"

%"option::Option<u32>" = type { i32, [0 x i32], [1 x i32] }

define void @pow(%"option::Option<u32>"* noalias nocapture sret dereferenceable(8), i32) unnamed_addr {
start:
  %v3 = tail call i32 @llvm.ctlz.i32(i32 %1, i1 false)
  %v4 = trunc i32 %v3 to i8
  %v7 = shl i8 1, %v4
  %v8 = zext i8 %v7 to i32
  %v9 = getelementptr inbounds %"option::Option<u32>", %"option::Option<u32>"* %0, i16 0, i32 0
  store i32 1, i32* %v9, align 4
  %v10 = getelementptr inbounds %"option::Option<u32>", %"option::Option<u32>"* %0, i16 0, i32 2, i16 0
  store i32 %v8, i32* %v10, align 4
  ret void
}

declare i32 @llvm.ctlz.i32(i32, i1)
```

compiling it to AVR crashes with

```
#0 0x00000000017210d5 llvm::sys::PrintStackTrace(llvm::raw_ostream&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x17210d5)
#1 0x000000000171f0ee llvm::sys::RunSignalHandlers() (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x171f0ee)
#2 0x000000000171f252 SignalHandler(int) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x171f252)
#3 0x00007f25e76e3330 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x10330)
#4 0x00000000010a2b55 llvm::LiveVariables::runOnBlock(llvm::MachineBasicBlock*, unsigned int) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10a2b55)
#5 0x00000000010a357d llvm::LiveVariables::runOnMachineFunction(llvm::MachineFunction&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10a357d)
#6 0x00000000010e8573 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10e8573)
#7 0x000000000138ba53 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138ba53)
#8 0x000000000138bafc llvm::FPPassManager::runOnModule(llvm::Module&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138bafc)
#9 0x000000000138c85f llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138c85f)
#10 0x0000000000697f74 compileModule(char**, llvm::LLVMContext&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x697f74)
#11 0x0000000000646a90 main (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x646a90)
#12 0x00007f25e688af45 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21f45)
#13 0x000000000068d2b5 _start (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x68d2b5)
Stack dump:
0.	Program arguments: /home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc --mcpu=atmega328p a.ll --debug 
1.	Running pass 'Function Pass Manager' on module 'a.ll'.
2.	Running pass 'Live Variable Analysis' on function '@pow'
```

gergoerdi, 2017-05-10:
```
# Machine code for function pow: IsSSA, TracksLiveness
Function Live Ins: %R25R24 in %vreg0, %R21R20 in %vreg1, %R23R22 in %vreg2

0B	BB#1: derived from LLVM BB %start
	    Predecessors according to CFG: BB#0 BB#2
16B		%vreg131<def> = PHI %vreg69, <BB#0>, %vreg130, <BB#2>; DREGS:%vreg131,%vreg69 IWREGS:%vreg130
32B		%vreg132<def> = COPY %vreg131:sub_lo; GPR8:%vreg132 DREGS:%vreg131
48B		CPRdRr %vreg132, %R0, %SREG<imp-def>; GPR8:%vreg132
64B		BREQk <BB#4>, %SREG<imp-use>
	    Successors according to CFG: BB#3(?%) BB#4(?%)

80B	BB#3: derived from LLVM BB %start
	    Predecessors according to CFG: BB#1 BB#3
96B		%vreg138<def> = PHI %vreg44, <BB#1>, %vreg139, <BB#3>; GPR8:%vreg138,%vreg139 LD8:%vreg44
112B		%vreg136<def> = PHI %vreg132, <BB#1>, %vreg137, <BB#3>; LD8:%vreg136,%vreg137 GPR8:%vreg132
128B		%vreg139<def,tied1> = LSLRd %vreg138<tied0>, %SREG<imp-def>; GPR8:%vreg139,%vreg138
144B		%vreg137<def,tied1> = SUBIRdK %vreg136<tied0>, 1, %SREG<imp-def>; LD8:%vreg137,%vreg136
160B		BRNEk <BB#3>, %SREG<imp-use>
	    Successors according to CFG: BB#4(?%) BB#3(?%)

176B	BB#4: derived from LLVM BB %start
	    Predecessors according to CFG: BB#1 BB#3
192B		%vreg133<def> = PHI %vreg44, <BB#1>, %vreg139, <BB#3>; GPR8:%vreg133,%vreg139 LD8:%vreg44
208B		%vreg134<def> = ZEXT %vreg133<kill>, %SREG<imp-def,dead>; DREGS:%vreg134 GPR8:%vreg133
224B		%vreg135<def> = COPY %vreg0; PTRDISPREGS:%vreg135 DREGS:%vreg0
240B		STDWPtrQRr %vreg135, 4, %vreg134<kill>; mem:ST2[%v10](align=4) PTRDISPREGS:%vreg135 DREGS:%vreg134
256B		RET

272B	BB#2: derived from LLVM BB %start
	    Predecessors according to CFG: BB#0
288B		RJMPk <BB#1>
	    Successors according to CFG: BB#1(?%)

304B	BB#0: derived from LLVM BB %start
	    Live Ins: %R25R24 %R21R20 %R23R22
320B		%vreg2<def> = COPY %R23R22; DREGS:%vreg2
336B		%vreg1<def> = COPY %R21R20; DREGS:%vreg1
352B		%vreg0<def> = COPY %R25R24; DREGS:%vreg0
368B		%vreg3<def> = LDIWRdK 1; DLDREGS:%vreg3
384B		%vreg5<def> = COPY %vreg0; PTRREGS:%vreg5 DREGS:%vreg0
400B		%vreg4<earlyclobber,def,tied1> = STWPtrPiRr %vreg5<tied0>, %vreg3<kill>, 2; mem:ST2[%v91](align=4) PTRREGS:%vreg4,%vreg5 DLDREGS:%vreg3
416B		%vreg6<def> = LDIWRdK 0; DLDREGS:%vreg6
432B		%vreg7<def> = COPY %vreg4; PTRDISPREGS:%vreg7 PTRREGS:%vreg4
448B		STWPtrRr %vreg7<kill>, %vreg6; mem:ST2[%v91+2] PTRDISPREGS:%vreg7 DLDREGS:%vreg6
464B		%vreg8<def> = COPY %vreg0; PTRDISPREGS:%vreg8 DREGS:%vreg0
480B		STDWPtrQRr %vreg8, 6, %vreg6; mem:ST2[%v10+2] PTRDISPREGS:%vreg8 DLDREGS:%vreg6
496B		%vreg9<def,tied1> = LSRWRd %vreg2<tied0>, %SREG<imp-def,dead>; DREGS:%vreg9,%vreg2
512B		%vreg10<def,tied1> = ORWRdRr %vreg2<tied0>, %vreg9<kill>, %SREG<imp-def,dead>; DREGS:%vreg10,%vreg2,%vreg9
528B		%vreg11<def,tied1> = LSRWRd %vreg10<tied0>, %SREG<imp-def,dead>; DREGS:%vreg11,%vreg10
544B		%vreg12<def,tied1> = LSRWRd %vreg11<tied0>, %SREG<imp-def,dead>; DREGS:%vreg12,%vreg11
560B		%vreg13<def,tied1> = ORWRdRr %vreg10<tied0>, %vreg12<kill>, %SREG<imp-def,dead>; DREGS:%vreg13,%vreg10,%vreg12
576B		%vreg14<def,tied1> = LSRWRd %vreg13<tied0>, %SREG<imp-def,dead>; DREGS:%vreg14,%vreg13
592B		%vreg15<def,tied1> = LSRWRd %vreg14<tied0>, %SREG<imp-def,dead>; DREGS:%vreg15,%vreg14
608B		%vreg16<def,tied1> = LSRWRd %vreg15<tied0>, %SREG<imp-def,dead>; DREGS:%vreg16,%vreg15
624B		%vreg17<def,tied1> = LSRWRd %vreg16<tied0>, %SREG<imp-def,dead>; DREGS:%vreg17,%vreg16
640B		%vreg18<def,tied1> = ORWRdRr %vreg13<tied0>, %vreg17<kill>, %SREG<imp-def,dead>; DREGS:%vreg18,%vreg13,%vreg17
656B		%vreg19<def,tied1> = LSRWRd %vreg18<tied0>, %SREG<imp-def,dead>; DREGS:%vreg19,%vreg18
672B		%vreg20<def,tied1> = LSRWRd %vreg19<tied0>, %SREG<imp-def,dead>; DREGS:%vreg20,%vreg19
688B		%vreg21<def,tied1> = LSRWRd %vreg20<tied0>, %SREG<imp-def,dead>; DREGS:%vreg21,%vreg20
704B		%vreg22<def,tied1> = LSRWRd %vreg21<tied0>, %SREG<imp-def,dead>; DREGS:%vreg22,%vreg21
720B		%vreg23<def,tied1> = LSRWRd %vreg22<tied0>, %SREG<imp-def,dead>; DREGS:%vreg23,%vreg22
736B		%vreg24<def,tied1> = LSRWRd %vreg23<tied0>, %SREG<imp-def,dead>; DREGS:%vreg24,%vreg23
752B		%vreg25<def,tied1> = LSRWRd %vreg24<tied0>, %SREG<imp-def,dead>; DREGS:%vreg25,%vreg24
768B		%vreg26<def,tied1> = LSRWRd %vreg25<tied0>, %SREG<imp-def,dead>; DREGS:%vreg26,%vreg25
784B		%vreg27<def,tied1> = ORWRdRr %vreg18<tied0>, %vreg26<kill>, %SREG<imp-def,dead>; DREGS:%vreg27,%vreg18,%vreg26
800B		%vreg28<def,tied1> = COMWRd %vreg27<tied0>, %SREG<imp-def,dead>; DREGS:%vreg28,%vreg27
816B		%vreg29<def,tied1> = LSRWRd %vreg28<tied0>, %SREG<imp-def,dead>; DLDREGS:%vreg29 DREGS:%vreg28
832B		%vreg30<def,tied1> = ANDIWRdK %vreg29<tied0>, 21845, %SREG<imp-def,dead>; DLDREGS:%vreg30,%vreg29
848B		%vreg31<def,tied1> = SUBWRdRr %vreg28<tied0>, %vreg30<kill>, %SREG<imp-def,dead>; DLDREGS:%vreg31,%vreg30 DREGS:%vreg28
[... 87 lines cut ...]
2256B		%vreg113<def,tied1> = LSLWRd %vreg112<tied0>, %SREG<imp-def,dead>; DREGS:%vreg113,%vreg112
2272B		%vreg114<def,tied1> = LSLWRd %vreg113<tied0>, %SREG<imp-def,dead>; DREGS:%vreg114,%vreg113
2288B		%vreg115<def,tied1> = LSLWRd %vreg114<tied0>, %SREG<imp-def,dead>; DREGS:%vreg115,%vreg114
2304B		%vreg116<def,tied1> = LSLWRd %vreg115<tied0>, %SREG<imp-def,dead>; DREGS:%vreg116,%vreg115
2320B		%vreg117<def,tied1> = LSLWRd %vreg116<tied0>, %SREG<imp-def,dead>; DREGS:%vreg117,%vreg116
2336B		%vreg118<def,tied1> = LSLWRd %vreg117<tied0>, %SREG<imp-def,dead>; DREGS:%vreg118,%vreg117
2352B		%vreg119<def,tied1> = LSLWRd %vreg118<tied0>, %SREG<imp-def,dead>; DREGS:%vreg119,%vreg118
[... cut ...]
2496B		%vreg128<def,tied1> = LSRWRd %vreg127<tied0>, %SREG<imp-def,dead>; DREGS:%vreg128,%vreg127
2512B		%vreg129<def,tied1> = LSRWRd %vreg128<tied0>, %SREG<imp-def,dead>; IWREGS:%vreg129 DREGS:%vreg128
2528B		%vreg130<def,tied1> = ADIWRdK %vreg129<tied0>, 16, %SREG<imp-def,dead>; IWREGS:%vreg130,%vreg129
2544B		CPWRdRr %vreg2, %vreg6, %SREG<imp-def>; DREGS:%vreg2 DLDREGS:%vreg6
2560B		RBRNEk <BB#1>, %SREG<imp-use>
2576B		RJMPk <BB#2>
	    Successors according to CFG: BB#2(?%) BB#1(?%)

# End machine code for function pow.
```

gergoerdi, 2017-05-10:
```
********** PROCESS IMPLICIT DEFS **********
********** Function: pow
BB#0 has 2 implicit defs.
Processing %vreg111<def> = IMPLICIT_DEF; DREGS:%vreg111
Processing %vreg51<def> = IMPLICIT_DEF; DREGS:%vreg51
```

gergoerdi, 2017-05-10:
That's because

```
  MachineFunction::iterator I;
  for (I = MF->begin(); I != MF->end() && &(*I) != MBB; ++I);
  MF->insert(I, trueMBB);
  MF->insert(I, falseMBB);
```

is off by one compared to the original code. So it should be

```
  MachineFunction::iterator I;
  for (I = MF->begin(); I != MF->end() && &(*I) != MBB; ++I);
  if (I != MF->end()) ++I;
  MF->insert(I, trueMBB);
  MF->insert(I, falseMBB);
```

gergoerdi, 2017-05-11:
How do I add some kind of 'has-patch' tag?

shepmaster, 2017-05-11:
> How do I add some kind of 'has-patch' tag?

Once you've got a commit that's been merged into LLVM, notify us here and we can cherry-pick it into our LLVM fork.

I'm not 100% sure what the LLVM submission process is, but I'm sure @dylanmckay would be willing to point the way if you can't find anything online.

dylanmckay, 2017-05-11:
> I'm not 100% sure what the LLVM submission process is, but I'm sure @dylanmckay would be willing to point the way if you can't find anything online.

[This][http://llvm.org/docs/DeveloperPolicy.html#making-and-submitting-a-patch) guide is really good. Prefer submitting the patch via [Phabricator](https://reviews.llvm.org/) instead of email because it's much easier to review.

Feel free to add me as a reviewer (my phab username is `dylanmckay`) and I can sign off and commit it for you.

Nice debugging!

dylanmckay, 2017-05-13:
I've added the testcase from this issue into the LLVM patch and I've also upstreamed it in `r302973`.

dylanmckay, 2017-05-13:
Cherry picked into avr-rust in 6df57710ab656e3472e8d6122397d568967d0a55.

Thanks for the work @gergoerdi!

closed 2017-05-13 by dylanmckay

jackpot51, 2017-05-13:
@gergoerdi Were you able to compile libcore with these changes?

shepmaster, 2017-05-13:
@jackpot51 you can see @gergoerdi's [blog post](https://gergo.erdi.hu/blog/2017-05-12-rust_on_avr__beyond_blinking/) and [comments on Reddit](https://www.reddit.com/r/rust/comments/6ar4q7/rust_on_avr_beyond_blinking/).

TL;DR, no. Like we've needed to do before, a [subset of libcore was created](https://github.com/gergoerdi/rust-avr-libcore-mini).

gergoerdi, 2017-05-13:
@shepmaster @dylanmckay Thanks for covering the last mile on this. I'd like to add similar test cases to my other commits, so I tried following http://llvm.org/docs/TestingGuide.html. How do I run LLVM tests from inside the rustc build tree? I've tried

```
$ pwd # As you can see, this is inside rustc's build tree
/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/build
$ ./bin/llvm-lit ../../../../../rust/src/llvm/test/CodeGen/AVR/
llvm-lit: /home/cactus/prog/rust/rust-avr/rust/src/llvm/test/lit.cfg:159: fatal: No site specific configuration available!
```

dylanmckay, 2017-05-13:
That looks to be related to the fact that LLVM is compiled inside the Rust build dir.

Normally when you compile and run LLVM, it will run a configure script which will fill a file named `lit.site.cfg` with configure settings.

I'm not sure how to work around it, but I know for certain that if you build LLVM on its own, it will work fine.

I think in order to compile it on its own, all you need to do it `cmake ~/path/to/rust/src/llvm -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=AVR -DCMAKE_BUILD_TYPE=Debug && make`

dylanmckay, 2017-05-31:
The fix has been upstreamed in `r302973`.

dylanmckay, 2017-05-31:
Fix has been cherry picked into `avr-support` branch