avr-rust/rust-legacy-fork #54: Should we include commented-out libcore in avr-rust?

dylanmckay, 2017-05-13:
I was wondering what your thoughts were @shepmaster and @gergoerdi on pushing one of the minified versions of `libcore` you both have so that it is a part of this repo?

There isn't much point in maintaining separate local `libcore` libraries.

One concern is that it may mean that we get more merge conflicts. We would probably have to comment out parts of libcore in this repository anyway for stuff like float/libcore support.

gergoerdi, 2017-05-13:
My version is here: https://github.com/gergoerdi/rust-avr-libcore-mini I'm happy to move it wherever. It is based on `libcore` as of https://github.com/avr-rust/rust/commit/e80f644aa8f3310259150cdae3e14c31d4087e6e with the following modules (and references to them) removed:

* `any.rs`
* `cell`
* `char`
* `char_private`
* `fmt::*`
* `hash::*`
* `num::{bignum, dec2flt, flt2dec, diy_float}`
* `raw`
* `slice::sort`
* `str`
* `sync::*`
* `tuple`

shepmaster, 2017-05-13:
The big trick to me is that we need the full libcore to build the compiler for the host but when targeting AVR we need the minimized libcore for now.

Commenting out wouldn't be sufficient for that, unless we actually had two directories, at which point the merge conflicts would go to zero but the updating process would get much harder ;-)

Perhaps `#[cfg(not(target_arch = "avr"))]` (or whatever is the real value) would be better than commenting out stuff. Would also be good markers for use to remove as we go along.

Beyond that, I'd be happy to see it present here.

shepmaster, 2017-05-13:
As a rough scope:

```
$ diff --unified -r libcore/ rust-avr-libcore-mini/src/ | wc
    2084    8801   68577
```

This includes changes like 

```diff
-#[derive(Copy, Clone, Eq, PartialEq, Ord, PartialOrd, Debug, Hash)]
+#[derive(Copy, Clone, Eq, PartialEq, Ord, PartialOrd)]
```

Which I'm sure was annoying to do and will be annoying to bracket in `cfg` blocks.

shepmaster, 2017-05-13:
I feel like *most* of the `Debug` and formatting differences would be taken care of by #37, which is why I'd be more excited to see that fixed instead of commenting out every `Debug` implementation.

gergoerdi, 2017-05-14:
In the meantime, couldn't we reduce the diff considerably by keeping `Debug` and `Hash` and just stubbing them out, instead of removing them? Let me try and see how it'd work.

gergoerdi, 2017-05-14:
I've got great news! I've just pushed a new version of `libcore-mini` (overwriting `master`) that is almost identical to the real `libcore`, but still works well enough that I can compile my CHIP-8 program with it:

```
18:58:26 [cactus@galaxy src]$ diff -u $CORE .|wc
    103     473    4525
```

gergoerdi, 2017-05-14:
The remaining changes compare to `libcore` are, I believe, rather benign and concentrated (no more "I had to remove `Hash` and `Debug` derivation from every. single. type.")

* Float and `i128`/`u128` support is removed
* All the panicking macros just `loop{}`
* The `sync` module is removed (I haven't investigated yet why this is needed)
* Most of the `fmt` module is stubbed out
* String <-> number conversion is removed

gergoerdi, 2017-05-14:
```
19:22:23 [cactus@galaxy src]$ diff -qr $CORE .
Only in /home/cactus/prog/rust/rust-avr/rust/src/libcore: Cargo.toml
Only in /home/cactus/prog/rust/rust-avr/rust/src/libcore: benches
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/fmt/mod.rs and ./fmt/mod.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/fmt/num.rs and ./fmt/num.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/iter/range.rs and ./iter/range.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/lib.rs and ./lib.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/macros.rs and ./macros.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/num/mod.rs and ./num/mod.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/num/wrapping.rs and ./num/wrapping.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/panicking.rs and ./panicking.rs differ
Files /home/cactus/prog/rust/rust-avr/rust/src/libcore/str/mod.rs and ./str/mod.rs differ
Only in /home/cactus/prog/rust/rust-avr/rust/src/libcore: tests
```

gergoerdi, 2017-05-17:
More details on the magnitude of the difference from upstream `libcore`:

```
 src/fmt/mod.rs      | 310 ++--------------------------------------------------
 src/fmt/num.rs      |   6 +-
 src/iter/range.rs   |   1 -
 src/lib.rs          |   9 +-
 src/macros.rs       |   9 +-
 src/num/mod.rs      | 282 +----------------------------------------------
 src/num/wrapping.rs |   2 +-
 src/panicking.rs    |  12 +-
 src/str/mod.rs      |  27 +----
 9 files changed, 26 insertions(+), 632 deletions(-)
```

closed 2017-07-04 by dylanmckay