avr-rust/rust-legacy-fork #56: "Cannot select" for str::slice_error_fail

gergoerdi, 2017-05-14:
This came up while trying to add `str` back to `libcore-mini`:

LLVM IR, minimized as far as I could:

```
target triple = "avr-atmel-none"

; Function Attrs: cold noinline noreturn nounwind uwtable
define void @_ZN4core3str16slice_error_fail17ha9abe7236f6b8737E(i8* noalias nonnull readonly, i16, i16, i16) {
bb7.i.i23:           
  %vnot..i = icmp ugt i16 %1, %2
  br i1 %vnot..i, label %"is_char_boundary17h47e8e2cd965eb5cbE.exit", label %bb24.preheader

bb24.preheader:  
  %vchar_start.0.ph = phi i16 [ %2, %bb7.i.i23 ], [ %2, %"is_char_boundary17h47e8e2cd965eb5cbE.exit" ], [ %3, %"exit.thread" ]
  %v16 = icmp eq i16 %vchar_start.0.ph, 0
  %v17 = icmp eq i16 %vchar_start.0.ph, %1
  %vor.cond.i1353 = or i1 %v16, %v17
  br i1 %vor.cond.i1353, label %bb33, label %bb33

"is_char_boundary17h47e8e2cd965eb5cbE.exit":
  %v18 = getelementptr inbounds i8, i8* %0, i16 %2
  %v19 = load i8, i8* %v18, align 1
  %v20 = icmp sgt i8 %v19, -65
  br i1 %v20, label %"exit.thread", label %bb24.preheader

"exit.thread": 
  br label %bb24.preheader
  
bb33:
  br label %bb33
}
```

Error message:

```
LLVM ERROR: Cannot select: 
t27: ch = RBRCOND t0, BasicBlock:ch<bb33.preheader 0x4255a68>, Constant:i8<3>, t25
  t26: i8 = Constant<3>
  t25: glue = CMP t8, Constant:i8<-64>
    t8: i8,ch = load<LD1[%v18]> t0, t5, undef:i16
      t5: i16 = add t2, t4
        t2: i16,ch = CopyFromReg t0, Register:i16 %vreg0
          t1: i16 = Register %vreg0
        t4: i16,ch = CopyFromReg t0, Register:i16 %vreg2
          t3: i16 = Register %vreg2
      t7: i16 = undef
    t21: i8 = Constant<-64>
In function: _ZN4core3str16slice_error_fail17ha9abe7236f6b8737E
```

dylanmckay, 2017-07-04:
I cannot reproduce this - I've got current LLVM `trunk` and am compiling with `./bin/llc -march=avr ./issue56.ll -mcpu=atmega328p`, and compilation succeeds.

gergoerdi, 2017-08-26:
This allowed me to revert one of the workarounds in `libcore-mini`!

dylanmckay, 2017-08-26:
Nice!