avr-rust/rust-legacy-fork #60: Can't link using Cargo directly (potentially PEBKAC)

gergoerdi, 2017-05-20:
I don't know yet if this Problem Exists Between Keyboard And Chair or if it's a real bug in.. something? `rustc`? `cargo`? How LLVM/AVR does linking? but anyway, I've documented my problems trying to use `cargo` for linking here: http://stackoverflow.com/q/44082307/477476

The short of it is that `cargo build` succeeds with an ELF file that contains an empty `.text` section, but the same linker command, when run manually from the shell, produces a correct ELF.

dylanmckay, 2017-05-23:
I doubt Rust will correctly link an AVR executable - I don't think I've ever tried it.

We would need to teach it to pass the correct CRT library, otherwise our program won't even have initialisation routines.

Basically, AVR linking hasn't even been looked at for AVR.

gergoerdi, 2017-05-23:
Right, but I'm explicitly telling `rustc` to use `avr-gcc` as the linker, and it gets to calling `avr-gcc` correctly, yet the correct invocation still results in an empty ELF file.

gergoerdi, 2017-06-08:
I knew it was going to be a PEBKAC.

The problem was that I had this in my target `.json` file:

```
  "pre-link-args": {
    "gcc": ["-Os -mmcu=atmega328p"]
  },
```
instead of
 
```
  "pre-link-args": {
    "gcc": ["-Os","-mmcu=atmega328p"]
  },
```

gergoerdi, 2017-06-08:
This is cool because now I can build using Xargo without polluting my `Cargo.toml` files with explicit `libcore-mini` references.

closed 2017-06-08 by gergoerdi

dylanmckay, 2017-06-09:
@gergoerdi: What did you need to do to set up Xargo for AVR? I would've assumed it would be quite tricky.