avr-rust/rust-legacy-fork #61: Assertion `!NodePtr->isKnownSentinel()' failed.

gergoerdi, 2017-05-20:
Reduced LLMV IR:

```
target triple = "avr-atmel-none"

%str_slice = type { i8*, i16 }
%Machine = type { i16, [0 x i8], i16, [0 x i8], [16 x i8], [0 x i8] }

define internal fastcc void @step(%Machine* dereferenceable(20)) unnamed_addr #0 {
 ret void
}

define void @main() unnamed_addr #3 {
start:
  %machine = alloca %Machine, align 8
  %v0 = bitcast %Machine* %machine to i8*
  %v1 = getelementptr inbounds %Machine, %Machine* %machine, i16 0, i32 2
  %v2 = load i16, i16* %v1, align 2
  br label %bb2.i5

bb2.i5: 
  %v18 = load volatile i8, i8* inttoptr (i16 77 to i8*), align 1
  %v19 = icmp sgt i8 %v18, -1
  br i1 %v19, label %bb2.i5, label %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit6

_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit6: 
  %v20 = load volatile i8, i8* inttoptr (i16 78 to i8*), align 2
  br label %bb7

bb7:                                              
  call fastcc void @step(%Machine* nonnull dereferenceable(20) %machine) #7
  br label %bb7
}
```

dylanmckay, 2017-07-04:
Here is the backtrace I get via gdb

```
#0  0x00007fffe3b09670 in raise () from /usr/lib/libc.so.6
#1  0x00007fffe3b0ad00 in abort () from /usr/lib/libc.so.6
#2  0x00007fffe3b0245a in __assert_fail_base () from /usr/lib/libc.so.6
#3  0x00007fffe3b024d2 in __assert_fail () from /usr/lib/libc.so.6
#4  0x00007fffe9f53bbb in llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::MachineInstr, true, true, void>, false, false>::operator* (this=0x7fffffffc870)
    at /home/dylan/projects/llvm-project/llvm/include/llvm/ADT/ilist_iterator.h:140
#5  0x00007fffe9f54255 in llvm::MachineInstrBundleIterator<llvm::MachineInstr, false>::operator* (this=0x7fffffffc870)
    at /home/dylan/projects/llvm-project/llvm/include/llvm/CodeGen/MachineInstrBundleIterator.h:179
#6  0x00007fffe9f8cf4a in llvm::AVRRegisterInfo::eliminateFrameIndex (this=0x6d7348, II=..., SPAdj=0, FIOperandNum=1, RS=0x0)
    at /home/dylan/projects/llvm-project/llvm/lib/Target/AVR/AVRRegisterInfo.cpp:169
#7  0x00007fffe7f25036 in (anonymous namespace)::PEI::replaceFrameIndices (this=0x6fb5b0, BB=0x7236c8, Fn=..., SPAdj=@0x7fffffffcb6c: 0)
    at /home/dylan/projects/llvm-project/llvm/lib/CodeGen/PrologEpilogInserter.cpp:1123
#8  0x00007fffe7f22996 in (anonymous namespace)::PEI::replaceFrameIndices (this=0x6fb5b0, Fn=...)
    at /home/dylan/projects/llvm-project/llvm/lib/CodeGen/PrologEpilogInserter.cpp:1034
#9  0x00007fffe7f20819 in (anonymous namespace)::PEI::runOnMachineFunction (this=0x6fb5b0, Fn=...)
    at /home/dylan/projects/llvm-project/llvm/lib/CodeGen/PrologEpilogInserter.cpp:216
#10 0x00007fffe7de1e61 in llvm::MachineFunctionPass::runOnFunction (this=0x6fb5b0, F=...) at /home/dylan/projects/llvm-project/llvm/lib/CodeGen/MachineFunctionPass.cpp:62
#11 0x00007fffe746862f in llvm::FPPassManager::runOnFunction (this=0x6f1a10, F=...) at /home/dylan/projects/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1519
#12 0x00007fffe7468945 in llvm::FPPassManager::runOnModule (this=0x6f1a10, M=...) at /home/dylan/projects/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1540
#13 0x00007fffe74690da in (anonymous namespace)::MPPassManager::runOnModule (this=0x6c42a0, M=...) at /home/dylan/projects/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1596
#14 0x00007fffe7468c06 in llvm::legacy::PassManagerImpl::run (this=0x6c3db0, M=...) at /home/dylan/projects/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1699
#15 0x00007fffe74695f1 in llvm::legacy::PassManager::run (this=0x7fffffffd970, M=...) at /home/dylan/projects/llvm-project/llvm/lib/IR/LegacyPassManager.cpp:1730
#16 0x0000000000423115 in compileModule (argv=0x7fffffffe248, Context=...) at /home/dylan/projects/llvm-project/llvm/tools/llc/llc.cpp:617
#17 0x000000000042123c in main (argc=3, argv=0x7fffffffe248) at /home/dylan/projects/llvm-project/llvm/tools/llc/llc.cpp:358
```

Looking at the error, I'm guessing that somehow we've corrupted the intrusive linked list structure. The assertion we are hitting looks like it's just validating the structure of the list.

We are _probably_ doing something wrong around `AVRRegisterInfo.cpp:169`.

dylanmckay, 2017-07-04:
The problem is not that the list is corrupted - it is that we are calling `std::next` on an iterator that has ended.

Here is the entire basic block in question for the reproduction code in the description

```
%R24<def,dead> = INRdA 46; mem:Volatile LD1[inttoptr (i16 78 to i8*)](align=2)
%R17R16<def> = MOVWRdRr %R29R28, 0, %SREG<imp-def,dead>
```

We get right up to the `foldFrameOffset` call with and so in this call here

```
foldFrameOffset(*std::next(II), Offset, DstReg);
```

We give a strange result for the first argument.

dylanmckay, 2017-07-04:
It looks like the `foldFrameOffset` is just an optimisation to combine two `ADD` instructions together. Easy workaround is to comment the line out, but I will find a proper fix.

dylanmckay, 2017-07-04:
Here is the basic block with the `FRMIDX` before we mutate it

```
BB#2: derived from LLVM BB %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit6
    Live Ins: %R29R28
    Predecessors according to CFG: BB#1
        %R24<def,dead> = INRdA 46; mem:Volatile LD1[inttoptr (i16 78 to i8*)](align=2)
        %R17R16<def> = FRMIDX <fi#0>, 0, %SREG<imp-def,dead>
    Successors according to CFG: BB#3(?%)
```

`foldFrameOffset` itself has a bunch of checks that return early if the next instruction is not relevant. We should be able to just add a predicate to the call of `foldFrameOffset`.

dylanmckay, 2017-07-04:
Fixed upstream in [r307057](http://llvm.org/viewvc/llvm-project?view=revision&revision=307057) and cherry-picked in 897020690cab72e17280af259dfc310b17aa49cc.

closed 2017-07-04 by dylanmckay