avr-rust/rust-legacy-fork #62: Another MBB reordering bug

gergoerdi, 2017-05-20:
This one seems similar to https://github.com/avr-rust/rust/issues/49. I don't have a shrunk example yet.

LLVM IR:

```
target triple = "avr-atmel-none"

%str_slice = type { i8*, i16 }
%"core::option::Option<chip8_engine::opcodes::Op>" = type { i16, [0 x i16], [2 x i16] }
%"chip8_engine::machine::Machine" = type { i16, [0 x i8], i16, [0 x i8], [16 x i8], [0 x i8] }

@str.0 = internal constant [68 x i8] c"/home/cactus/prog/rust/avr/stripped8/chip8-engine-avr/src/machine.rs"
@panic_bounds_check_loc.1 = internal unnamed_addr constant { %str_slice, i32 } { %str_slice { i8* getelementptr inbounds ([68 x i8], [68 x i8]* @str.0, i32 0, i32 0), i16 68 }, i32 41 }, align 4
@panic_bounds_check_loc.2 = internal unnamed_addr constant { %str_slice, i32 } { %str_slice { i8* getelementptr inbounds ([68 x i8], [68 x i8]* @str.0, i32 0, i32 0), i16 68 }, i32 36 }, align 4
@"_ZN38_$LT$core..option..Option$LT$T$GT$$GT$6unwrap14_MSG_FILE_LINE17h1ad372c436feeebeE" = external global { i32, [0 x i8], %str_slice, [0 x i8], %str_slice, [0 x i8] }

; Function Attrs: noinline norecurse nounwind uwtable
define internal fastcc void @_ZN9chip8_avr3spi5setup17he6a9e8a8cb3a0461E() unnamed_addr #1 {
start:
  %0 = load volatile i8, i8* inttoptr (i16 37 to i8*), align 1
  %1 = or i8 %0, 4
  store volatile i8 %1, i8* inttoptr (i16 37 to i8*), align 1
  %2 = load volatile i8, i8* inttoptr (i16 36 to i8*), align 4
  %3 = or i8 %2, 44
  store volatile i8 %3, i8* inttoptr (i16 36 to i8*), align 4
  %4 = load volatile i8, i8* inttoptr (i16 36 to i8*), align 4
  %5 = and i8 %4, -17
  store volatile i8 %5, i8* inttoptr (i16 36 to i8*), align 4
  store volatile i8 80, i8* inttoptr (i16 76 to i8*), align 4
  ret void
}

; Function Attrs: noreturn nounwind uwtable
define void @main() unnamed_addr {
start:
  %_12.i = alloca %"core::option::Option<chip8_engine::opcodes::Op>", align 8
  %machine = alloca %"chip8_engine::machine::Machine", align 8
  tail call fastcc void @_ZN9chip8_avr3spi5setup17he6a9e8a8cb3a0461E() #6
  %0 = bitcast %"chip8_engine::machine::Machine"* %machine to i8*
  call void @_ZN12chip8_engine7machine7Machine3new17h5e85f0933e39358cE(%"chip8_engine::machine::Machine"* noalias nocapture nonnull sret dereferenceable(20) %machine) #6
  %1 = getelementptr inbounds %"chip8_engine::machine::Machine", %"chip8_engine::machine::Machine"* %machine, i16 0, i32 2
  %2 = load i16, i16* %1, align 2
  %3 = lshr i16 %2, 8
  %4 = trunc i16 %3 to i8
  store volatile i8 %4, i8* inttoptr (i16 78 to i8*), align 2
  br label %bb2.i

bb2.i:                                            ; preds = %bb2.i, %start
  %5 = load volatile i8, i8* inttoptr (i16 77 to i8*), align 1
  %6 = icmp sgt i8 %5, -1
  br i1 %6, label %bb2.i, label %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit

_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit:  ; preds = %bb2.i
  %7 = load volatile i8, i8* inttoptr (i16 78 to i8*), align 2
  %8 = trunc i16 %2 to i8
  store volatile i8 %8, i8* inttoptr (i16 78 to i8*), align 2
  br label %bb2.i1

bb2.i1:                                           ; preds = %bb2.i1, %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit
  %9 = load volatile i8, i8* inttoptr (i16 77 to i8*), align 1
  %10 = icmp sgt i8 %9, -1
  br i1 %10, label %bb2.i1, label %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit2

_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit2: ; preds = %bb2.i1
  %11 = load volatile i8, i8* inttoptr (i16 78 to i8*), align 2
  store volatile i8 66, i8* inttoptr (i16 78 to i8*), align 2
  br label %bb2.i3

bb2.i3:                                           ; preds = %bb2.i3, %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit2
  %12 = load volatile i8, i8* inttoptr (i16 77 to i8*), align 1
[... 239 lines cut ...]
	movw	r16, r28
	subi	r16, 255
	sbci	r17, 255
	mov	r24, r16
	mov	r25, r17
	call	_ZN12chip8_engine7machine7Machine3new17h5e85f0933e39358cE
	ldd	r24, Y+3
	ldd	r25, Y+4
	out	46, r25
	ori	r16, 2
LBB1_1:                                 ; %"_ZN75_$LT$chip8_avr..Board$u20$as$u20$chip8_engine..peripherals..Peripherals$GT$12scan_key_row17haa3ed42a3d3a5068E.exit.i"
	lsl	r21
	subi	r25, 1
	breq	.+2
	rjmp	LBB1_1
	rjmp	LBB1_2
LBB1_6:                                 ; %bb2.i
	in	r18, 45
	tst	r18
	brmi	.+2
	rjmp	LBB1_6
; BB#7:                                 ; %_ZN9chip8_avr3spi4sync17hb1a38c1bcef3b0caE.exit
	in	r18, 46
	out	46, r24
```

gergoerdi, 2017-05-20:
This seems fixable with the exact same change as I did on the `Select8`/`Select16` one.

gergoerdi, 2017-05-20:
While the IR from my original comment works with this change, it now breaks `libcore-mini` with

```
llc: /home/cactus/prog/rust/rust-avr/rust/src/llvm/include/llvm/Support/GenericDomTree.h:586: llvm::DomTreeNodeBase<NodeT>* llvm::DominatorTreeBase<NodeT>::addNewBlock(NodeT*, NodeT*) [with NodeT = llvm::MachineBasicBlock]: Assertion `IDomNode && "Not immediate dominator specified for block!"' failed.

#8 0x00000000010d8726 llvm::DominatorTreeBase<llvm::MachineBasicBlock>::addNewBlock(llvm::MachineBasicBlock*, llvm::MachineBasicBlock*) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10d8726)
#9 0x00000000010d8e4e llvm::MachineDominatorTree::applySplitCriticalEdges() const (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10d8e4e)
#10 0x00000000010d315f (anonymous namespace)::MachineCSE::runOnMachineFunction(llvm::MachineFunction&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10d315f)
#11 0x00000000010e8603 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x10e8603)
#12 0x000000000138bae3 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138bae3)
#13 0x000000000138bb8c llvm::FPPassManager::runOnModule(llvm::Module&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138bb8c)
#14 0x000000000138c8ef llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x138c8ef)
#15 0x0000000000697f74 compileModule(char**, llvm::LLVMContext&) (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x697f74)
#16 0x0000000000646a90 main (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x646a90)
#17 0x00007f761a427f45 __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21f45)
#18 0x000000000068d2b5 _start (/home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc+0x68d2b5)
Stack dump:
0.	Program arguments: /home/cactus/prog/rust/rust-avr/build/build/x86_64-unknown-linux-gnu/llvm/bin/llc core-0b8560bbca9581ff.0.ll 
1.	Running pass 'Function Pass Manager' on module 'core-0b8560bbca9581ff.0.ll'.
2.	Running pass 'Machine Common Subexpression Elimination' on function '@"_ZN4core3num20_$LT$impl$u20$u8$GT$25checked_next_power_of_two17he287dc882e996a17E"'
```

gergoerdi, 2017-05-20:
To reproduce with https://github.com/gergoerdi/llvm-avr/commit/f7931dedd0dc90418d41da5ed6266e1d6d7ac3fc:

```
target triple = "avr-atmel-none"

; Function Attrs: nounwind readnone uwtable
define i16 @"_ZN4core3num20_$LT$impl$u20$u8$GT$25checked_next_power_of_two17he287dc882e996a17E"(i8) unnamed_addr #0 {
start:
  %1 = add i8 %0, -1
  %2 = tail call i8 @llvm.ctlz.i8(i8 %1, i1 false) #15
  %3 = sub nsw i8 0, %2
  %4 = and i8 %3, 7
  %5 = shl i8 1, %4
  %not. = icmp uge i8 %5, %0
  %. = zext i1 %not. to i16
  %_0.sroa.3.0.insert.ext = zext i8 %5 to i16
  %_0.sroa.3.0.insert.shift = shl nuw i16 %_0.sroa.3.0.insert.ext, 8
  %_0.sroa.0.0.insert.insert = or i16 %_0.sroa.3.0.insert.shift, %.
  ret i16 %_0.sroa.0.0.insert.insert
}

; Function Attrs: nounwind readnone
declare i8 @llvm.ctlz.i8(i8, i1) #14
```

but I have a movie to catch RN :)

gergoerdi, 2017-05-20:
Hah, commit simply had a bug where I forgot to remove the old `I++` while copy-pasting in a new one from the `Select8` implementation. Fix is now good.

dylanmckay, 2017-05-23:
Looks like the fix is in avr-rust/llvm#2

closed 2017-06-18 by gergoerdi

dylanmckay, 2017-09-26:
My [`check-everything-upstreamed.rb`](https://gist.github.com/dylanmckay/221f127db7b52cd9e168dd9abc7450e6) script found that I never ended up upstreaming the fix for this to LLVM proper, doing now

dylanmckay, 2017-09-26:
Committed in [r314179](http://llvm.org/viewvc/llvm-project?view=revision&revision=314179).