avr-rust/rust-legacy-fork #66: Best Way to get started?

XOSplicer, 2017-06-30:
Hey, what would you guys suggest to get started with Rust on an Arduino uno? 

Im thinking about how would one get to have rustc with target AVR enabeled. If this is not yet possible, what are other options. What are the current issues to ne solved for this ?

Thanks in advance

gergoerdi, 2017-06-30:
Have you looked at https://github.com/gergoerdi/rust-avr-chip8-avr/blob/master/README.md ? I have put detailed instructions there for getting rustc built with the right branch of LLVM.

XOSplicer, 2017-07-01:
yeaha, I've built it. My first rust ELF
But I needed a little help from this one:
https://www.reddit.com/r/rust/comments/6i6yz5/build_instructions_for_chip8avr/
it did not work out of the box with xargo v0.3.8, I needed to install version 0.3.7 explicitly
Thanks for your hint. I will continue to investigate what I need to create my own avr programs with rust.
The only thing I need is an Arduino in my Hotel room now...

closed 2017-07-01 by XOSplicer