avr-rust/rust-legacy-fork #67: Inline weirdness

dsvensson, 2017-07-04:
The following code ported from avr-libc for dealing with crude delays works as long as it's not inlined. I hope I've ported it correctly, but the real question is why it breaks if I allow it to be inlined?

```
#[inline(never)]
pub fn _delay_loop_2(count: u16) {
    unsafe {
        asm!("1: sbiw $0, 1
                 brne 1b"
             : : "r"(count))
    }
}
```

Removing inline yields this error:
```
   Compiling hello v0.1.0 (file:///home/daniel/Development/avr/hello)
error: <inline asm>:1:10: error: invalid operand for instruction
        1: sbiw r18, 1
                ^

  --> src/main.rs:41:9
   |
41 | /         asm!("1: sbiw $0, 1
42 | |                  brne 1b"
43 | |              : : "r"(count))
   | |____________________________^

error: <inline asm>:1:10: error: invalid operand for instruction
        1: sbiw r18, 1
                ^

  --> src/main.rs:41:9
   |
41 | /         asm!("1: sbiw $0, 1
42 | |                  brne 1b"
43 | |              : : "r"(count))
   | |____________________________^

error: aborting due to 2 previous errors

error: Could not compile `hello`.

To learn more, run the command again with --verbose.
```

The delay in avr-libc can be found here:
http://www.nongnu.org/avr-libc/user-manual/delay__basic_8h_source.html

dsvensson, 2017-07-04:
Adding "volatile" as an option to asm!() makes it compile. The code doesn't look completely correct though, as it will consume the counter parameter which will never be restored, but that should be fixable. I'll look into this tonight, maybe the issue can be closed.

dsvensson, 2017-07-04:
After much bruteforcing this seems like a workable solution

```
// http://www.nongnu.org/avr-libc/user-manual/group__util__delay__basic.html
#[inline(always)]
pub fn _delay_loop_2(count: u16) {
    unsafe {
        let mut __count = count;
        asm!("1: sbiw $0, 1
                 brne 1b"
             : "=w"(__count)
             : "0"(__count)
             :
             : "volatile")
    }
}
```

So this issue should be closed. But the question is if I should open tickets about all the weirdness that accepted (when building via xargo, rustc rejected some of the more broken syntax) my faulty asm!() usage, or is that a generic rust issue?

dylanmckay, 2017-07-04:
I think this might be because SBIW only accepts the top 4 GPR register pairs (there are a few instructions which do this) and the `r` constraint represents any register and so the compiler has no problem allocating `count` to `r18` which then errors out when used in `sbiw` because it isn't a top four gpr pair.

You might need to use a more specific constraint.

It is probably just chance that the online code happens to allocate registers that aren't in the top four pairs, whereas the non inlined version does.

dylanmckay, 2017-07-04:
*inline

dylanmckay, 2017-07-04:
I have a feeling that gcc would do this too, unless it was pretty smart about looking through inline assembly and allocating outside variables accordingly

dsvensson, 2017-07-05:
The problem was that asm!() swallowed bad syntax without emitting any warnings. It works correctly now that I've figured out how to write what I intend it to do.

closed 2017-07-05 by dsvensson