avr-rust/rust-legacy-fork #69: Switch lookup tables end up in .data instead of .text by default

gergoerdi, 2017-07-15:
I've been struggling for a couple days now on why my CHIP-8 firmware doesn't work anymore, even when downgrading to the exact same rustc/llvm versions as before. Then I found out that it's not because of anything in the compiler, but it's because in the meantime I switched over to using Xargo for linking instead of a custom shell script.

The following program should demonstrate the problem. It is the result of compiling a switch, not unlike the one in #47. The stripped-down IR is as follows:

```
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-atmel-none"

@switch.table = private unnamed_addr constant [4 x i8] c"\01\02\03\0C"
@switch.table.1 = private unnamed_addr constant [4 x i8] c"\04\05\06\0D"
@switch.table.2 = private unnamed_addr constant [4 x i8] c"\07\08\09\0E"
@switch.table.3 = private unnamed_addr constant [4 x i8] c"\0A\00\0B\0F"

; Function Attrs: noinline norecurse nounwind uwtable
define internal fastcc void @spi_setup() {
start:
  %0 = load volatile i8, i8* inttoptr (i16 37 to i8*), align 1
  %1 = or i8 %0, 4
  store volatile i8 %1, i8* inttoptr (i16 37 to i8*), align 1
  %2 = load volatile i8, i8* inttoptr (i16 36 to i8*), align 4
  %3 = or i8 %2, 44
  store volatile i8 %3, i8* inttoptr (i16 36 to i8*), align 4
  %4 = load volatile i8, i8* inttoptr (i16 36 to i8*), align 4
  %5 = and i8 %4, -17
  store volatile i8 %5, i8* inttoptr (i16 36 to i8*), align 4
  store volatile i8 80, i8* inttoptr (i16 76 to i8*), align 4
  ret void
}

; Function Attrs: noinline norecurse nounwind uwtable
define internal fastcc void @spi_sync(i8) unnamed_addr #1 {
start:
  store volatile i8 %0, i8* inttoptr (i16 78 to i8*), align 2
  br label %bb2

bb2:                                              ; preds = %bb2, %start
  %1 = load volatile i8, i8* inttoptr (i16 77 to i8*), align 1
  %2 = icmp sgt i8 %1, -1
  br i1 %2, label %bb2, label %bb3

bb3:                                              ; preds = %bb2
  %3 = load volatile i8, i8* inttoptr (i16 78 to i8*), align 2
  ret void
}

; Function Attrs: noinline uwtable
define internal fastcc i8 @coords_key(i16) unnamed_addr #2 {
start:
  %abi_cast.sroa.0.0.extract.trunc = trunc i16 %0 to i8
  %abi_cast.sroa.4.0.extract.shift = lshr i16 %0, 8
  %abi_cast.sroa.4.0.extract.trunc = trunc i16 %abi_cast.sroa.4.0.extract.shift to i8
  switch i8 %abi_cast.sroa.0.0.extract.trunc, label %bb17 [
    i8 0, label %bb18
    i8 1, label %bb19
    i8 2, label %bb20
    i8 3, label %bb21
  ]

bb17:                                             ; preds = %bb21, %bb20, %bb19,
  unreachable

bb18:                                             ; preds = %start
  %1 = icmp ult i8 %abi_cast.sroa.4.0.extract.trunc, 4
  br i1 %1, label %switch.lookup, label %bb17

bb19:                                             ; preds = %start
  %2 = icmp ult i8 %abi_cast.sroa.4.0.extract.trunc, 4
  br i1 %2, label %switch.lookup1, label %bb17

bb20:                                             ; preds = %start
[... 400 lines cut ...]

The customization compared to the default `avr-gcc` linker script is as follows (it moves `.rodata` to `.text`):

```
--- default.ld	2017-07-15 15:20:01.762345763 +0800
+++ lookup-text.ld	2017-07-15 15:01:27.386390685 +0800
@@ -146,6 +146,8 @@
     *(.fini0)  /* Infinite loop after program termination.  */
     KEEP (*(.fini0))
      _etext = . ;
+    *(.rodata)  /* We need to include .rodata here if gcc is used */
+    *(.rodata*) /* with -fdata-sections.  */
   }  > text
   .data	  : AT (ADDR (.text) + SIZEOF (.text))
   {
@@ -155,8 +157,6 @@
        line will have no effect, see PR13697.  Thus, keep .data  */
     KEEP (*(.data))
     *(.data*)
-    *(.rodata)  /* We need to include .rodata here if gcc is used */
-    *(.rodata*) /* with -fdata-sections.  */
     *(.gnu.linkonce.d*)
     . = ALIGN(2);
      _edata = . ;
```

gergoerdi, 2017-07-15:
I realize that this would be correct behavior if we didn't use progmem for lookup tables (but we want to use progmem for them, as agreed in #47) and copied `.rodata` to RAM at startup. So instead, we should put some custom annotation on the lookup table `constant`s to tell LLVM that we are only ever going to access them via bespoke, `lpm`-using code (so we'll never try taking a pointer to them), and put those in the `.text` section.

dylanmckay, 2017-07-16:
I'm confused - isn't this a duplicate of #47? Both are switch tables that are placed into the wrong address space.

By the way, I've got a patch that fixes #47 and #65 locally, still need to upstream it though.

gergoerdi, 2017-07-17:
I thought #47 was, basically, about figuring out that these particular constants should be loaded via `lpm`, and this ticket is about what to tell the linker about them. But yeah, they're related.

dylanmckay, 2017-07-19:
I think I may have misunderstood this issue before - I need to read up a bit on which sections are actually used on AVR.

From what I understand, the issue is that even when switch lookup tables are placed in the program memory address space, they are located in the wrong section?

If this is the case, _I think_ we can add hooks to `AVRELFStreamer.cpp`/`AVRELFObjectWriter` that do this, but I don't actually know anything about this area.

dylanmckay, 2017-07-19:
By the way, I don't think I can delete "dylanmckay marked this as a duplicate of #47 "

gergoerdi, 2017-08-27:
I think I managed to un-mark it.

dylanmckay, 2019-01-18:
We should be able to solve this by placing the switch table into one of the sections merged into `.text` by default, like `.lowtext` (arbitrarily picked).