avr-rust/rust-legacy-fork #70: Look into finding a way to not require target specification files

dylanmckay, 2017-08-23:
Currently, all binaries that use avr-rust need to create a target specification file to instruction Rust to compile for a specific microcontroller.

The specification itself is always identical across devices, with the only difference being the CPU name. This means that if you want to support microcontroller <x> in your executable, you need to create a target specifcation for it.

It might be easier for everyone if there is another method like a command line argument for it. I am not sure of any solutions yet, it needs some investigation.

shepmaster, 2017-08-23:
> specification itself is always identical across devices

I'm not sure that I fully agree with that, as I'm pretty sure mine are different from what most other people use (as I avoid the AVR-GCC standard lib).

That being said, it certainly makes sense to have some easier way to deal with this.

dylanmckay, 2017-08-23:
[rust-arduino-blink-led-no-core-with-cargo/blink/arduino.json](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/master/blink/arduino.json)

Interesting, that makes sense.

gergoerdi, 2017-08-27:
For what it's worth, I also use a custom target file in `chip8`: https://github.com/gergoerdi/rust-avr-chip8-avr/blob/f6e1ded1354dce26061c9ef4c2fbd4641aee701b/avr-atmega328p.json

The reasons for that are:
* Workaround for https://github.com/avr-rust/rust/issues/69 by using a custom linker script
* Linking in a non-Rust object file

The first should be done for all projects unconditionally until the linked issue is fixed. But I don't know how to do the second correctly. Is there a way to tell Cargo to compile and link in a C source? There must be a solution for this for FFI-using packages, I'd hope.

Alternatively, I could replace those C sources with Rust if we had a way of explicitly marking something (a global `const &'static str` maybe?) then I wouldn't have the C dependency anymore.

dylanmckay, 2017-08-27:
> Is there a way to tell Cargo to compile and link in a C source? 

Yes. In a build script, you can invoke GCC, generate an object file, and then print a formatted string (something like `println!("cargo:link=object=<path>")`) and it will do the linking automatically.

The [gcc](https://crates.io/crates/gcc) crate will do this all for you though, which makes it easy.

> I could replace those C sources with Rust if we had a way of explicitly marking something

Do you mean marking something as living in program memory? If so, we should be able to add a new attribute pretty easily I think. I can't think of any more idiomatic ways of doing it without drastically changing the syntax.