avr-rust/rust-legacy-fork #71: Copy globals from program memory into RAM

dylanmckay, 2017-08-24:


dylanmckay, 2017-08-24:
It looks like this function is implemented in `libgcc`, which is good news for us.

I will check if we currently call this. If not, it's possible for us to emit a call to it.

shepmaster, 2017-08-24:
And [my ASM to attempt to do the same thing](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/d85d11171152d25d11c58ccb19270cffa545b18b/initialize_memory.S)

shepmaster, 2017-08-24:
> it's possible for us to emit a call to it.

Which I'd like to be controllable, please :-)

dylanmckay, 2017-08-24:
I compiled avr-rust/blink. The interrupt table is correctly there, as well as the `__ctors_end` function which seems to be initialising the stack.

This `__ctors_end` function calls `main` directly after this, with no mention of `__do_stack_copy`.

shepmaster, 2017-08-24:
> avr-rust/blink

Neat, I missed that!

> which seems to be initialising the stack

For reference, I [have this as well](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/d85d11171152d25d11c58ccb19270cffa545b18b/blink/src/main.rs#L65-L71). The key parts are setting the zero register and populating the stack size, then copying globals to RAM.

> The interrupt table is correctly there

Which I [have separately](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/d85d11171152d25d11c58ccb19270cffa545b18b/interrupt_vector.S) and it's all glued together with [linker flags](https://github.com/shepmaster/rust-arduino-blink-led-no-core-with-cargo/blob/d85d11171152d25d11c58ccb19270cffa545b18b/blink/arduino.json#L17-L54).

dylanmckay, 2017-08-24:
I've looked into this a little further and I believe the `do_copy_data` insertion is done based on some sort of logic implemented in the `avr-gcc` (not `avr-ld`) frontend.

I have confirmed this by taking a C program that reads a global, compiling it to an object file (which has not been linked with libgcc and so does not contain `do_copy_data`), and then linking it specifically with `avr-ld`.

This generates an executable which does not copy data to RAM, and is therefore broken.

I would like to make LLVM output code that triggers `avr-gcc `s `do_copy_data` function. I suspect we will need to place some variables into a specific section, or set a flag on the symbols in ELF.

You will still be able to avoid the insertion of the `do_copy_data` in the same way you can today with GCC - by using `avr-ld` directly

shepmaster, 2017-08-24:
I'm mostly just mirroring what I discovered GCC to do in my research. My true hope is to have a completely non-GCC reliant solution to allow for the flexibility to have competing solutions. I'm also interested in the possibility of using lld ;-)

dylanmckay, 2017-08-24:
I'm definitely interested in `lld` as well, do you know if there are any plans to embed it into Rust and prefer it over binutils?

dylanmckay, 2017-08-24:
There's a small skeleton for AVR support already in lld, which is neat.

shepmaster, 2017-08-24:
> linking it specifically with avr-ld.
> 
> This generates an executable which does not copy data to RAM, and is therefore broken.

That's quite a surprising result! I wonder if there's an avr-ld flag that might control the setting...

> any plans to embed it into Rust

I've heard discussion about some plans here, but I'm not 100% sure what the main motivation for the broader Rust community is. I *think* lld might be faster than ld or gold? Maybe it's just to avoid having to install a C compiler / linker to run Hello World?

That being said, we already have to specify the specific linker now, so presumably specifying `lld` shouldn't be *too* hard...

dylanmckay, 2017-08-24:
I've found this in AVR-gcc (`config/avr/avr.c`)

```c
avr_need_copy_data_p = (STR_PREFIX_P (name, ".data")
                         || STR_PREFIX_P (name, ".gnu.linkonce.d"));
```

Note that `STR_PREFIX_P` is a macro that is true if the first arguments starts with the second argument.

I believe this is the logic which decides whether or not we need a copy_data section. 

When I disassemble Rust blink, I note that there are dozens of symbols in the output program, but there is only one variable inside a section starting with `.data`

```
00000000 l    d  .data.rel.ro.panic_loc.2       00000000 .data.rel.ro.panic_loc.2
```

I'm not sure why this function isn't triggering the `do_copy_data`, maybe it's related to the panic not actually being hit, I'm not sure.

When I modify the blink example to load from a constant static `FOO`, I get this

```
00000000 l     O .rodata._ZN5blink3FOO17h3a721bc44740f6b9E      00000004 _ZN5blink3FOO17h3a721bc44740f6b9E
```

So the constant is getting placed into `.rodata`, which also wouldn't trigger the `do_copy_data`.

When I change the constant static to a mutable static, the symbols do make it into the `.data` section.

```
00000000 l     O .data._ZN5blink3FOO17h3a721bc44740f6b9E        00000004 _ZN5blink3FOO17h3a721bc44740f6b9E
00000000 l    d  .data._ZN5blink3FOO17h3a721bc44740f6b9E        00000000 .data._ZN5blink3FOO17h3a721bc44740f6b9E
```

And now for the moment of truth - does this generate `do_copy_data`.

No, it does not.

dylanmckay, 2017-08-24:
Interestingly, it looks like `avr-gcc` also checks for `.rodata` in the same function. There is something else going on.

dylanmckay, 2017-08-24:
I will message the avr-gcc mailing list

dylanmckay, 2017-09-11:
I've looked into this more and I think I'm getting mixed up.

The C compiler of AVR-GCC is declaring `do_copy_data`, which leads to us calling into it via the CRT.

The only line required to trigger copying is `.global __do_copy_data` at some point in the assembly file. If the symbol is declared, it will be used, otherwise no data will be copied.

dylanmckay, 2017-09-11:
Fixed in [r312905](http://llvm.org/viewvc/llvm-project?view=revision&revision=312905), cherry-picked in 6092c82d02b2498b413cbdee5b8617186b70425d.

closed 2017-09-11 by dylanmckay

shepmaster, 2017-09-11:
> The only line required to trigger copying is .global __do_copy_data at some point in the assembly file. If the symbol is declared, it will be used, otherwise no data will be copied.

Can you give me the TL;DR of how to opt-in to my own non-GCC-provided code?

dylanmckay, 2017-09-11:
Alright, so `__do_copy_data` is defined in the CRT libraries.

It appears that if you declare the symbol, the CRT library will call it before it calls main. If you do not declare it, it isn't called and the function doesn't make it into the final object file.

Thus, if you don't want to use it, then don't link the CRT. I believe this works well because you're trying to avoid libc/crt code, and so there's no issue there.

In the case where no CRT is linked, it just means there is one extra symbol in the executable, but I imagine that it will get trimmed out as there are no references to it.

For the other readers, linking with `avr-gcc` will always link the CRT and avr-libc. This is because using a C compiler to link will result in a C-centric executable. Because of this, using `avr-gcc` to link with avr-llvm now will always result in data being copied to RAM and BSS being cleared, which is good.

Now, if you link with `avr-ld`, you do not automatically include the CRT or avr-libc. This means that doing this will not result in `do_copy_data` being called. In this case, you will need to roll that functionality yourself.

shepmaster, 2017-09-12:
Thanks!

> linking with avr-gcc will always link the CRT and avr-libc

I've been cheating and actually using `avr-gcc`, so I guess I'll have to figure out how to switch... 🤞

dylanmckay, 2017-09-12:
I have a feeling that `-nostartfiles` will probably do it