avr-rust/rust-legacy-fork #85: Develop crate for compile-time execution delays

dylanmckay, 2017-11-13:
We should develop a crate that lets people do something like this

```rust
fn main() {
    loop {
        write_volatile(PORTB, 0xff);

        delay_ms!(1000);

        write_volatile(PORTB, 0x00);

        delay_ms!(1000);
    }
}
```

This will mean we won't have issues with hacky "long empty for loop" delays like in avr-rust/blink#3.

We should develop an easy to use crate with macros for delays and encourage its use. We could potentially set up a small (markdown?) web app/github pages site that lists common patterns in AVR like timers/delays and then has a small blurb that shows a small example.

shepmaster, 2017-11-14:
Why would this be a completely separate crate as opposed to some part of the arduino crate?

shepmaster, 2017-11-14:
(I'm also generally "meh" on the idea of busy loops as timers should always be a better solution)

dylanmckay, 2017-11-14:
> Why would this be a completely separate crate as opposed to some part of the arduino crate?

Just so we don't tie people to that crate. We can easily reexport the timer functionality in the arduino crate, but I think there's a good case for flexibility.

neu-rah, 2017-11-14:
when not using busy loops... what can we use? timers?
that can collide with user code if they also happen to use that timer or at least denies that timer for user code.

using it with non-specified timer makes it more flexible but also more complex... well once timers are done and a good map of the different avr timers is available that should be more easy

here some arduino standard core about delay

https://github.com/arduino/ArduinoCore-avr/blob/fb7c2cc02ebc38319ec86901d0a3c9f7249fb041/cores/arduino/wiring.c#L120

pusherofbrooms, 2017-11-14:
Busy-wait loops seem to be popular for instructional code (like a simple blink), and for simple code that doesn't need accuracy or interrupts.

Arduino packages all this stuff up in a big ball, but I can see value in splitting out the functionality into individual crates, and maybe having an over-arching crate pull them in for those who want convenience.

pusherofbrooms, 2017-11-14:
I started work on a busy-wait delay library. https://github.com/pusherofbrooms/avr-delay. It's functional, but I don't write a lot of rust, and I've certainly never written rust for public consumption, so this is mostly warts.

I'm about to write up some issues for the code, but the highlights:
1) get rid of the magic number
2) figure out how to handle the MCU_SPEED constant so that it can be set outside the library
3) add a delay_us()
4) consider converting the public functions to macros, though I don't think I understand the advantage.

dylanmckay, 2017-11-14:
That looks neat!

I think the best way we can guarantee accuracy is by defining a naked function that executes for a well-defined known of CPU cycles. If we use an instruction with the same clock delay on all cores, it won't complicate delays when the naked delay function takes different times on different CPUs.

This function would likely by written in something like assembly.

The `delay_ms` function you have could adjust its computation to calculate how many iterations of our naked function needs to be executed, account for the few cycles it takes to call it (if desired), and then call it for that many iterations.

Also, we should utilize the `const fn` support in nightly to ensure we don't offset delays due to the iteration calculation. We can benefit from `const fn` support compared to AVR-GCC, which requires programs to be compiled at a specific optimisation level which the AVR-GCC developers know will fully inline/optimise the `_delay_ms` call.

Also, I'm curious what your magic number represents @pusherofbrooms - is it some sort of cycle count estimate of the loop?

dylanmckay, 2017-11-14:
I'm going to give implementing this a go

pusherofbrooms, 2017-11-14:
That's the way the arduino delay library does it. They have a carefully timed ASM loop wrapped by a C function.

The magic number is just a cycle count estimate. I need to break that down into its components.

pusherofbrooms, 2017-11-19:
I did some work to get this crate into better shape. https://github.com/pusherofbrooms/avr-delay HEAD contains the results. I put as much of the looping as I could in asm. I ignore about 13 to 20 cycles of setup time and about 11 cycles per outer loop.

dylanmckay, 2017-11-19:
I am a fan of your implementation - once we're sure it more-or-less works, we could move it to the avr-rust organisation (if you want - no problem if you'd rather leave it on your personal profile).

pusherofbrooms, 2017-11-19:
I'm good with maintaining it under avr-rust @dylanmckay . Does it make sense to select the most used LICENSE file from the avr-rust project and use that? Did you want to handle the mechanics of the move, or will you just open a new repo and let me push the code up? Once it's moved, I'll update the README

dylanmckay, 2017-11-20:
> Does it make sense to select the most used LICENSE file from the avr-rust project and use that? 

I've pretty much used MIT for everything although LLVM itself is licensed under an apache derivative IIRC. Rust also continues to keep its upstream license, whatever that is (MIT?)

> Did you want to handle the mechanics of the move, or will you just open a new repo and let me push the code up?

I think there's a way for you to initiate a transfer (under repo settings), and us to approve it. If you can't find out how to do that, I can happily create an empty repo and give you push access.

shepmaster, 2017-11-21:
> I think there's a way for you to initiate a transfer (under repo settings)

I think you actually will need to transfer it to a person in the organization first (such as myself or Dylan) who can then move it into the organization. You could also be made a member of the organization.

pusherofbrooms, 2017-11-21:
As @shepmaster says, I don't have permissions to transfer it directly into AVR-Rust. Also, @dylanmckay has a fork, so I can't yet transfer it to him (same name repository). I'm happy to do whatever is most convenient.

I added an apache and an MIT license file such that it lines up with a lot of rust projects.

dylanmckay, 2017-11-23:
@pusherofbrooms 

I have deleted my fork - feel free to transfer to me and I will do the rest and comment here.

dylanmckay, 2017-11-24:
This has now been completed - the crate can now be found at [avr-rust/delay](https://github.com/avr-rust/delay)

If we have any issues or feature requests, we can raise them there. Closing this, nice work @pusherofbrooms!

closed 2017-11-24 by dylanmckay