avr-rust/rust-legacy-fork #86: Revert temporary cherry-picked fix for #72 in favor of upstream patch

dylanmckay, 2017-11-14:
I cherry-picked a commit to fix #72. This commit has not been reviewed or committed into LLVM upstream yet.

Once it has, we should revert the old commit and cherry-pick the new one (assuming the diff changes during code review). This ensures that our fork is a close as possible to LLVM trunk which also means less merge conflicts.

dylanmckay, 2017-12-09:
Upstream commit landed in [r320243](https://reviews.llvm.org/rL320243)

I attempted to cherry-pick, but there have been a number of other changes to these files which mean we would have to modify quite a bit of code to get the patch to apply cleanly.

We should do this when we next upgrade LLVM

shepmaster, 2018-11-03:
We've recently upgraded to LLVM 8 🎉 I'm going to close any bug that is reported against an older version of LLVM. If you are still having this issue with the LLVM-8 based code, **please** ping me and I can reopen the issue!

closed 2018-11-03 by shepmaster