avr-rust/rust-legacy-fork #89: back end crashes on this IL: LowerFormalArguments didn't emit the correct number of values for a tuple return value

carlos4242, 2017-12-28:
From this swift code....

public func i2cGetByte(slaveAddress address: UInt8) -> (success: Bool, byte: UInt8) {
	return (success: true, byte: _i2cReadSingleByte(address))
}

...the llvm IR produced is...

define { i1, i8 } @_TF3AVR10i2cGetByteFT12slaveAddressVs5UInt8_T7successSb4byteS0__(i8) #2 {
entry:
  %1 = call zeroext i8 @_i2cReadSingleByte(i8 zeroext %0)
  %2 = insertvalue { i1, i8 } { i1 true, i8 undef }, i8 %1, 1
  ret { i1, i8 } %2
}

...but this crashes llc...

"/Users/carlpeto/llvm/llvm-avr/llc" -O=0 -march=avr -mcpu=atmega328p -filetype=obj -o serial.o serial.ll
Assertion failed: (InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"), function LowerArguments, file /Users/carlpeto/llvm/llvm2/llvm-avr/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp, line 8141.
0  llc                      0x000000010e43ac0c llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 60
1  llc                      0x000000010e43b189 PrintStackTraceSignalHandler(void*) + 25
2  llc                      0x000000010e4371a9 llvm::sys::RunSignalHandlers() + 425
3  llc                      0x000000010e43b622 SignalHandler(int) + 354
4  libsystem_platform.dylib 0x00007fffd0bfdb3a _sigtramp + 26
5  libsystem_platform.dylib 0x0100000000000000 _sigtramp + 792732896
6  libsystem_c.dylib        0x00007fffd0a82420 abort + 129
7  libsystem_c.dylib        0x00007fffd0a49893 basename_r + 0
8  llc                      0x000000010e194384 llvm::SelectionDAGISel::LowerArguments(llvm::Function const&) + 4772
9  llc                      0x000000010e23fe39 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 5049
10 llc                      0x000000010e23d0fb llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 1643
11 llc                      0x000000010d1384bb llvm::AVRDAGToDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 59
12 llc                      0x000000010d78b461 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 449
13 llc                      0x000000010dc8d5af llvm::FPPassManager::runOnFunction(llvm::Function&) + 399
14 llc                      0x000000010dc8dab5 llvm::FPPassManager::runOnModule(llvm::Module&) + 117
15 llc                      0x000000010dc8e884 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) + 2196
16 llc                      0x000000010dc8dd76 llvm::legacy::PassManagerImpl::run(llvm::Module&) + 342
17 llc                      0x000000010dc8f5b1 llvm::legacy::PassManager::run(llvm::Module&) + 33
18 llc                      0x000000010cd57662 compileModule(char**, llvm::LLVMContext&) + 21810
19 llc                      0x000000010cd51ef3 main + 659
20 libdyld.dylib            0x00007fffd09ee235 start + 1
21 libdyld.dylib            0x0000000000000008 start + 794893780
Stack dump:
0.	Program arguments: /Users/carlpeto/llvm/llvm-avr/llc -O=0 -march=avr -mcpu=atmega328p -filetype=obj -o serial.o serial.ll 
1.	Running pass 'Function Pass Manager' on module 'serial.ll'.
2.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@_TF3AVR10i2cGetByteFT12slaveAddressVs5UInt8_T7successSb4byteS0__'
Abort trap: 6

dylanmckay, 2018-02-08:
Here's a reproduction that compiles standalone

```llvm
declare i8 @do_something(i8 %val)

define { i1, i8 } @main(i8) #2 {
entry:
  %1 = call zeroext i8 @do_something(i8 zeroext %0)
  %2 = insertvalue { i1, i8 } { i1 true, i8 undef }, i8 %1, 1
  ret { i1, i8 } %2
}
```

This still occurs on

dylanmckay, 2018-02-08:
This is likely a duplicate of #57.

That bug is currently titled "Assertion failure on i128". That title is not really accurate though, as from memory I believe when I investigated it the bug did not seem to be `i128` specific.

I will retitle that issue so that it is also more discoverable if anybody else gets the same error output.

carlos4242, 2018-02-08:
>  didn't emit the correct number of values (original source had a tuple return
>  value) (#89)

Cool. Sounds good. :)

https://github.com/avr-rust/rust/issues/89#issuecomment-364068243

closed 2018-02-08 by dylanmckay