avr-rust/rust-legacy-fork #98: Structs are passed by pointer in FFI?

mb64, 2018-04-25:
It seems to pass a `struct` via a pointer in an `extern "C"` function, instead of copying the value.

I'm not sure if this is a bug in the AVR LLVM backend, AVR Rust, `rustc` itself (the AVR branch is still on 1.22), or just if my expectations are just way off.

Here's a quick example:

```rust
#[derive(Copy,Clone)]
#[repr(C,packed)]
pub struct ContainsUsize {
    value: usize,
}

extern "C" {
    fn takes_wrapped(x: ContainsUsize);
    fn takes_usize(x: usize);
}

#[no_mangle]
pub extern "C" fn main() {
    let x = ContainsUsize {
        value: 82,
    };
    unsafe {
        takes_wrapped(x);
        takes_usize(x.value);
    }
}
```
This produces the following assembly, trimmed down to show the specific place and annotated:
```assembly
	; ... init stack, Y reg
	ldi	r16, 82	; r16r17 contains x
	ldi	r17, 0
	std	Y+1, r16	; So does *(Y+1)
	std	Y+2, r17
	movw	r24, r28
	adiw	r24, 1	; Passing Y+1
	call	takes_wrapped
	movw	r24, r16	; Passing r16r17
	call	takes_usize
	; ...
```

(As an aside, the corresponding C code produces what I would expect:
```C
#pragma pack
struct ContainsSizeT {
    size_t value;
};

void takes_wrapped(struct ContainsSizeT);
void takes_size_t(size_t);

void main() {
    struct ContainsSizeT x = {value: 22};
    takes_wrapped(x);
    takes_size_t(x.value);
}
```
This produces:
```assembly
	ldi r24,lo8(22)
	ldi r25,0
	rcall takes_wrapped
	ldi r24,lo8(22)
	ldi r25,0
	rjmp takes_size_t
```
Here `avr-gcc` passes both by value.)

shepmaster, 2018-04-26:
> the corresponding C code produces what I would expect

Can you try with Clang? That would help decide where to focus.

mb64, 2018-04-27:
So I built `clang` with AVR support (wow, it uses a lot of RAM), and it also passes it by pointer:
```assembly
main:
	ldi	r24, lo8(main.x) ; Passing address of main.x
	ldi	r25, hi8(main.x)
	call	takes_wrapped
	ldi	r24, 22 ; Passing the value
	ldi	r25, 0
	call	takes_size_t
	ret
main.x:
	.short	22
```

mb64, 2018-04-28:
Potentially bad news... I don't think it's "just" the LLVM backend.  I just took a look at the IR that Clang was generating, and it looks like it originates in the generated IR:

```C
struct T {
    int x;
};
extern void use_T(struct T);

void do_use_T(struct T x) {
    use_T(x);
}
```
`clang -S -O0 -emit-llvm --target=avr-unknown-unknown simple.c` gives the following (along with some attributes):
```llvm
%struct.T = type { i16 }

; Function Attrs: noinline nounwind optnone
define void @do_use_T(%struct.T* byval align 1) #0 {
  call void @use_T(%struct.T* byval align 1 %0)
  ret void
} 

declare void @use_T(%struct.T* byval align 1) #1
```
It doesn't do this with other targets.

Is there some writing-LLVM-IR library that both Clang and rustc use?  This would be too much of a coincidence otherwise.

shepmaster, 2018-04-29:
> Is there some writing-LLVM-IR library that both Clang and rustc use?

Not to my knowledge, or at least nothing that wouldn't be in LLVM itself.

> `%struct.T* byval align 1`

Isn't that saying it takes a *pointer* to a `struct T` by value, not the struct itself by value?

mb64, 2018-04-29:
Yeah... I would've expected it to just be `%struct.T` or similar.  Both Clang and rustc produce, on `x86_64-unknown-linux-gnu`:
```llvm
declare void @takes_wrapped(i64) unnamed_addr #2
```
Even with `-C opt-level=0` or `-O0`, both compilers just cast it to the correspondingly-sized integer type, still passing it by value.
```rust
#[repr(C,packed)]
pub struct Wrapped {
    value: i32,
    another: i32,
}

extern "C" {
    fn takes_wrapped(x: Wrapped);
}

#[no_mangle]
pub extern "C" fn main() {
    let x = Wrapped {
        value: 82,
        another: 45,
    };
    unsafe { takes_wrapped(x); }
}
```
Here's `main` on x86_64:
```llvm
define void @main() unnamed_addr #0 personality i32 (...)* @rust_eh_personality {
start:
  tail call void @takes_wrapped(i64 193273528402) ; (45 << 32) + 82
  ret void
}
```
And on AVR:
```llvm
define void @main() unnamed_addr #0 personality i32 (...)* @rust_eh_personality {
start:
  %arg = alloca { i32, i32 }, align 1
  %0 = getelementptr inbounds { i32, i32 }, { i32, i32 }* %arg, i16 0, i32 0
  store i32 82, i32* %0, align 1
  %1 = getelementptr inbounds { i32, i32 }, { i32, i32 }* %arg, i16 0, i32 1
  store i32 45, i32* %1, align 1
  call void @takes_wrapped({ i32, i32 }* noalias nocapture nonnull dereferenceable(8) %arg)
  ret void
}
```

mb64, 2018-04-29:
(Sorry, accidentally pushed "Close and comment" instead of "Comment")

shepmaster, 2018-04-29:
> Isn't that saying it takes a pointer to a struct T by value, not the struct itself by value?

Never mind me, my brain mixed up which behavior you were seeing vs what you wanted.

shepmaster, 2018-04-29:
Here's some random thoughts:

x86 might have some extra LLVM optimization passes — not sure where to check for that, or which passes (if any) are run before the LLVM IR is dumped.

Passing 64 bits by value on AVR would take 4 registers, while by-reference is only one — could check the function argument lowering code to see

`takes_wrapped` is an extern function, so the optimizer cannot do whatever it feels like, it has to adhere to the ABI — we might not have encoded the AVR C ABI rules correctly in LLVM.

leo60228, 2018-07-16:
Would this be easy to fix?