avr-rust/rust-legacy-fork #100: Unicode implementation uses out-of-range literals.

shepmaster, 2018-04-29:
Whatever this fix is, it can probably be upstreamed without the rest of AVR.

```
warning: literal out of range for usize
  --> /.../libcore/src/unicode/bool_trie.rs:48:23
   |
48 |         } else if c < 0x10000 {
   |                       ^^^^^^^
   |
   = note: #[warn(overflowing_literals)] on by default
   = note: the literal `0x10000` (decimal `65536`) does not fit into an `usize` and will become `0usize`
```

shepmaster, 2018-04-29:
The actual arrays are small, so this can probably be solved with some judicious sizes and casts.

kitling, 2018-05-01:
Seems like `c < 0x10000` would be equivilant to `c <= 0xFFFF`, which would fix the issue?

kitling, 2018-05-02:
So I actually took a look at the code now, and I understand a bit more what the actual issue is. Perhaps usize should just be changed to u32 (or u64?) instead of usize because of platforms where usize is less that u32.

shepmaster, 2018-05-02:
> When you say "compiler should just give a warning" do you mean something like "conditional will never be false"?

Oh, I was thinking that if the compiler could tell that:

1. It was a `<` comparison
1. The value was only out of range by 1

Then it could suggest changing to `<=`. It's a very silly thing, but Rust error messages are usually good. ;-)

shepmaster, 2018-05-02:
> because of platforms where usize is less that u32.

Like AVR, where it's 16 bit 😇 

That sounds like my original idea, which should be a reasonable fallback. I think your idea of adjusting the literal is a cleaner alternative if it works.

kitling, 2018-05-06:
That won't properly work because the full range of unicode requires 32 bits to express. There is an `else` statement after the `else if` we're working with. This really should work with 32bits at minimum.

shepmaster, 2018-05-06:
> This really should work with 32bits at minimum.

then yeah, changing from `usize` to `u32` and doing the needed casts at the right point makes the most sense.

dylanmckay, 2018-11-09:
I've had similar problems with using `bindgen` with AVR headers because `enum`s are represented as `usize = u16`, which is too small to cover a lot of discriminators.

crlf0710, 2019-04-14:
Now that `c` is converted to u32 instead of usize, i wonder if this issue can be closed now?
https://github.com/avr-rust/rust/blob/avr-support/src/libcore/unicode/bool_trie.rs#L45

closed 2019-04-14 by shepmaster