avr-rust/rust-legacy-fork #101: [LLVM 6.0] flt2dec causes assertion failed: Wrong topological sorting

shepmaster, 2018-05-03:
This is with my [recently-rebased branch](https://github.com/shepmaster/rust/tree/avr-support-rust-2018-04-24-190a6c) based on upstream Rust 190a6c. I've added the patch from #99 locally.

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-b213db3.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

; Function Attrs: uwtable
define align 1 void @core__num__flt2dec__strategy__dragon__mul_pow() unnamed_addr  {
start:
  %ret.i = alloca [40 x i32], align 1
  %0 = zext i32 undef to i64
  %1 = add nuw nsw i16 0, 11
  br label %bb19.i105.11

bb19.i105.11:                                     ; preds = %start
  %2 = getelementptr inbounds [40 x i32], [40 x i32]* %ret.i, i16 0, i16 %1
  %3 = load i32, i32* %2, align 1
  %4 = mul nuw i64 %0, 2788392729
  %5 = zext i32 %3 to i64
  %6 = add nuw nsw i64 0, %5
  %7 = add i64 %6, %4
  %8 = trunc i64 %7 to i32
  store i32 %8, i32* %2, align 1
  unreachable
}
```

```
Assertion failed: (Node2Index[SU.NodeNum] > Node2Index[PD.getSUnit()->NodeNum] && "Wrong topological sorting"), function InitDAGTopologicalSorting, file /Users/shep/Projects/avr-rust/src/llvm/lib/CodeGen/ScheduleDAG.cpp, line 518.
```

shepmaster, 2018-05-03:
/cc @brainlag

TimNN, 2018-10-08:
I've been looking into this issue a bit (with a slightly [different repro](https://gist.github.com/149b9e5b1967eb0a4d004c34ff73f728) since I'm using a newer LLVM version). The issue appears to be that whatever data is passed to `ScheduleDAGTopologicalSort`, is **not** a DAG. The example where the assertion happens contains a cycle.

Note that this seems to be related to optimizations. Compiling with `llc -O0` is fine, `llc -O1` is not.

TimNN, 2018-10-08:
In case that helps anyone, here is what I believe is the relevant part of the `-debug` output: https://gist.github.com/620098d70297404f3d02a8555363b916

The "LLVM ERROR: out of memory" seems easily explainable to me: LLVM apparently wants to compute the depth of the "DAG" for debug printing. Since the DAG contains a cycle, we get an infinite loop.

brainlag, 2018-10-08:
If you compile llvm with asserts you will hit that and don't run into OOM error.
I debugged this in the past and IIRC the problem is because 2 nodes in the DAG are glued together which lead to the cycle. If these 2 nodes would not glued together it would be fine.

TimNN, 2018-10-08:
> the problem is because 2 nodes in the DAG are glued together

I'm not quite sure what you mean with "glued together". In my case there is a cycle between three nodes (2, 3 and 6 in the Graph below).

<details><summary>Graph</summary>

[image]

</details>

brainlag, 2018-10-08:
There are glue nodes which glues 2 (or more?) nodes together

TimNN, 2018-10-09:
Ah, I wasn't aware that "glue nodes" are an LLVM concept. Anyway, I think I understand now what you meant and see it for my example as well. So, IIUC, the next step would probably be figuring out why the two nodes are glued together.

In any case, I'm including the relevant graphs below, in case they are useful to anyone:

<details><summary>ISel Input</summary>

[image]

</details>
<details><summary>Sched Input</summary>

[image]

</details>
<details><summary>Scheduled</summary>

[image]

</details>

TimNN, 2018-10-09:
Looking at some more graphs, I believe that the issue occurs during the `combine2` phase:

Before, we already have the glue node, but not yet a potential cycle. After `combine2` we now have `t40` and the potential cycle.

<details><summary>Before Combine 2</summary>

[image]

</details>

<details><summary>After Combine 2</summary>

[image]

</details>

TimNN, 2018-10-09:
I think I know now what happens: In `CombineToPostIndexedLoadStore`, LLVM checks if `t21` and `t35` can be combined.

This may only be done if

> Op must be independent of N, i.e. Op is neither a predecessor nor a successor of N. Otherwise, if Op is folded that would create a cycle.

The check for "!Op->isPredecessorOf(N) && !N->isPredecessorOf(Op)" then "correctly" identifies that they are independent, if one considers only the edges actually present in the graph.

The issue now is that the glue link between `t28` and `t29` will in effect be bidirectional (rather than unidirectional) once the two nodes are combined.

I can see three solutions to this:

1) This could be considered a bug in `CombineToPostIndexedLoadStore`, thus `CombineToPostIndexedLoadStore` should be adapted to handle this case.
2) Not generate the glue in the first place. (If it only exists for cosmetic / performance reasons, it could just be dropped. If it *must* be present, then I believe the glue would need to be replaced by a pseudo / virtual instruction).
3) Ignore the glue in the schedule when it would lead to a cycle.

Intuitively, I don't think that (1) is feasible and don't like (3) as a solution. I'll try to figure out why / how / where the glue is inserted in the first place.

TimNN, 2018-10-09:
The glue nodes are generated by LLVM in `DAGTypeLegalizer::ExpandIntRes_ADDSUB`. Given that, I don't believe that solution (2) is easily feasible anymore, since any changes here would affect much more than just AVR.

brainlag, 2018-10-09:
Imho, and i might be wrong, this is flaw in llvm. Either way I think you should discuss this with the llvm developers.

Also, when I looked up the code for `CombineToPostIndexedLoadStore` I found [this](https://github.com/llvm-mirror/llvm/commit/ea11fa660db4d3e54ca911c821f200ce766e98f4).

TimNN, 2018-10-09:
Yeah, I think you are correct that this can be considered an LLVM bug. I'll write to the mailing list or open a bug tomorrow.

> Also, when I looked up the code for `CombineToPostIndexedLoadStore` I found [this](https://github.com/llvm-mirror/llvm/commit/ea11fa660db4d3e54ca911c821f200ce766e98f4).

That commit should be a non-functional change. It just ensures that `Op->isPredecessorOf(N)` and `N->isPredecessorOf(Op)` don't do duplicate work.

TimNN, 2018-10-10:
llvm-dev thread (or rather, initial message): http://lists.llvm.org/pipermail/llvm-dev/2018-October/126851.html

TimNN, 2018-10-10:
Per the first reply on the ML, I've tried to disable the `(add|sub)(c|e)` instructions for AVR ([patch](https://github.com/TimNN/llvm/commit/1c45e300e51e48f690d5179959154c8b71cdbc21)) and that makes my reproduction compile. 

I don't know yet if that breaks something else. Worstcase, I hope that this only makes things slower, which could be fixed by adding support for the `(ADD|SUB)CARRY` instructions.

**Live-Edit:** I've just ran the AVR CodeGen tests and took a closer look at `add.ll`. As far as I can tell, the [generated assembly](https://gist.github.com/11945b85aa7dad25344d1c08854cba22) is _much_ worse (for comparison, the assembly [without the patch](https://gist.github.com/005be27708fe27853d7c298a30ef804b)).

shepmaster, 2018-10-10:
And someone has a [proposed fix for the glue](https://reviews.llvm.org/D53106)

shepmaster, 2018-11-03:
We've recently upgraded to LLVM 8 🎉 I'm going to close any bug that is reported against an older version of LLVM. If you are still having this issue with the LLVM-8 based code, **please** ping me and I can reopen the issue!

crlf0710, 2019-04-14:
Built the official 8.0 release code with AVR target enabled, llced the code in OP, and it seems `llc` falls into a dead loop.

dylanmckay, 2019-06-07:
The way forward here is to convert the AVR backend from the old `ADDE/ADDC/SUBE/SUBC` DAG opcodes to the `UADDO/ADDCARRY/USUBO/SUBCARRY` DAG opcodes.

https://reviews.llvm.org/D53106#1443059

> Summarizing offline discussion with jyknight for future reference:
>
> This should only be necessary due to glued nodes that are not Copy{To,From}Reg and which have additional predecessors/successors than nodes on the glue.
> This appear to only occur in the case of ADDC/ADDE nodes have been deprecated.
>
> This patch was originally written for such an instance in the AVR backend.

dylanmckay, 2019-06-07:
Hm, I can't reproduce this on LLVM trunk anymore, on either testcase.

```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-95356ba.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

%"num::bignum::Big32x40" = type { [0 x i8], i16, [0 x i8], [40 x i32], [0 x i8] }

; Function Attrs: uwtable
define void @"core::num::dec2flt::num::digits_to_big"() unnamed_addr {
start:
  %_7.sroa.0.0..sroa_idx.i = getelementptr inbounds %"num::bignum::Big32x40", %"num::bignum::Big32x40"* undef, i16 0, i32 3, i16 0
  switch i2 undef, label %bb5.i10 [
    i2 0, label %bb2.i
    i2 1, label %bb3.i
    i2 -2, label %bb4.i
  ]

bb2.i:                                            ; preds = %start
  ret void

bb3.i:                                            ; preds = %start
  %0 = add i8 0, -48
  %1 = zext i8 %0 to i32
  %2 = load i32, i32* %_7.sroa.0.0..sroa_idx.i, align 1
  %3 = call { i32, i1 } @llvm.uadd.with.overflow.i32(i32 %2, i32 %1)
  %4 = extractvalue { i32, i1 } %3, 0
  store i32 %4, i32* %_7.sroa.0.0..sroa_idx.i, align 1
  unreachable

bb4.i:                                            ; preds = %start
  unreachable

bb5.i10:                                          ; preds = %start
  unreachable
}

; Function Attrs: nounwind readnone speculatable
declare { i32, i1 } @llvm.uadd.with.overflow.i32(i32, i32)
```

### Using avr-rust at `avr-support-aa35e73b`

```bash
./llc -march=avr -o - -filetype=asm ~/projects/builds/llvm-project/llvm/foo.ll
```

Gives

```
        .text                                                                                                                                                                                 
        .file   "bugpoint-output-95356ba.bc"                                                                                                                                                  
llc: /home/dylan/projects/avr-rust/rust/src/llvm/lib/CodeGen/ScheduleDAG.cpp:507: void llvm::ScheduleDAGTopologicalSort::InitDAGTopologicalSorting(): Assertion `Node2Index[SU.NodeNum] > Node
2Index[PD.getSUnit()->NodeNum] && "Wrong topological sorting"' failed.                                                                                                                        
Stack dump:                                                                                                                                                                                   
0.      Program arguments: ./llc -march=avr -o - -filetype=asm /home/dylan/projects/builds/llvm-project/llvm/foo.ll                                                                           
1.      Running pass 'Function Pass Manager' on module '/home/dylan/projects/builds/llvm-project/llvm/foo.ll'.                                                                                
2.      Running pass 'AVR DAG->DAG Instruction Selection' on function '@"core::num::dec2flt::num::digits_to_big"'                                                                             
#0 0x00007f5841e1dd9a llvm::sys::PrintStackTrace(llvm::raw_ostream&) /home/dylan/projects/avr-rust/rust/src/llvm/lib/Support/Unix/Signals.inc:499:3                                           
#1 0x00007f5841e1c234 llvm::sys::RunSignalHandlers() /home/dylan/projects/avr-rust/rust/src/llvm/lib/Support/Signals.cpp:67:5                                                                 
#2 0x00007f5841e1c3b5 SignalHandler(int) /home/dylan/projects/avr-rust/rust/src/llvm/lib/Support/Unix/Signals.inc:348:31                                                                      
#3 0x00007f58414a44d0 __restore_rt (/usr/lib/libpthread.so.0+0x124d0)                                                                                                                         
#4 0x00007f584101482f __GI_raise (/usr/lib/libc.so.6+0x3782f)
#5 0x00007f5840fff672 __GI_abort (/usr/lib/libc.so.6+0x22672)
#6 0x00007f5840fff548 _nl_load_domain.cold.0 (/usr/lib/libc.so.6+0x22548)
#7 0x00007f584100cdb6 (/usr/lib/libc.so.6+0x2fdb6)
#8 0x00007f58422c3deb llvm::ScheduleDAGTopologicalSort::InitDAGTopologicalSorting() /home/dylan/projects/avr-rust/rust/src/llvm/lib/CodeGen/ScheduleDAG.cpp:506:7
#9 0x00007f58424b1d75 (anonymous namespace)::ScheduleDAGRRList::Schedule() /home/dylan/projects/avr-rust/rust/src/llvm/lib/CodeGen/SelectionDAG/ScheduleDAGRRList.cpp:371:3
[... 29 lines cut ...]
        ld      r24, Z
        ldd     r25, Z+1
        ldd     r18, Z+2
        ldd     r19, Z+3
        subi    r24, 48
        sbci    r25, 255
        sbci    r18, 255
        sbci    r19, 255
        st      Z, r24
        std     Z+1, r25
        std     Z+2, r18
        std     Z+3, r19
.Lfunc_end0:
        .size   "core::num::dec2flt::num::digits_to_big", .Lfunc_end0-"core::num::dec2flt::num::digits_to_big"
                                        ; -- End function

        ; Declaring this symbol tells the CRT that it should
        ;copy all variables from program memory to RAM on startup
        .globl  __do_copy_data
        ; Declaring this symbol tells the CRT that it should
        ;clear the zeroed data section on startup
        .globl  __do_clear_bss
```

So perhaps an LLVM upgrade will fix this?

dylanmckay, 2019-06-08:
I cherry-picked avr-rust/llvm-project@56d676b08b956051de3f781e7afd73365d62a7a9 and avr-rust/llvm-project@a5b647806b43a505da004f54e19b41476121c98b into the `avr-support-aa35e73b` branch (#137), and this test no longer fails.