avr-rust/rust-legacy-fork #102: [LLVM 6.0] LowerFormalArguments didn't emit the correct number of values

jhwgh1968, 2018-05-03:
Reduced test case:

```
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-7b39239.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

; Function Attrs: nounwind readnone uwtable
define { i64, i16 } @"_ZN59_$LT$f32$u20$as$u20$core..num..dec2flt..rawfp..RawFloat$GT$6unpack17h2977bc660656d2faE"(float %self) unnamed_addr #1 {
start:
  ret { i64, i16 } zeroinitializer
}
```

The full error message:
```
llc: /home/admin/Software/avr-rust/src/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:8736: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function&): Assertion `InVals.size() == Ins.size() && "LowerFormalArguments didn't emit the correct number of values!"' failed.
```

This is after applying #92 to @shepmaster's rebased branch.

cc @brainlag

brainlag, 2018-05-03:
#worksforme

You are missing a patch, probably [this one]( https://gist.github.com/brainlag/bfb4840a12b74b60557f34ed33aa9481)

Duplicate of #92.

shepmaster, 2018-05-03:
Yeah, that seemingly never got into LLVM, either... I'll see about pushing up some new versions with more of the recent patches applied.

jhwgh1968, 2018-05-04:
That is the only patch I *did* apply, actually. It got me past the original problem of running out of registers.

Having said that, I would not be surprised if I am missing something else. I guess I'll wait for @shepmaster to update his branch with all the known fixes across the issue tracker before I start digging for more bugs.

jhwgh1968, 2018-05-04:
I figured out there was a second patch for #92 that I did not apply, as suspected. Closing.

closed 2018-05-04 by jhwgh1968