avr-rust/rust-legacy-fork #104: [LLVM 6.0] Cannot getTypeInfo() on a type that is unsized

shepmaster, 2018-05-05:
```llvm
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-ec12b48.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

%"num::bignum::Big32x40" = type { [0 x i8], i16, [0 x i8], [40 x i32], [0 x i8] }

; Function Attrs: cold noinline noreturn uwtable
declare void @"core::slice::slice_index_len_fail"(i16, i16) unnamed_addr

; Function Attrs: uwtable
define { [0 x i32]*, i16 } @"core::num::bignum::Big32x40::digits"(%"num::bignum::Big32x40"* noalias readonly dereferenceable(162) %self) unnamed_addr  {
start:
  tail call void @"core::slice::slice_index_len_fail"(i16 undef, i16 40)
  unreachable
}
```

```
Assertion failed: (Ty->isSized() && "Cannot getTypeInfo() on a type that is unsized!"), function getTypeSizeInBits, file /Users/shep/Projects/avr-rust/src/llvm/include/llvm/IR/DataLayout.h, line 531.
0  libLLVM.dylib            0x000000010ee55ee8 llvm::sys::PrintStackTrace(llvm::raw_ostream&) + 40
1  libLLVM.dylib            0x000000010ee565f6 SignalHandler(int) + 694
2  libsystem_platform.dylib 0x00007fff7fb21f5a _sigtramp + 26
3  libsystem_platform.dylib 0x0001000000000002 _sigtramp + 2152587458
4  libsystem_c.dylib        0x00007fff7f8bf1ae abort + 127
5  libsystem_c.dylib        0x00007fff7f8871ac basename_r + 0
6  libLLVM.dylib            0x000000010eee3205 llvm::DataLayout::getTypeSizeInBits(llvm::Type*) const + 661
7  libLLVM.dylib            0x000000010f46be5c llvm::TargetLowering::LowerCallTo(llvm::TargetLowering::CallLoweringInfo&) const + 2220
8  libLLVM.dylib            0x000000010f4893e4 llvm::SelectionDAGBuilder::lowerInvokable(llvm::TargetLowering::CallLoweringInfo&, llvm::BasicBlock const*) + 836
9  libLLVM.dylib            0x000000010f474d55 llvm::SelectionDAGBuilder::LowerCallTo(llvm::ImmutableCallSite, llvm::SDValue, bool, llvm::BasicBlock const*) + 2005
10 libLLVM.dylib            0x000000010f45fd75 llvm::SelectionDAGBuilder::visitCall(llvm::CallInst const&) + 1685
11 libLLVM.dylib            0x000000010f454569 llvm::SelectionDAGBuilder::visit(llvm::Instruction const&) + 105
12 libLLVM.dylib            0x000000010f4ed223 llvm::SelectionDAGISel::SelectBasicBlock(llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>, false, true>, llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::Instruction, true, false, void>, false, true>, bool&) + 147
13 libLLVM.dylib            0x000000010f4ec2f7 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) + 9015
14 libLLVM.dylib            0x000000010f4e8e3b llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) + 2059
15 libLLVM.dylib            0x000000010f12e284 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) + 180
16 libLLVM.dylib            0x000000010ef7187d llvm::FPPassManager::runOnFunction(llvm::Function&) + 541
17 libLLVM.dylib            0x000000010ef71af3 llvm::FPPassManager::runOnModule(llvm::Module&) + 67
18 libLLVM.dylib            0x000000010ef7204b llvm::legacy::PassManagerImpl::run(llvm::Module&) + 971
19 llc                      0x000000010ed35c2f compileModule(char**, llvm::LLVMContext&) + 10239
20 llc                      0x000000010ed3321b main + 1275
21 libdyld.dylib            0x00007fff7f813015 start + 1
Stack dump:
0.	Program arguments: llc -march=avr -mcpu=atmega328p -filetype=obj ./a-type-that-is-unsized.2.ll
1.	Running pass 'Function Pass Manager' on module './a-type-that-is-unsized.2.ll'.
2.	Running pass 'AVR DAG->DAG Instruction Selection' on function '@"core::num::bignum::Big32x40::digits"'
```

shepmaster, 2018-05-05:
/cc @brainlag

jhwgh1968, 2018-05-28:
I managed to reproduce this after applying various patches and commenting a bunch of things out. I got it in a different place, though, which seemed to be something related to ``Option<T>``.

I am commenting primarily to point out: this issue is much more of a blocker for LLVM 6 than most (maybe all?) of the rest. Many have mitigations, and the rest have been in functions that we can disable and still have a minimally functional libcore. But I am not sure if we can call it Rust anymore if we have to disable everything that requires ``&[T]`` and ``Option<T>``! :smile: 

If anyone with LLVM expertise has time to look at this project, I think this is the highest priority issue that remains completely unmitigated. (Perhaps @shepmaster could add a tag to reflect this if he agrees.)

brainlag, 2018-05-28:
Imho all bugs are more or less equal important. Because a bug is only hit by a unimportant libcore function  doesn't mean user code will not hit it all the time.

shepmaster, 2018-05-29:
> I got it in a different place, though, which seemed to be something related to `Option<T>`.

The bugs that we surface happen in arbitrary locations, generally whatever was compiled "first" (which is complicated by multicore builds and other sources of non-determinism). Then we use tools like bugpoint which don't find *all* cases of a given error, just *one* of them. 

All that's to say that I agree with @brainlag — I feel that all bugs are equivalent at this point.

> if we have to disable everything that requires `&[T]` and `Option<T>`

I'm a little surprised here because I thought I'd compiled my blink program with the LLVM 6 branch (which I never pushed to master, I see).

jhwgh1968, 2018-05-30:
Points taken, @shepmaster and @brainlag. Do either of you think there is value in reconstructing and posting my alternative bugpoint, or is the existing one sufficient?

shepmaster, 2018-05-30:
> posting my alternative bugpoint

If you had it laying around, I'd probably suggest posting it. There's the possibility that a different variant is smaller, more obvious to diagnose, or has a completely different root cause. I don't know if I'd say you should spend time on making it again though.

dylanmckay, 2018-06-12:
I tried this testcase with LLVM trunk and it compiled fine. I'm building an avr-rust toolchain to try it there now.

```bash
dylan@dylan-pc:~/p/b/l/llvm:> cat issue-104.ll                                                                                                                               15:58:50
; ModuleID = 'bugpoint-reduced-simplified.bc'
source_filename = "bugpoint-output-ec12b48.bc"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

%"num::bignum::Big32x40" = type { [0 x i8], i16, [0 x i8], [40 x i32], [0 x i8] }

; Function Attrs: cold noinline noreturn uwtable
declare void @"core::slice::slice_index_len_fail"(i16, i16) unnamed_addr

; Function Attrs: uwtable
define { [0 x i32]*, i16 } @"core::num::bignum::Big32x40::digits"(%"num::bignum::Big32x40"* noalias readonly dereferenceable(162) %self) unnamed_addr  {
start:
  tail call void @"core::slice::slice_index_len_fail"(i16 undef, i16 40)
  unreachable
}
```

brainlag, 2018-06-12:
Yeah I couldn't reproduce it either with LLVM trunk a month ago.

dylanmckay, 2018-06-12:
It compiled fine with the avr-rust/llvm repository too

```bash
./bin/llc -march=avr -mcpu=atmega328 ~/projects/builds/llvm-project/llvm/issue-104.ll  -o -
```

```asm
        .text
        .file   "/home/dylan/projects/builds/llvm-project/llvm/issue-104.ll"
        .globl  "core::num::bignum::Big32x40::digits"
        .p2align        1
        .type   "core::num::bignum::Big32x40::digits",@function
"core::num::bignum::Big32x40::digits":  ; @"core::num::bignum::Big32x40::digits"
; BB#0:                                 ; %start
        ldi     r22, 40
        ldi     r23, 0
        call    core::slice::slice_index_len_fail
.Lfunc_end0:
        .size   "core::num::bignum::Big32x40::digits", .Lfunc_end0-"core::num::bignum::Big32x40::digits"

        ; Declaring this symbol tells the CRT that it should
        ;copy all variables from program memory to RAM on startup
        .globl  __do_copy_data
        ; Declaring this symbol tells the CRT that it should
        ;clear the zeroed data section on startup
        .globl  __do_clear_bss
```

leo60228, 2018-07-16:
Could this be fixed by updating the LLVM fork?

brainlag, 2018-07-16:
yes

shepmaster, 2018-11-03:
We've recently upgraded to LLVM 8 🎉 I'm going to close any bug that is reported against an older version of LLVM. If you are still having this issue with the LLVM-8 based code, **please** ping me and I can reopen the issue!

closed 2018-11-03 by shepmaster