avr-rust/rust-legacy-fork #106: LLVM ERROR: expected relocatable expression from fairly simple LLVM IR

carlos4242, 2018-07-02:
The attached llvm ir is the minimum test case I've found for seeing this error occur when attempting to compile object files for AVR using llc.

[cause-relocation-error.ll.txt](https://github.com/avr-rust/rust/files/2157130/cause-relocation-error.ll.txt)

Note that it's possible to create assembly files but these then do not compile themselves using llvm-mc.

Attached are the assembly code generated using the AVR backend...
[cause-relocation-error.avr.s.txt](https://github.com/avr-rust/rust/files/2157129/cause-relocation-error.avr.s.txt)

using the command
`llc -O3 -march=avr -mcpu=atmega328p -filetype=asm -o cause-relocation-error-avr.s cause-relocation-error.ll`

...and the X86 backend...
[cause-relocation-error.x86.s.txt](https://github.com/avr-rust/rust/files/2157131/cause-relocation-error.x86.s.txt)

using the command
`llc -O3 -march=x86-64 -filetype=asm -o cause-relocation-error-x86.s cause-relocation-error.ll`

If you compile the x86 assembly it works fine and produces an object file, with relocatables in it.

If you compile the AVR assembly you get these errors...

cause-relocation-error-avr.s:44:2: error: expected relocatable expression
        .long   (__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)
        ^
cause-relocation-error-avr.s:45:2: error: expected relocatable expression
        .long   (__unnamed_5&65535)-((_TMRfV4main10Brightness&65535)+20)

(using the command `llvm-mc -arch=avr -mcpu=atmega328p -filetype=obj -o cause-relocation-error-avr.o cause-relocation-error-avr.s`)

Likewise, compiling direct to object files produces the same results. With the x86 backend this...
`llc -O3 -march=x86-64 -filetype=obj -o cause-relocation-error-x86.o cause-relocation-error.ll`
...works fine.

But...
`llc -O3 -march=avr -mcpu=atmega328p -filetype=obj -o cause-relocation-error-avr.o cause-relocation-error.ll`
...crashes with just...
LLVM ERROR: expected relocatable expression

It looks like there's something about the machine code being generated that's not valid.

As a last throw of the dice, I tried to use GCC to compile the assembly output by llc...
`avr-gcc  -mmcu=atmega328p -o cause-relocation-error-avr.o cause-relocation-error-avr.s`

This gave alternative errors...

Carls-MacBook-Air:avr carlpeto$ avr-gcc  -mmcu=atmega328p -o min-2-avr.o min-2-avr.s
min-2-avr.s: Assembler messages:
min-2-avr.s:44: Error: invalid operands (__TEXT,__swift3_typeref, regular, no_dead_strip and *ABS* sections) for `&'
min-2-avr.s:44: Error: invalid operands (__TEXT,__swift3_fieldmd, regular, no_dead_strip and *ABS* sections) for `&'
min-2-avr.s:45: Error: invalid operands (__TEXT,__swift3_reflstr, regular, no_dead_strip and *ABS* sections) for `&'
min-2-avr.s:45: Error: invalid operands (__TEXT,__swift3_fieldmd, regular, no_dead_strip and *ABS* sections) for `&'

carlos4242, 2018-07-03:
Also, if I change the assembly directly to remove the odd looking &65535 it compiles.

`.long (__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)`

causes the error but

`.long __unnamed_4-(_TMRfV4main10Brightness+16)` is fine.

I don't know if this semantically changes the output but I can't see how it would as all addresses must be less than 65535 anyway?

dylanmckay, 2018-07-10:
Good catch.

This will be a bug in the AVR fixup resolution code [somewhere around here](
https://github.com/llvm-mirror/llvm/blob/master/lib/Target/AVR/MCTargetDesc/AVRMCExpr.cpp#L72).

We've got custom logic to handle stuff like `lo8(symbol)`, `-hi8(symbol)`. It looks like that logic probably doesn't work with `(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)`.

I know that you can perform pointer subtraction or addition in relocations, but I didn't know you could also perform `&`.

> I don't know if this semantically changes the output but I can't see how it would as all addresses must be less than 65535 anyway?

I agree, all pointers are 16-bits and so `<any pointer> & 0xffff` is always redundant.

dylanmckay, 2018-07-10:
How the fixup looks with different backends.

# AVR

```bash
llc -march=avr -mcpu=atmega328p cause-relocation-error.ll -filetype=obj -debug 2>&1|grep unnamed_4
        Fixups:[<MCFixup Offset:0 Value:(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16) Kind:2>]>,
           (__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16) Kind:2>]>,
           (__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16) Kind:2>]>,
           (__unnamed_4, Index:0, ),
```

# X86

```bash
llc cause-relocation-error.ll -filetype=obj -debug 2>&1|grep unnamed_4                                              
        Fixups:[<MCFixup Offset:0 Value:.L__unnamed_4-(_TMRfV4main10Brightness+16) Kind:2>]>,
           (.L__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:.L__unnamed_4-(_TMRfV4main10Brightness+16) Kind:2>]>,
           (.L__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:.L__unnamed_4-(_TMRfV4main10Brightness+16) Kind:2>]>,
           (.L__unnamed_4, Index:0, ),
```

# Mips

```bash
llc -march=mips cause-relocation-error.ll -filetype=obj -debug 2>&1|grep unnamed_4                                 
        Fixups:[<MCFixup Offset:0 Value:(($__unnamed_4)&4294967295)-((_TMRfV4main10Brightness&4294967295)+16) Kind:2>]>,
           ($__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:(($__unnamed_4)&4294967295)-((_TMRfV4main10Brightness&4294967295)+16) Kind:2>]>,
           ($__unnamed_4, Index:0, ),
        Fixups:[<MCFixup Offset:0 Value:(($__unnamed_4)&4294967295)-((_TMRfV4main10Brightness&4294967295)+16) Kind:2>]>,
           ($__unnamed_4, Index:0, ),
```

Both AVR and Mips are masking the pointers, but X86 is not.

carlos4242, 2018-07-10:
I guess my first instinct was that the masking looked weird, and maybe we should remove whatever was lowering the mixups into this odd looking masked format. But MIPS is a mature platform and is doing it too, so that leads me to think we should look at fixing the "assembler" part of the back end that is interpreting these fixups. (p.s. What is the difference between a "fixup" versus a "relocation"?)

carlos4242, 2018-07-13:
Interestingly, this might run a little deeper than I'd hoped. I just did a long debug session on llvm-mc to watch it attempting to handle the fixup ```.long	(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)```. I'm no longer confident that llvm can handle this for any target architecture, I think this might be a bigger problem. Whatever is producing this llvm IR (some apple internals) might be relying on some aspect of the x86 and arm64/32 platforms that it has never needed to worry about before now.

Stepping through... ```MCAssembler::evaluateFixup``` calls into ```MCExpr::evaluateAsRelocatableImpl``` (this is all in the main LLVM code, not yet in the AVR target specific code). ```(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)``` exists in memory as an MCExpr of type Binary with opCode MCBinaryExpr::Sub, the LHS is itself an MCExpr of type Binary with opCode MCBinaryExpr::And, the RHS is an MCExpr of type Binary with opCode MCBinaryExpr::Add... etc.

The first thing the code for evaluating the binary does is...
````
    if (!ABE->getLHS()->evaluateAsRelocatableImpl(LHSValue, Asm, Layout, Fixup,
                                                  Addrs, InSet) ||
        !ABE->getRHS()->evaluateAsRelocatableImpl(RHSValue, Asm, Layout, Fixup,
                                                  Addrs, InSet))
      return false;
````
...which iterates into the LHS and RHS, evaluating them.

When it goes into evaluate the MCBinaryExpr::And operation, it ultimately hits this block...

````
    // We only support a few operations on non-constant expressions, handle
    // those first.
    if (!LHSValue.isAbsolute() || !RHSValue.isAbsolute()) {
      switch (ABE->getOpcode()) {
      default:
        return false;
      case MCBinaryExpr::Sub:
        // Negate RHS and add.
        // The cast avoids undefined behavior if the constant is INT64_MIN.
        return EvaluateSymbolicAdd(Asm, Layout, Addrs, InSet, LHSValue,
                                   RHSValue.getSymB(), RHSValue.getSymA(),
                                   -(uint64_t)RHSValue.getConstant(), Res);

      case MCBinaryExpr::Add:
        return EvaluateSymbolicAdd(Asm, Layout, Addrs, InSet, LHSValue,
                                   RHSValue.getSymA(), RHSValue.getSymB(),
                                   RHSValue.getConstant(), Res);
      }
    }
````

And this returns false. So it seems the code is not willing to evaluate ```__unnamed_4&65535``` or any other & operation between a symbol and a mask. So the processing craps out and the error is printed.

I think the only "fix" is either some serious rework in core LLVM (no thanks!) or try to find out what upstream code is causing this elaborate fixup to be generated. There must be something about the lowering from LLVM IR into assembly that's creating this. Interestingly, as Dylan has proved, it's not specific to AVR as he's seeing something similar with MIPS.

leo60228, 2018-07-16:
So, does this mean it's a problem with upstream and not AVR-Rust?

brainlag, 2018-07-16:
as far as I understand it it is a swift frontend problem

carlos4242, 2018-07-16:
I'm not quite sure how to call it. I think the best way to describe it is "not a bug in LLVM", either avr-rust or upstream.

As far as I can figure out, I'm using a front end that is outputting for a target triple with a data layout involving 64 bit pointers. That produces LLVM IR like...

```
i32 trunc
(i64 sub
(i64 ptrtoint ([9 x i8]* @3 to i64),
i64 add (i64 ptrtoint (<{ i32, i16, i16, i32, i32, i32, i32 }>* @_TMRfV4main10Brightness to i64),
i64 16)
) to i32)
```

In my toolchain the LLVM IR is then read into my LLVM AVR build tools, which have the correct target triple and data layout for AVR.

When AsmPrinter tries to lower to assembly/machine code, it recognises that the pointers are too big and masks them off. It creates an assembly expression like ```(__unnamed_4&65535)-((_TMRfV4main10Brightness&65535)+16)```. These expressions, though convoluted, are actually fine in most places, the assembler can work with them in many parts of assembly but not as fixups, hence the error.

I considered trying to patch upstream to allow these as fixups but (i) it might be a large change and not popular, given that the rationale for doing it is to fix a broken situation (above) and it shouldn't occur normally and (ii) in the bigger picture I have a vague feeling that fixups can become relocations that are written into an object file and that might not be legal or possible if they're in this format. Plus (iii) it's way outside my limited understanding of LLVM for sure!

I'm going to mark this as closed, unless anyone objects?

closed 2018-07-16 by carlos4242