avr-rust/rust-legacy-fork #107: Compile error originating from LLVM.

Debily, 2018-07-06:
There is a compile error (type mismatch) originating from llvm.
I am not familiar with llvm-internals, so I don't know how to patch this error.

Here is the error message:

```
In file included from rust-avr/src/llvm/tools/lli/lli.cpp:30:
rust-avr/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h: In member function 'llvm::Expected<std::vector<char> > llvm::orc::remote::OrcRemoteTargetClient<ChannelT>::readMem(char*, llvm::JITTargetAddress, uint64_t)':
rust-avr/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h:696:26: error: could not convert '((llvm::orc::remote::OrcRemoteTargetClient<ChannelT>*)this)->callB<llvm::orc::remote::OrcRemoteTargetRPCAPI::ReadMem>(Src, Size)' from 'Expected<vector<unsigned char,allocator<unsigned char>>>' to 'Expected<vector<char,allocator<char>>>'
     return callB<ReadMem>(Src, Size);
            ~~~~~~~~~~~~~~^~~~~~~~~~~
```

dylanmckay, 2018-07-10:
This is a weird issue - the code here is unrelated to AVR.

dylanmckay, 2018-07-19:
A duplicate issue was raised at #108.

I'll close this issue because that one has more details.

Thanks for the report @Debily!

closed 2018-07-19 by dylanmckay