avr-rust/rust-legacy-fork #108: Can not build llvm: OrcRemoteTargetClient.h:696:26: error: could not convert '((llvm::orc::remote::OrcRemoteTargetClient<ChannelT>*)this)->callB<llvm::orc::remote::OrcRemoteTargetRPCAPI::ReadMem>(Src, Size)'

BloodyWulf, 2018-07-17:
Hey, i'm interested in using Rust on AVR so i wanted to give it a try.

But unfortunately i can't build the project.
I'm using Arch Linux and this software versions are installed:
```text
make 4.2.1
cmake 3.11.4
ninja 1.8.2

gcc (GCC) 8.1.1 20180531
clang version 6.0.1 (tags/RELEASE_601/final)

Python 3.6.6
Python 2.7.15

rustc 1.27.1 (5f2b325f6 2018-07-07) (default)
rustc 1.29.0-nightly (1ecf6929d 2018-07-16)
cargo 1.27.0 (1e95190e5 2018-05-27)

```

What i've done so far:

```text
git clone https://github.com/avr-rust/rust.git

cd rust

./x.py build
```

And than the build failed (at around 88% of progress):
```text
In file included from [...]/rust/src/llvm/tools/lli/lli.cpp:30:
[...]/rust/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h: In member function 'llvm::Expected<std::vector<char> > llvm::orc::remote::OrcRemoteTargetClient<ChannelT>::readMem(char*, llvm::JITTargetAddress, uint64_t)':

[...]/rust/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h:696:26: error: could not convert
	'((llvm::orc::remote::OrcRemoteTargetClient<ChannelT>*)this)->callB<llvm::orc::remote::OrcRemoteTargetRPCAPI::ReadMem>(Src, Size)'
   from 'Expected<vector<unsigned char,allocator<unsigned char>>>'
     to 'Expected<vector<char,allocator<char>>>'

     return callB<ReadMem>(Src, Size);
            ~~~~~~~~~~~~~~^~~~~~~~~~~
```

Can someone help me please?

dylanmckay, 2018-07-19:
Could you try this

```bash
cd rust/src/llvm
git show
```

And see what version of LLVM you're on.

BloodyWulf, 2018-07-19:
Hey Dylan, thank you for your response.

```text
cd rust/src/llvm
git show

commit 3d84391d86e0d8eece971ad15b1438fc4a816b29 (HEAD -> avr-support, origin/avr-support, origin/HEAD)
Author: Dylan McKay <me@dylanmckay.io>
Date:   Tue Jun 12 03:53:09 2018 +1200

    [AVR] Cherry-pick the fix for #99

    This cherry-picks LLVM commit r334409.

    https://llvm.org/viewvc/llvm-project?view=revision&revision=334409

diff --git a/src/llvm b/src/llvm
index 0149e8bd35..cad3f9b46c 160000
--- a/src/llvm
+++ b/src/llvm
@@ -1 +1 @@
-Subproject commit 0149e8bd35f09463404affe879aac2e1466a6bd6
+Subproject commit cad3f9b46c6969a3dbfc12021a2e95d49b372654
```

I was able to build it after i changed the function signature from 'char' to
'unsigned char' but i don't know if that's the way to go.

I can imagine 2 reasons why this happen.

First reason:
The scripts are checking out a wrong version of llvm on my machine (don't know why this should happen).

Second reason:
Compilers have a switch how the datatype 'char' is treated. Maybe there is a global flag
on your machine that is set to 'treat char as unsigned char'.

That's the only reasons i can imagine why it build for you, but not for me.
But i think reason one is the one which make sense.

And at the moment i can't checkout the whole project.
```text
$ ./x.py

Updating submodules
Submodule 'book' (https://github.com/rust-lang/book.git) registered for path 'src/doc/book'
[...]
Submodule 'src/tools/rustfmt' (https://github.com/rust-lang-nursery/rustfmt.git) registered for path 'src/tools/rustfmt'

Cloning into '[...]/rust/src/doc/book'...
[...]
Cloning into '[...]/rust/src/tools/rustfmt'...

Submodule path 'src/doc/book': checked out 'd09c9e8144ed32170b7596abb145ade8b097acaf'
Submodule path 'src/doc/nomicon': checked out 'a4322ccb289a43cc238d4536982f184a3eec9ba7'
Submodule path 'src/doc/reference': checked out '266d429a48468371d2d90669f6a30dd659bb4bdb'
Submodule path 'src/jemalloc': checked out '1f5a28755e301ac581e2048011e4e0ff3da482ef'
Submodule path 'src/libcompiler_builtins': checked out 'ef4951582f620c589cd9e18ec182538bf116bce3'
Submodule 'compiler-rt/compiler-rt-cdylib/compiler-rt' (https://github.com/rust-lang/compiler-rt) registered for path 'src/libcompiler_builtins/compiler-rt'

Cloning into '[...]/rust/src/libcompiler_builtins/compiler-rt'...

Submodule path 'src/libcompiler_builtins/compiler-rt': checked out 'c8a8767c56ad3d3f4eb45c87b95026936fb9aa35'
Submodule path 'src/liblibc': checked out '136bf195dfa546a3589f414c1dfe693aa6ff1018'

error: Server does not allow request for unadvertised object ec492397b689f18ae29e6b852dd501dc04c6e164

Fetched in submodule path 'src/llvm', but it did not contain ec492397b689f18ae29e6b852dd501dc04c6e164. Direct fetching of that commit failed.
failed to run: git submodule update --init --recursive src/llvm src/rt/hoedown src/jemalloc src/tools/rust-installer src/liblibc src/doc/nomicon src/tools/cargo src/doc/reference src/doc/book src/tools/rls src/libcompiler_builtins src/tools/clippy src/tools/rustfmt src/tools/miri
```

dylanmckay, 2018-07-25:
I've recently setup a Docker container that runs AVR-Rust builds on Jenkins [here](http://jenkins.dylanmckay.io/job/avr-rust/job/rust/). This is using the [jenkins](https://hub.docker.com/r/jenkins/jenkins/) dockerfile, which is probably just ubuntu.

It compiles fine in both that environment, and my local environment. I too am surprised by the char signedness issue. Could be different `clang` versions on our machine maybe.

> And at the moment i can't checkout the whole project.

That was my bad! I cherry-picked a fix to avr-rust/rust but I forgot to push it to avr-rust/llvm. I noticed it a few days ago and fixed it.

jhwgh1968, 2018-08-01:
I just tried to build the same commit numbers you had, @dylanmckay, and I got the same error @BloodyWulf did.

However, I recognize this error from another project entirely. I also use Arch Linux, but have LLVM libraries tracked from upstream on my machine for other reasons (currently sitting at r337240). About 3-4 months ago, I started having to patch this error out of other unrelated projects I linked against it. I think someone is actually changing the API.

I don't know why that would affect a 4.0.1 branch + some cherry-picks, though.

FuzzyLitchi, 2018-08-22:
```
rust/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h:696:26: error: could not convert ‘((llvm::orc::remote::OrcRemoteTargetClient<ChannelT>*)this)->callB<llvm::orc::remote::OrcRemoteTargetRPCAPI::ReadMem>(Src, Size)’ from ‘Expected<vector<unsigned char,allocator<unsigned char>>>’ to  Expected<vector<char,allocator<char>>>’
     return callB<ReadMem>(Src, Size);
```
I tried building it too, I get the same error.
I _also_ use Arch Linux
```
GNU Make 4.2.1
cmake version 3.11.4
clang version 6.0.1 (tags/RELEASE_601/final)
gcc (GCC) 8.2.0
```
Rust commit 9f8c381a275b9cf6927ea89f915c7c60b9f54449
LLVM commit avr-rust/llvm@ec492397b689f18ae29e6b852dd501dc04c6e164

BloodyWulf, 2018-09-06:
I tried around a bit the last weeks, and it was a bit frustrating.
For the moment i patch it for my self, and it works.

What i've done to get it work:

```text
git clone https://github.com/avr-rust/rust.git

cd rust
./x.py
```

After that all sources are downloaded. Then i patched the file
```text
rust/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h
```
in line 690. Change the return type from char to unsigned char.

Befor patch:
```text
Expected<std::vector<char>> readMem(char *Dst, JITTargetAddress Src,
                                      uint64_t Size) {
    // Check for an 'out-of-band' error, e.g. from an MM destructor.
    if (ExistingError)
      return std::move(ExistingError);

    return callB<ReadMem>(Src, Size);
  }
```

After patch:
```text
Expected<std::vector<unsigned char>> readMem(char *Dst, JITTargetAddress Src,
                                      uint64_t Size) {
    // Check for an 'out-of-band' error, e.g. from an MM destructor.
    if (ExistingError)
      return std::move(ExistingError);

    return callB<ReadMem>(Src, Size);
  }
```

After that i copied the file config.toml.example to config.toml. There are a few
things that must be changed (as far as i get it) in config.toml.

In the [build] section:
```text
# Indicate whether submodules are managed and updated automatically.
#submodules = true
```

 must be changed to 
```text
submodules = false
```
so the edited file will not be updated.

In [install] section i set the prefix where the toolchain shall be installed.
I changed it to (just for showing what i mean):
```text
prefix = "~/devel/avr-rust/toolchain"
```

Now it should compile.

I copied the 2 edited files to a new location, so i can copy them back when i
make a new git clone (example on an other computer). My script for doing that:

```text
#!/bin/bash

rustup toolchain uninstall avr

rm -rfv rust/
rm -rfv toolchain/

git clone https://github.com/avr-rust/rust.git

cd rust
./x.py

cd ..
mkdir -v toolchain
cp -v OrcRemoteTargetClient.h rust/src/llvm/include/llvm/ExecutionEngine/Orc/OrcRemoteTargetClient.h
cp -v config.toml rust/config.toml

cd rust
./x.py build
./x.py dist
./x.py install

cd ..
mkdir -v ~/devel/avr-rust/toolchain/lib/rustlib/x86_64-unknown-linux-gnu/bin
rustup toolchain link avr ~/devel/avr-rust/toolchain
```

Filesystem befor executing my buildscript:
```text
~/devel/avr-rust/
                |
                +-> config.toml
                +-> OrcRemoteTargetClient.h
                +-> my_buildscript.sh
```

Filesystem after executing my buildscript:
```text
~/devel/avr-rust/
                |
                +-> config.toml
                +-> OrcRemoteTargetClient.h
                +-> my_buildscript.sh
                +-> rust/
                |       |
                |       +-> ...
                |
                +-> toolchain/
                        |
                        +-> ...
```

After that all i checked out the blink example. To build the example
there are 2 more lines that need to be executed:
```text
export XARGO_RUST_SRC=~/devel/avr-rust/toolchain/bin/rust
export RUST_TARGET_PATH=`pwd`
```
The 1st line can be added to .bashrc. The 2nd line i have to execute
in the blink example folder.

Now the blink example should compile.

I created an own project to play around with an Teensy++ 2.0 (AT90USB1286)
and it work. At the beginning there was a bit of struggeling with interrupts,
but now  it works fine.

But for my taste i have to do to much manually (what is caused by the
build error). We should look how we can fix that. After fixing the build
the project will be more interessting for other people that plays around in
the avr-world. (There is a big avr community in germany, where are also one
maintainer of avr-gcc and avrdude.)

The README.md should also be updated, because there is the build description
with configure and make and as far as i got it from other issues that
workflow is outdated.

I hope this helps others to get avr-rust working so they can enjoy rust on avr.

I can say that avr-rust is an awesome project. Thank you @dylanmckay
for all your work!