avr-rust/rust-legacy-fork #110: Failed to load source for a dependency on `num_cpus`

rivertam, 2018-09-16:
So I'm just going through the installation process and I got the following error:

```
info: looks like you are running this command under `sudo`
      and so in order to preserve your $HOME this will now
      use vendored sources by default. Note that if this
      does not work you should run a normal build first
      before running a command like `sudo make install`
Updating submodules
error: failed to load source for a dependency on `num_cpus`

Caused by:
  Unable to update registry https://github.com/rust-lang/crates.io-index

Caused by:
  failed to update replaced source `registry https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to read root of directory source: /home/ben/rust/src/vendor

Caused by:
  No such file or directory (os error 2)
failed to run: /home/ben/rust/build/build/x86_64-unknown-linux-gnu/stage0/bin/cargo build --manifest-path /home/ben/rust/
src/bootstrap/Cargo.toml --frozen
Build completed unsuccessfully in 0:00:03
Makefile:75: recipe for target 'install' failed
make: *** [install] Error 1
```

I'm not sure what information I can give otherwise, but I'm on Ubuntu 16.04 rustc 1.30.0-nightly (2ab3eba30 2018-09-14)

dylanmckay, 2018-09-23:
The commit you mentioned is very recent (`2ab3eba`) - are you currently using the LLVM 7.0 branch at #109?

This error is getting hit with the stage0 (precompiled snapshot from internet) compiler, which does not contain anything related to AVR.

```
No such file or directory (os error 2)
failed to run: /home/ben/rust/build/build/x86_64-unknown-linux-gnu/stage0/bin/cargo build --manifest-path /home/ben/rust/
```

Also, have you tried to compile rust under a regular user? I haven't seen this warning before.

```
info: looks like you are running this command under `sudo`
      and so in order to preserve your $HOME this will now
      use vendored sources by default. Note that if this
      does not work you should run a normal build first
      before running a command like `sudo make install`
```

shepmaster, 2018-11-03:
We've recently upgraded to LLVM 8 🎉 I'm going to close any bug that is reported against an older version of LLVM. If you are still having this issue with the LLVM-8 based code, **please** ping me and I can reopen the issue!

closed 2018-11-03 by shepmaster