avr-rust/rust-legacy-fork #113: Compiling the toolchain fails

eloydegen, 2018-10-15:
When I run `make`, it fails after a while with the following error:

```$ make
Updating submodules
Traceback (most recent call last):
  File "/rust/rust/src/bootstrap/bootstrap.py", line 773, in <module>
    main()
  File "/rust/rust/src/bootstrap/bootstrap.py", line 756, in main
    bootstrap()
  File "/rust/rust/src/bootstrap/bootstrap.py", line 734, in bootstrap
    build.download_stage0()
  File "/rust/rust/src/bootstrap/bootstrap.py", line 344, in download_stage0
    self._download_stage0_helper(filename, pattern)
  File "/rust/rust/src/bootstrap/bootstrap.py", line 377, in _download_stage0_helper
    get("{}/{}".format(url, filename), tarball, verbose=self.verbose)
  File "/rust/rust/src/bootstrap/bootstrap.py", line 36, in get
    download(sha_path, sha_url, False, verbose)
  File "/rust/rust/src/bootstrap/bootstrap.py", line 69, in download
    _download(path, url, probably_big, verbose, True)
  File "/rust/rust/src/bootstrap/bootstrap.py", line 93, in _download
    exception=exception)
  File "/rust/rust/src/bootstrap/bootstrap.py", line 143, in run
    ret = subprocess.Popen(args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1025, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 1
```

fluxchief, 2018-10-16:
My guess is that you don't have curl installed.

eloydegen, 2018-10-16:
That is the problem, thank you.

closed 2018-11-03 by shepmaster