avr-rust/rust-legacy-fork #118: Support for rust 2018 edition

peacememories, 2018-11-17:
I recently tried adding the new futures-preview 0.3-alpha.9 to my avr rust project, and now my build fails because the avr rustc does not recognize the `--edition` option. What would be needed for avr-rust/rust to merge support for the editions flag (and the 2018 edition, too) from rust/rust?
I'm assuming this would be non-trivial to say the least, but I thought I'd ask regardless, because I'd really like to create a futures runtime for avr :)

TimNN, 2018-11-17:
When did you last rebuild your local avr compiler? My current local build (which should approximately match the `avr-support` branch in this repo) definitely has the `--edition` flag:

```sh
 $ rustc +avr --help -v | grep edition                                                                                                                               
        --edition 2015|2018
                        Specify which edition of the compiler to use when
```

peacememories, 2018-11-17:
Oh nice, I've built it about a month ago, and I'm currently building now, but I searched for `edition` in the whole repo and came up blank, so I assumed it wasn't merged yet. I'll see if everything works once the build finishes. Thanks for the quick response :)

TimNN, 2018-11-17:
> but I searched for edition in the whole repo and came up blank, 

Did you use GitHub search? If so, not finding "edition" can probably be explained by GitHub (annoyingly) not indexing forks...

peacememories, 2018-11-17:
Yeah I did. That's... unfortunate 😅

peacememories, 2018-11-17:
Hmm, I rebuilt the newest avr rustc, and now xargo cannot compile libcore anymore, because it seems to be too old. I tried using cargo instead (so, using the normal libcore i guess?), but now it doesn't seem to understand `asm!("SEI")` anymore O.o

dylanmckay, 2018-11-18:
@peacememories Are you still using the [avr-rust/libcore](https://github.com/avr-rust/libcore) fork? A few people have had problems with their own projects still including the explicit `libcore` dependency in `Xargo.toml` which can be fixed by deleting `Xargo.toml`

shepmaster, 2018-11-18:
> I recently tried adding the new futures-preview 0.3-alpha.9 to my avr rust project

This compiles for me, but without allocation support, it's *very* hard to actually work.

peacememories, 2018-11-18:
@dylanmckay I've tried a few methods now:

* Switch to `cargo +avr build --release --target avr-atmega1280`

    This fails at link-time because it cannot find a libcore

* Use xargo without any options ( `rustup run avr xargo build --release --target avr-atmega1280` plus the two environment variables so the target file can be found)

    This tries to compile with the avr-rust/libcore fork, which has old syntax like `...` and other incompatible stuff in it, apparently

* Use xargo with avr-rust/libcore fork specified in Cargo.toml

   Same as above

* Use cargo with the core lib pointing to the checked-out libcore folder in avr-rust/rust

    Libcore fails to compile because of "target-incompatible datalayout"

The avr-atmega1280.json: https://hastebin.com/ludicipigu.json

peacememories, 2018-11-18:
Oh, I misread, I totally forgot the Xargo.toml

peacememories, 2018-11-18:
With Xargo.toml deleted, it's now searching in `$(pwd)/libcore` for a libcore O.o

peacememories, 2018-11-18:
So, I revisited the xargo example in the readme, and there seem to have been a few changes. I've gotten my base program to comple, albeit not as well as before. Specifically, if I don't turn on lto, the binary of my (not very big) program overflows the text section of my target. And that is with `-Os` and `-Wl,--gc-sections` enabled.

Also, specifying anything other than `lto="thin"` results in an assertion error from the compiler (somewhere in some c++ code?!)

And finally, adding the futures-preview to my dependencies makes the build fail because the core lib seems to not have `core::pin::Pin`, only `core::pin::PinMut`

shepmaster, 2018-11-18:
So, you are able to compile using edition 2018 and this issue is addressed?

shepmaster, 2018-11-18:
> the core lib seems to not have `core::pin::Pin`, only `core::pin::PinMut`

Right. Like I mentioned earlier, actually using this crate is a pain; you need to disable most of the features:

```
[dependencies.futures-preview]
version = "0.3.0-alpha.9"
default-features = false
```

Additionally, you cannot use `async`/`await` syntax in a no-std environment yet, either.

TimNN, 2018-11-18:
> Also, specifying anything other than lto="thin" results in an assertion error from the compiler (somewhere in some c++ code?!)

The assertion wouldn't be the one described in https://github.com/avr-rust/rust/issues/117, by any chance?

If so, that issue has an LLVM patch at the end (https://github.com/TimNN/llvm/commit/fce776910a1fe4295f05386e34683536d82d3cb1), which should fix the assertion.

peacememories, 2018-11-18:
> So, you are able to compile using edition 2018 and this issue is addressed?

Yes, I can build with the new compiler, so I'll close this issue. Thanks for all your help! :) But I'd still like to borrow your time for a sec:

> > if I don't turn on lto,
> 
> I would recommend always using LTO for an embedded device.

I know I should use it, and I can understand that it would otherwise result in binaries too big for the target (libcore alone is pretty big), I'm just confused why it *started* overflowing with the new version. Probably some subtle changes in the libcore or something?

> > Also, specifying anything other than lto="thin" results in an assertion error from the compiler (somewhere in some c++ code?!)
> 
> The assertion wouldn't be the one described in #117, by any chance?
> 
> If so, that issue has an LLVM patch at the end ([TimNN/llvm@fce7769](https://github.com/TimNN/llvm/commit/fce776910a1fe4295f05386e34683536d82d3cb1)), which should fix the assertion.

Yes, that's the exact problem I think. I'll look into the patch. For now, though, `lto="thin"` seems to work (I'll probably have to look into what "thin" and "fat" mean in this context^^) so it's not the highest priority for me.

> Right. Like I mentioned earlier, actually using this crate is a pain; you need to disable most of the features:
>
> ```
> [dependencies.futures-preview]
> version = "0.3.0-alpha.9"
> default-features = false [...]

@shepmaster I know that a lot of the features of the crate are not supported without std, and I'm building with `futures-preview = { version = "0.3.0-alpha.9", default-features = false }` in my `[dependencies]` key, which *should* do the same as you described.
But it seems weird for `Pin` to be one of the things not compatible with avr, and even more weird for `core::pin` to supply `PinMut` but not `Pin`.
You mentioned you got the crate to compile - would you mind checking if your libcore contains `Pin`?

Again, thank you all so much for helping me out here :)

closed 2018-11-18 by peacememories

peacememories, 2018-11-18:
> 
> 
> > your libcore contains `Pin`
> 
> Yes, my branch of avr-rust has been updated to be based on Rust upstream's [42959a2](https://github.com/avr-rust/rust/commit/42959a24a14e86804b81a3f8f8ecf78c367f370f), while the branch provided here is slightly older. I was waiting for a few things to land in master before pushing that branch.

Is it remotely feasible for someone with little knowledge of rust-avr to rebase to that commit, or should I just wait until the rebase is pushed?

shepmaster, 2018-11-21:
> Is it remotely feasible for someone with little knowledge of rust-avr to rebase to that commit

It's not terrible, but there are some conflicts that I've addressed already that would probably be easier to use.

> should I just wait until the rebase is pushed?

This is the lazy solution, so I recommend it! 😉

shepmaster, 2018-11-21:
I've updated the avr-support branch to be based off of rust-lang/rust@f37247f8.