avr-rust/rust-legacy-fork #121: Getting "overlaps section" when trying to link object files...

carlos4242, 2018-12-11:
So I've got this relatively(?) simple LLVM IR...

````
; ModuleID = 'main_clean.ll'
source_filename = "main.ll"
target datalayout = "e-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-atmel-linux-gnueabihf"

%TSi = type <{ i16 }>

@"$s4main1iSivp" = hidden local_unnamed_addr global %TSi zeroinitializer, align 2
@"$s4main1jSivp" = hidden local_unnamed_addr global %TSi zeroinitializer, align 2
@"$s4main1kSivp" = hidden global %TSi zeroinitializer, align 2
@__swift_reflection_version = linkonce_odr hidden constant i16 3

; Function Attrs: noreturn
define protected i16 @main(i16, i8** nocapture readnone) local_unnamed_addr #0 {
entry:
  %access-scratch = alloca [6 x i8], align 2
  tail call swiftcc void @"$s3AVR11SetupSerial8baudRateys6UInt16V_tF"(i16 9600)
  store i16 8, i16* getelementptr inbounds (%TSi, %TSi* @"$s4main1iSivp", i16 0, i32 0), align 2
  store i16 9, i16* getelementptr inbounds (%TSi, %TSi* @"$s4main1jSivp", i16 0, i32 0), align 2
  store i16 0, i16* getelementptr inbounds (%TSi, %TSi* @"$s4main1kSivp", i16 0, i32 0), align 2
  %2 = getelementptr inbounds [6 x i8], [6 x i8]* %access-scratch, i16 0, i16 0
  call void @llvm.lifetime.start.p0i8(i64 -1, i8* nonnull %2)
  call void @swift_beginAccess(i8* bitcast (%TSi* @"$s4main1kSivp" to i8*), [6 x i8]* nonnull %access-scratch, i16 1, i8* null) #3
  br label %3

; <label>:3:                                      ; preds = %3, %entry
  %4 = load i16, i16* getelementptr inbounds (%TSi, %TSi* @"$s4main1kSivp", i16 0, i32 0), align 2
  %5 = add i16 %4, 8
  store i16 %5, i16* getelementptr inbounds (%TSi, %TSi* @"$s4main1kSivp", i16 0, i32 0), align 2
  call swiftcc void @"$s3AVR5print11unsignedInt10addNewlineys6UInt16V_SbtF"(i16 0, i1 true)
  br label %3
}

; Function Attrs: argmemonly nounwind
declare void @llvm.lifetime.start.p0i8(i64, i8* nocapture) #1

declare swiftcc void @"$s3AVR11SetupSerial8baudRateys6UInt16V_tF"(i16) local_unnamed_addr #2

declare swiftcc void @"$s3AVR5print11unsignedInt10addNewlineys6UInt16V_SbtF"(i16, i1) local_unnamed_addr #2

; Function Attrs: nounwind
declare void @swift_beginAccess(i8*, [6 x i8]*, i16, i8*) local_unnamed_addr #3

attributes #0 = { noreturn "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" }
attributes #1 = { argmemonly nounwind }
attributes #2 = { "no-frame-pointer-elim"="true" "no-frame-pointer-elim-non-leaf" }
attributes #3 = { nounwind }

!swift.module.flags = !{!0}

!0 = !{!"standard-library", i1 false}
````

Compiling it using LLC with the avr back end produces something that looks like a valid object file, but it cannot be linked into an executable.

`/Users/carlpeto/llvm/llvm-avr/llc -O3 -march=avr -mcpu=atmega328p -filetype=obj -o main.o main_clean_opt.ll`

The linker error is...
````
"/usr/local/CrossPack-AVR/bin/avr-gcc" -mmcu=atmega328p -B"/usr/local/CrossPack-AVR/bin" -Wl,-u,vfprintf -lprintf_flt -lm -o main.elf main.o  strings.o "AVR/libAVR.a" "uSwift/libSwift.a" "CAVR/libavr.a"
/usr/local/CrossPack-AVR-20131216/lib/gcc/avr/4.8.1/../../../../avr/bin/ld: section .swift1_autolink_entries loaded at [00002488,00002493] overlaps section .data loaded at [00002488,000024b5]
collect2: error: ld returned 1 exit status
make: *** [main.elf] Error 1
````

I'm not much of an expert on linkers (they're a bit of a black box to me).

I tried to debug it using avr-objdump...
````
main.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000074  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .bss          00000006  00000000  00000000  000000a8  2**1
                  ALLOC
  2 .rodata       00000002  00000000  00000000  000000a8  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .swift1_autolink_entries 00000006  00000000  00000000  000000aa  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
````

I realised it might be some of the files archived into `AVR/libAVR.a` so I tried to dump that. (Bear with me if I'm doing something dumb.)

````
In archive AVR/libAVR.a:

utils.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         000000f0  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .rodata       00000002  00000000  00000000  00000124  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .swift1_autolink_entries 00000006  00000000  00000000  00000126  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA

pins.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000084  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .rodata       00000002  00000000  00000000  000000b8  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .swift1_autolink_entries 00000006  00000000  00000000  000000ba  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA

serial.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000278  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .rodata       00000002  00000000  00000000  000002ac  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .swift1_autolink_entries 00000006  00000000  00000000  000002ae  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA

async.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         0000002c  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .rodata       00000002  00000000  00000000  00000060  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .swift1_autolink_entries 00000006  00000000  00000000  00000062  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, DATA

constants.o:     file format elf32-avr

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000000  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  00000034  2**0
                  ALLOC
  3 .rodata       00000008  00000000  00000000  00000034  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .comment      00000012  00000000  00000000  0000003c  2**0
                  CONTENTS, READONLY
````

Can anyone with a little more understanding of linking point me to what's wrong?

I'm wondering if the llc back end isn't outputting custom sections correctly somehow?

Thanks for any help you guys can offer.

dylanmckay, 2018-12-18:
It's possible, and common (perhaps even the common case, I can't remember my avr-gcc Makefiles), to explicitly specify the addresses of 

```
avr-ld --help
...
       -Tbss=org
       -Tdata=org
       -Ttext=org
           Same as --section-start, with ".bss", ".data" or ".text" as the
           sectionname.
...
```

I wonder if plugging in similar command line linker arguments to Arduino proper might help.

carlos4242, 2018-12-19:
> Also, if this is an LLVM bug, it's probably a bug in the LLVM AVR ELF logic.
>
> We can check this; compile the IR to AVR assembly and use avr-gcc or avr-as to link/build it. If it works fine, then the bug is definitely in the AVR backend.

I was going to write back last night but didn’t know what to say. Glad you wrote. :)

Ok. That sounds like a good idea. I’m still not great on the layout of the stuff pre linking. I’ll take a look.

Seems like it must be some kind of bug between llvm IR and machine code somewhere.

Is there any way the LLVM IR could be malformed by the front end? It looks like it doesn’t contain any attributes to do with section layout? Just something that says what section a top level symbol should go into etc.? Unless there are restrictions on the section names?

If it isn’t in the llvm ir I guess that leaves lowering to assembly/MC or elf emission, like you say.

I’ll have a go at that test tonight. Good News is its one of my last obstacles. :)

I hope you’re well? Haven’t chatted in ages!

Carl

carlos4242, 2019-03-31:
OK, I eventually managed to get this working and the investigation was a long, painful process!

I got llvm to output assembly then got avr-gcc to compile it.  The big sticking point was that avr-gcc couldn't compile my code because swift produces symbol names with $ in. After long investigation, the problem is AVR assembly language has a little known peculiarity. For some reason, $ can be used as a line ending to put multiple statements on one line, a bit like ; in C. Very weird. So the symbols were valid for normal gcc but *not* for avr-gcc, which saw them as a line break!

After resolving that with upstream hacks, I managed to get assembly compiled by gcc... linked it and... got exactly the same overlaps section error!!

So at this point, we've ruled out LLVM completely, except for the .section assembly directives.

After more experimentation and some luck, I found the issue. It was just old tools. I was using a version of gcc and binutils from 2013 that had some issues handling these sections somehow.

I've upgraded and the problem seems to have gone away, at least while compiled via assembly and gcc. The next step will be to check that the code still works on a device.

Then go back to making object files directly with llvm and check that everything still works.

But I'm closing this issue now as I no longer believe it's an LLVM error. I think it's my toolchain.

closed 2019-03-31 by carlos4242

dylanmckay, 2019-04-01:
> For some reason, $ can be used as a line ending to put multiple statements on one line, a bit like ; in C. Very weird. So the symbols were valid for normal gcc but not for avr-gcc, which saw them as a line break!

TIL, surprising!