avr-rust/rust-legacy-fork #131: make install fails on macOS

ratkins, 2019-02-11:
I ran into a confusing beginner problem trying to build `avr-rust` on macOS Mojave (10.14.3). `make` worked fine, but `make install` told me it `could not canonicalize /opt/avr-rust`.

In order to fix this I had to edit `build/config.toml` and set `prefix = '/usr/local/avr-rust'`, and then make sure the directory was there and had the right ownership (`sudo mkdir /usr/local/avr-rust && sudo chown $USER:admin /usr/local/avr-rust`).

After that `make install` worked.

Finally, for the "Register the toolchain with rustup" step of [Building and Installation](https://github.com/avr-rust/rust#building-and-installation), I had to manually install GNU Coreutils to get `realpath` (`brew install coreutils`).

Is it worth trying to automate this (any hints to where I should start?), or should I just submit a PR on the readme noting the extra steps on macOS?

carlos4242, 2019-03-31:
Gosh, no one commented on this. Definitely raise a PR on the README, that would be a great addition.

I think not that many developers are building on mac but I am so it would be good for people like me when we start, I would have got stuck on that for sure!

ratkins, 2019-04-01:
Ok, [done](https://github.com/avr-rust/rust/pull/135).

dylanmckay, 2019-05-16:
Fixed in #135.

closed 2019-05-16 by dylanmckay

crazycarl, 2019-10-17:
I got the "could not canonicalize /opt/avr-rust" error on my Ubuntu 18 installation.
Following the note for fixing it on OSX has fixed it for me.

So it should be noted that this issues is not specific to OSX.