avr-rust/rust-legacy-fork #133: error: Error loading target specification: Could not find specification for target "avr-atmega328p"

1inguini, 2019-02-23:
the shell i am using is `fish, version 3.0.1`, the directory I installed avr-rust is `$HOME/.avr-rust/`.
versions of rustup and xargo are:
~~~
➤ rustup -V
rustup 1.16.0
~~~
~~~
➤ xargo -V
xargo 0.3.13
cargo 1.34.0-nightly (b33ce7fc9 2019-02-19)
~~~

I tried to build [blink](https://github.com/avr-rust/blink), only to get this error.
~~~
➤ env XARGO_RUST_SRC=$HOME/Desktop/rust/rust-avr/src/ env RUST_TARGET_PATH=`pwd` env RUST_BACKTRACE=1 xargo build --target avr-atmega328p --release
warning: Patch `rustc-std-workspace-core v1.0.0 (/home/linguini/Desktop/rust/rust-avr/src/tools/rustc-std-workspace-core)` was not used in the crate graph.
Check that the patched package version and available features are compatible
with the dependency requirements. If the patch has a different version from
what is locked in the Cargo.lock file, run `cargo update` to use the new
version. This may also occur with an optional dependency that is not enabled.
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --sysroot /home/linguini/.xargo -Z force-unstable-if-unmarked --target avr-atmega328p --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro` (exit code: 1)
--- stderr
error: Error loading target specification: Could not find specification for target "avr-atmega328p"
  |
  = help: Use `--print target-list` for a list of built-in targets

error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.IOntLav1ICuD/Cargo.toml" "--target" "avr-atmega328p" "-p" "core"` failed with exit code: Some(101)
stack backtrace:
   0: error_chain::make_backtrace::h90af62d5c4a6fbd6 (0x5635759c416b)
   1: <error_chain::State as core::default::Default>::default::h23f817a122243ac9 (0x5635759c4235)
   2: <std::process::Command as xargo::extensions::CommandExt>::run::h01346a6b4f35f543 (0x563575977d4b)
   3: xargo::sysroot::build::h240260972f2e09ed (0x56357597f541)
   4: xargo::sysroot::update::h94fd82f52808b493 (0x56357598365d)
   5: xargo::main::ha282b52dabbcaa07 (0x563575970c7a)
   6: std::rt::lang_start::{{closure}}::h6c099b348d99c7de (0x5635759748f2)
   7: std::rt::lang_start_internal::{{closure}}::h53f6e74893a27d59 (0x5635759ef162)
             at src/libstd/rt.rs:59
      std::panicking::try::do_call::h8945e3e0fe63365a
             at src/libstd/panicking.rs:307
   8: __rust_maybe_catch_panic (0x5635759fc1a9)
             at src/libpanic_unwind/lib.rs:102
   9: std::panicking::try::hee49dda4da841cb1 (0x5635759ef058)
             at src/libstd/panicking.rs:286
  10: std::panic::catch_unwind::h89d0b17e1bb724c9 (0x5635759e2665)
             at src/libstd/panic.rs:398
      std::rt::lang_start_internal::hcbec2a0475a1169a
             at src/libstd/rt.rs:58
  11: main (0x563575971a61)
  12: __libc_start_main (0x7f09fb9ef222)
  13: _start (0x5635759611ad)
  14: <unknown> (0x0)
~~~

1inguini, 2019-02-23:
It seems that the point is I can't compile the core.

carlos4242, 2019-03-14:
This is a duplicate of #128

hercek, 2019-04-17:
It does not look like a duplicate. I can compile core crate after the patches in #128. But the build does not continue since it cannot find compiler_builtins:
```
> rustup run avr-toolchain xargo build --target avr-atmega328p --release --verbose
+ "rustc" "--print" "sysroot"
+ "rustc" "--print" "target-list"
+ RUSTFLAGS="--sysroot /home/peter/.xargo -Z force-unstable-if-unmarked"
+ "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.hFrKDxzhtuuu/Cargo.toml" "--target" "avr-atmega328p" "-v" "-p" "core"
   Compiling core v0.0.0 (/home/peter/rust/rust/src/libcore)
     Running `rustc --crate-name core /home/peter/rust/rust/src/libcore/lib.rs --color never --crate-type lib --emit=dep-info,link -C opt-level=3 -C metadata=f7a2b86276dfcf90 -C extra-filename=-f7a2b86276dfcf90 --out-dir /tmp/xargo.hFrKDxzhtuuu/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.hFrKDxzhtuuu/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.hFrKDxzhtuuu/target/release/deps --sysroot /home/peter/.xargo -Z force-unstable-if-unmarked`
    Finished release [optimized] target(s) in 26.38s
+ RUSTFLAGS="--sysroot /home/peter/.xargo -Z force-unstable-if-unmarked"
+ "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.QQRqXfptYN1G/Cargo.toml" "--target" "avr-atmega328p" "-v" "-p" "compiler_builtins"
error: failed to load source for a dependency on `compiler_builtins`

Caused by:
  Unable to update /home/peter/rust/rust/src/libcompiler_builtins

Caused by:
  failed to read `/home/peter/rust/rust/src/libcompiler_builtins/Cargo.toml`

Caused by:
  No such file or directory (os error 2)
error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.QQRqXfptYN1G/Cargo.toml" "--target" "avr-atmega328p" "-v" "-p" "compiler_builtins"` failed with exit code: Some(101)
note: run with `RUST_BACKTRACE=1` for a backtrace
```
When I try to compile compiler_builtins from https://github.com/rust-lang-nursery/compiler-builtins then it failes because of std crate:
```
> rustup run avr-toolchain cargo xbuild --target avr-atmega328p --release --verbose 
+ "rustc" "--print" "sysroot"
+ "rustc" "--print" "target-list"
+ "/home/peter/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.jZbhk82SGOoA/Cargo.toml" "--target" "avr-atmega328p" "-v" "--" "-Z" "force-unstable-if-unmarked"
    Updating crates.io index
   Compiling compiler_builtins v0.1.10
   Compiling core v0.0.0 (/home/peter/rust/rust/src/libcore)
     Running `rustc --crate-name build_script_build /home/peter/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.10/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=33773de8e1bd096a -C extra-filename=-33773de8e1bd096a --out-dir /tmp/xargo.jZbhk82SGOoA/target/release/build/compiler_builtins-33773de8e1bd096a -L dependency=/tmp/xargo.jZbhk82SGOoA/target/release/deps --cap-lints allow`
     Running `rustc --crate-name core /home/peter/rust/rust/src/libcore/lib.rs --color never --crate-type lib --emit=dep-info,link -C opt-level=3 -C panic=abort -C metadata=93e18af0685e765c -C extra-filename=-93e18af0685e765c --out-dir /tmp/xargo.jZbhk82SGOoA/target/avr-atmega328p/release/deps --target avr-atmega328p -L dependency=/tmp/xargo.jZbhk82SGOoA/target/avr-atmega328p/release/deps -L dependency=/tmp/xargo.jZbhk82SGOoA/target/release/deps`
error[E0463]: can't find crate for `std`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: Could not compile `compiler_builtins`.

Caused by:
  process didn't exit successfully: `rustc --crate-name build_script_build /home/peter/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.10/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=33773de8e1bd096a -C extra-filename=-33773de8e1bd096a --out-dir /tmp/xargo.jZbhk82SGOoA/target/release/build/compiler_builtins-33773de8e1bd096a -L dependency=/tmp/xargo.jZbhk82SGOoA/target/release/deps --cap-lints allow` (exit code: 1)
warning: build failed, waiting for other jobs to finish...
error: build failed
error: `"/home/peter/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.jZbhk82SGOoA/Cargo.toml" "--target" "avr-atmega328p" "-v" "--" "-Z" "force-unstable-if-unmarked"` failed with exit code: Some(101)
note: run with `RUST_BACKTRACE=1` for a backtrace
```
What is the point that some kind of compiler_builtins (which is supposed some kind of very low level stuff to avoid dependency on C runtime) depends on std crate?
And what is the correct way to build blink at all?