avr-rust/rust-legacy-fork #134: Unit tests regressions!

carlos4242, 2019-03-06:
I've just run the unit tests on avr-rust-2019-01-18 and I'm getting regressions?

Testing Time: 2.56s
********************
Failing Tests (4):
    LLVM :: CodeGen/AVR/call.ll
    LLVM :: CodeGen/AVR/directmem.ll
    LLVM :: CodeGen/AVR/div.ll
    LLVM :: CodeGen/AVR/rem.ll

  Expected Passes    : 117
  Expected Failures  : 3
  Unexpected Failures: 4

carlos4242, 2019-03-10:
These are all fixed by avr-rust/llvm PR #10.

carlos4242, 2019-03-10:
LABEL AS: has local patch

hercek, 2019-03-15:
How do you run the unit tests (except running them for each file manually)?
`llvm-lit` does not work on my build. It does not have python variable `llvm_config` defined probably becase `rust/src/llvm/test/lit.site.cfg.py.in` is not processed to `lit.site.cfg.py`.

carlos4242, 2019-03-15:
When you build llvm you’ll have a bin directory where llc, opt, etc. are located. In there is llvm-lit.

Run it then pass the test directory.

For example in my set up, I have the llvm source code in ~/llvm. The build is in a sub folder ~/llvm/build. To run lit I would do:

~/llvm/build/bin/llvm-lit ~/test/CodeGen/AVR

Which runs all tests. Or append a single test.

hercek, 2019-03-16:
Thanks. That is it. When you let rust build system to build llvm then it does not build `llvm-lit` and trying to run it from `build/utils/llvm-lit` does not work as I described before.

dylanmckay, 2020-03-03:
Since fixed in LLVM trunk.

closed 2020-03-03 by dylanmckay