avr-rust/rust-legacy-fork #148: `core::hint::unreachable_unchecked()` corrupts execution path

Rahix, 2019-06-18:
The following code, when compiled with the profile below, results in a binary with a corrupted execution path.

## Code
```rust
#![no_std]
#![no_main]

extern crate panic_halt;

use core::str;
use core::hint;

#[inline(never)]
fn foo(n: usize, buf: &mut [u8]) ->&str {
    let mut n = n;
    let mut i = buf.len() - 1;

    loop {
        *buf.get_mut(i).unwrap_or_else(|| unsafe { hint::unreachable_unchecked() }) = (n % 10) as u8 + b'0';
        n = n / 10;

        if n == 0 {
            break;
        } else {
            i -= 1;
        }
    }

    unsafe { str::from_utf8_unchecked(buf.get(i..).unwrap_or_else(|| hint::unreachable_unchecked())) }
}

#[inline(never)]
fn brk() -> ! {
    unsafe { core::ptr::write_volatile(0x3E as *mut u8, 0x00); }
    loop {}
}

#[no_mangle]
pub extern fn main() -> ! {
    // Ensure the value can't be constant folded
    let n = 201;
    unsafe { core::ptr::write_volatile(0x3E as *mut u8, n); }
    let n = unsafe { core::ptr::read_volatile(0x3E as *mut u8) } as usize;
    let n2 = unsafe { core::ptr::read_volatile(0x3E as *mut u8) } as usize + 30;

    let mut buf = [0; 3];
    let s = foo(n, &mut buf);

    let mut buf2 = [0; 3];
    let s2 = foo(n2, &mut buf2);

    brk();
}
```

## Cargo Profile
```toml
[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = 3
```

## Observation
The function `foo` is called just once and instead of returning to main, execution falls through into `brk` immediately.  This behavior is very fragile; even small changes to the code make the program behave correctly.

When looking at the disassembly of `foo`, I can see there is no `ret` instruction at the end which I guess is the reason for this behavior.

### Disassembly (Excerpt)
```asm
foo:
	; truncated
LBB0_5:
	movw	r26, r12
	movw	r8, r30
	add	r26, r30
	adc	r27, r31

LBB0_6:
	ori	r24, 0x30	; 48
	st	X, r24
	movw	r24, r2
	cp	r14, r24
	cpc	r15, r25
	movw	r14, r10
	brcc	.-58     	; 0x124 <LBB0_3>
	ldi	r24, 0x04	; 4
	ldi	r25, 0x00	; 0
	cp	r8, r24
	cpc	r9, r25
	; No return here ...

brk:
	ldi	r24, 0x00	; 0
	out	0x1e, r24	; 30

LBB1_1:
	rjmp	.-2      	; 0x16a <LBB1_1>

main:
	push	r28
	push	r29
	push	r14
	; truncated
```

I have attached a zip with the binary, asm and llvm-ir: [unreachable-bug.zip](https://github.com/avr-rust/rust/files/3300814/unreachable-bug.zip)

shepmaster, 2019-06-18:
You may want to try removing `loop {}` due to https://github.com/rust-lang/rust/issues/28728

I'd suggest a volatile/read and then write of some value instead:

```rust
loop {
    write_volatile(ADDR, read_volatile(ADDR));
}
```

Of if you are allowing inline ASM, just `asm!("NOP")` in the loop should be fine.

Rahix, 2019-06-19:
Hmm, I see.  Though I tried and it did not change anything.  (And the assembly above looks fine):

```asm
brk:
	ldi	r24, 0x00	; 0
	out	0x1e, r24	; 30

LBB1_1:
	rjmp	.-2      	; 0x16a <LBB1_1>
```

Rahix, 2019-06-19:
I was able to further reduce the test-case:

```rust
#![no_std]
#![no_main]

extern crate panic_halt;

#[inline(never)]
fn foo(mut n: usize, buf: &mut [u8]) {
    let mut i = 2;

    loop {
        n = n / 10;

        if n == 0 {
            break;
        } else {
            i -= 1;
        }
    }

    buf.get(i..).unwrap_or_else(|| unsafe { core::hint::unreachable_unchecked() });
}

#[inline(never)]
fn brk() -> ! {
    loop {
        unsafe { core::ptr::write_volatile(0x3E as *mut u8, 0x00); }
    }
}

#[no_mangle]
pub extern fn main() -> ! {
    // Ensure the value can't be constant folded
    unsafe { core::ptr::write_volatile(0x3E as *mut u8, 10); }
    let n = unsafe { core::ptr::read_volatile(0x3E as *mut u8) } as usize;

    let mut buf = [0; 4];
    foo(n, &mut buf);
    foo(n, &mut buf);

    brk();
}
```

Rahix, 2019-06-19:
The llvm-ir of the above code is

```llvm
; ModuleID = 'bug.3uecfach-cgu.0'
source_filename = "bug.3uecfach-cgu.0"
target datalayout = "e-P1-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8"
target triple = "avr-unknown-unknown"

; bug::foo
; Function Attrs: noinline nounwind
define internal fastcc void @_ZN3bug3foo17h4c36797574a26bb6E(i16) unnamed_addr addrspace(1) #0 {
start:
  %1 = icmp ult i16 %0, 10
  br i1 %1, label %bb3, label %bb4

bb3.loopexit:                                     ; preds = %bb4
  %phitmp = icmp ult i16 %3, 5
  br label %bb3

bb3:                                              ; preds = %bb3.loopexit, %start
  %i.0.lcssa = phi i1 [ true, %start ], [ %phitmp, %bb3.loopexit ]
  tail call addrspace(1) void @llvm.assume(i1 %i.0.lcssa) #3
  ret void

bb4:                                              ; preds = %start, %bb4
  %n.02 = phi i16 [ %2, %bb4 ], [ %0, %start ]
  %i.01 = phi i16 [ %3, %bb4 ], [ 2, %start ]
  %2 = udiv i16 %n.02, 10
  %3 = add i16 %i.01, -1
  %4 = icmp ult i16 %n.02, 100
  br i1 %4, label %bb3.loopexit, label %bb4
}

; bug::brk
; Function Attrs: noinline norecurse noreturn nounwind
define internal fastcc void @_ZN3bug3brk17he3963db3659f852bE() unnamed_addr addrspace(1) #1 {
start:
  br label %bb0

bb0:                                              ; preds = %bb0, %start
  store volatile i8 0, i8* inttoptr (i16 62 to i8*), align 2
  br label %bb0
}

; Function Attrs: noreturn nounwind
define void @main() unnamed_addr addrspace(1) #2 {
start:
  store volatile i8 10, i8* inttoptr (i16 62 to i8*), align 2
  %0 = load volatile i8, i8* inttoptr (i16 62 to i8*), align 2
  %1 = zext i8 %0 to i16
; call bug::foo
  call fastcc addrspace(1) void @_ZN3bug3foo17h4c36797574a26bb6E(i16 %1)
; call bug::foo
  call fastcc addrspace(1) void @_ZN3bug3foo17h4c36797574a26bb6E(i16 %1)
; call bug::brk
  call fastcc addrspace(1) void @_ZN3bug3brk17he3963db3659f852bE()
  unreachable
}

; Function Attrs: nounwind
declare void @llvm.assume(i1) addrspace(1) #3

attributes #0 = { noinline nounwind "target-cpu"="atmega32u4" }
attributes #1 = { noinline norecurse noreturn nounwind "target-cpu"="atmega32u4" }
attributes #2 = { noreturn nounwind "target-cpu"="atmega32u4" }
attributes #3 = { nounwind }

!llvm.module.flags = !{!0}

!0 = !{i32 7, !"PIE Level", i32 2}
```

Rahix, 2020-09-03:
Looks like codegen has changed and this is no longer reproducible!  I now see a `ret` at the end of `foo`:

```asm
foo:
	push	r6
	push	r7
	push	r8
	push	r9
	push	r10
	push	r11
	push	r12
	push	r13
	push	r14
	push	r15
	push	r16
	push	r17
	movw	r12, r22
	movw	r14, r24
	ldi	r16, 0x0A	; 10
	ldi	r17, 0x00	; 0
	movw	r22, r16
	call	0x18e	; 0x18e <__udivmodhi4>
	ori	r24, 0x30	; 48
	movw	r30, r12
	std	Z+2, r24	; 0x02
	cp	r14, r16
	cpc	r15, r17
	brcs	.+58     	; 0x112 <foo::hd176b67340c5228c+0x6c>
	ldi	r30, 0x02	; 2
	ldi	r31, 0x00	; 0
	ldi	r16, 0x0A	; 10
	ldi	r17, 0x00	; 0
	ldi	r24, 0x64	; 100
	ldi	r25, 0x00	; 0
	movw	r6, r24
	movw	r24, r14
	movw	r22, r16
	movw	r8, r30
	call	0x18e	; 0x18e <__udivmodhi4>
	movw	r10, r22
	movw	r24, r10
	movw	r22, r16
	call	0x18e	; 0x18e <__udivmodhi4>
	movw	r30, r8
	sbiw	r30, 0x01	; 1
	movw	r26, r12
	add	r26, r30
	adc	r27, r31
	ori	r24, 0x30	; 48
	st	X, r24
	movw	r24, r6
	cp	r14, r24
	cpc	r15, r25
	movw	r14, r10
	brcc	.-44     	; 0xe6 <foo::hd176b67340c5228c+0x40>
	pop	r17
	pop	r16
	pop	r15
	pop	r14
	pop	r13
	pop	r12
	pop	r11
	pop	r10
	pop	r9
	pop	r8
	pop	r7
	pop	r6
	ret

brk:
	ldi	r24, 0x00	; 0
	out	0x1e, r24	; 30
	rjmp	.-2      	; 0x130 <brk::hb1a17e2f7472a5b0+0x4>

main:
	push	r14
	push	r15
	push	r16
	push	r17
	push	r28
	push	r29
	in	r28, 0x3d	; 61
	in	r29, 0x3e	; 62
	sbiw	r28, 0x06	; 6
	in	r0, 0x3f	; 63
	cli
	out	0x3e, r29	; 62
	out	0x3f, r0	; 63
	out	0x3d, r28	; 61
	ldi	r24, 0xC9	; 201
	out	0x1e, r24	; 30
	in	r24, 0x1e	; 30
	in	r17, 0x1e	; 30
	ldi	r16, 0x00	; 0
	std	Y+3, r16	; 0x03
	ldi	r18, 0x00	; 0
	ldi	r19, 0x00	; 0
	std	Y+1, r18	; 0x01
	std	Y+2, r19	; 0x02
	movw	r14, r18
	eor	r25, r25
	movw	r22, r28
	subi	r22, 0xFF	; 255
	sbci	r23, 0xFF	; 255
	call	0xa6	; 0xa6 <foo::hd176b67340c5228c>
	std	Y+6, r16	; 0x06
	std	Y+4, r14	; 0x04
	std	Y+5, r15	; 0x05
	mov	r24, r17
	eor	r25, r25
	adiw	r24, 0x1e	; 30
	movw	r22, r28
	subi	r22, 0xFC	; 252
	sbci	r23, 0xFF	; 255
	call	0xa6	; 0xa6 <test_broken_avr::foo::hd176b67340c5228c>
	call	0x12c	; 0x12c <test_broken_avr::brk::hb1a17e2f7472a5b0>
	call	0x1b6	; 0x1b6 <abort>
```

dylanmckay, 2020-10-29:
I'm going to close this one but if it happens again, feel free to reopen on the upstream Rust repo

closed 2020-10-29 by dylanmckay