avr-rust/rust-legacy-fork #150: Undefined reference to compare_and_swap

peacememories, 2019-09-21:
When I try to use `AtomicBool::compare_and_swap` I get the following linker error:

```
undefined reference to `__sync_val_compare_and_swap_1'
```

Is the `compare_and_swap` intrinsic not implemented in this compiler, or is there something I could do to fix this?

dylanmckay, 2019-09-26:
Perhaps similar to https://github.com/rust-lang/rust/issues/58500

dylanmckay, 2019-09-26:
@peacememories would you please be able to attach an LLVM IR file so I can reproduce this?

peacememories, 2019-09-26:
I’d be glad to. Where do I find the relevant files you need? :)

Rahix, 2019-09-28:
Generate them with `cargo rustc ... -- --emit=llvm-ir -o<...>` (if I recall everything correctly)

peacememories, 2019-10-02:
Okay, sadly I think I'll have to fix my rust-avr installation before I can do that. In my effort to eliminate error sources I updated it and it's broken right now (see https://github.com/avr-rust/rust/issues/151)