avr-rust/rust-legacy-fork #151: Trouble compiling on macOS

peacememories, 2019-09-23:
I recently pulled the latest version of rust-avr and tried to compile it again, but ran into the following problem:
https://gist.github.com/ab49f1171d442728d61a93398d0b046e
Short version:
```
...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
...
```

I'm running macOS 10.14.6 with xCode 11 (which I recently updated, so that could also be part of the problem)

Gaelan, 2020-04-23:
A little more investigation, since I ran into this as well: this message shows up in the logs.

```
note: ld: warning: ignoring file /var/folders/9b/sq7l7grn1_q_bm74x5x67z8r0000gn/T/rustcZOI6jz/librustc_llvm-e87e1352d21cfdff.rlib, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x53 0x59 0x4D 0x36 0x34 0x2F 0x20 )
```

Those hex bytes work out to `!<arch>\n/SYM64/` in ASCII; some googling suggests that this indicates some sort of 64-bit ar format, for archives larger than 4 GB. On that theory, I'm currently testing to see if a no-debug-symbol (which presumably produces substantially smaller binaries). I'll report back once that finishes.

Gaelan, 2020-04-23:
Yep, a non-debug build works.