avr-rust/rust-legacy-fork #153: Trouble compiling on Linux

adjivas, 2019-11-17:
Hello, I report this problem, I have tried to compile the last version of april-rust but I have gotten this error:
```bash
[ 86%] Linking CXX shared library ../../lib/libLLVM-8-rust-1.37.0-dev-275759724.so
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make[3]: *** [tools/llvm-shlib/CMakeFiles/LLVM.dir/build.make:254: lib/libLLVM-8-rust-1.37.0-dev-275759724.so] Error 1
make[3]: *** Deleting file 'lib/libLLVM-8-rust-1.37.0-dev-275759724.so'
make[3]: Leaving directory '/home/adjivas/build/build/x86_64-unknown-linux-gnu/llvm/build'
make[2]: *** [CMakeFiles/Makefile2:21176: tools/llvm-shlib/CMakeFiles/LLVM.dir/all] Error 2
make[2]: Leaving directory '/home/adjivas/build/build/x86_64-unknown-linux-gnu/llvm/build'
make[1]: *** [Makefile:152: all] Error 2
make[1]: Leaving directory '/home/adjivas/build/build/x86_64-unknown-linux-gnu/llvm/build'
thread 'main' panicked at '
command did not execute successfully, got: exit code: 2

build script failed, must exit now', /home/adjivas/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.38/src/lib.rs:813:5
note: Run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
        finished in 35256.288
failed to run: /home/adjivas/build/build/bootstrap/debug/bootstrap build
Build completed unsuccessfully in 14:19:51
make: *** [Makefile:18: all] Error 1
```
Here is the complet log: [log.txt](https://github.com/avr-rust/rust/files/3855717/log.txt)

adjivas, 2019-11-20:
I have reproducted the same incident with the [avr-support(37538aa13) - Update to LLVM 9.0 and current master](https://github.com/avr-rust/rust/pull/152).
```
[ 86%] Linking CXX shared library ../../lib/libLLVM-9-rust-1.40.0-dev-8adf9bdcc.so
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make[3]: *** [tools/llvm-shlib/CMakeFiles/LLVM.dir/build.make:248: lib/libLLVM-9-rust-1.40.0-dev-8adf9bdcc.so] Error 1
make[3]: *** Deleting file 'lib/libLLVM-9-rust-1.40.0-dev-8adf9bdcc.so'
make[3]: Leaving directory '/home/adjivas/rust/build/x86_64-unknown-linux-gnu/llvm/build'
make[2]: *** [CMakeFiles/Makefile2:22003: tools/llvm-shlib/CMakeFiles/LLVM.dir/all] Error 2
make[2]: Leaving directory '/home/adjivas/rust/build/x86_64-unknown-linux-gnu/llvm/build'
make[1]: *** [Makefile:152: all] Error 2
make[1]: Leaving directory '/home/adjivas/rust/build/x86_64-unknown-linux-gnu/llvm/build'
thread 'main' panicked at '
command did not execute successfully, got: exit code: 2

build script failed, must exit now', /home/adjivas/.cargo/registry/src/github.com-1ecc6299db9ec823/cmake-0.1.38/src/lib.rs:813:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
        finished in 40391.607
failed to run: /home/adjivas/rust/build/bootstrap/debug/bootstrap build
Build completed unsuccessfully in 12:59:46
make: *** [Makefile:18: all] Error 1
```
Here is the complet log: [log.txt](https://github.com/avr-rust/rust/files/3871258/log.txt)

Rahix, 2019-11-21:
The offending line in this log seems to be

```text
collect2: fatal error: ld terminated with signal 9 [Killed]
```

which points to something on your system killing the process ... Might be an OOM situation, how much RAM do you have and can you check whether the limit gets reached during the build?  Do you have swap-space?

dylanmckay, 2019-11-26:
OOM is definitely common - usually for me, the problem is that I want to _compile_ in parallel (`make -j4`), but link sequentially because the GNU linker can use a lot of memory.

adjivas, 2019-12-06:
Thank you for your informations, the specific issue who describe this problem is documented here: [Rust compiler creating huge files on tmp directory](https://github.com/rust-lang/rust/issues/64276). So this problem not really concerned the avr-rust project. In my case, I have redefined the _TMPDIR_ environement variable for fixe it.

closed 2019-12-06 by adjivas