avr-rust/rust-legacy-fork #160: [Docker][MacOs] Failed to build rustc

Smarco10, 2020-03-14:
Hi,

After trying to build avr-rust, I run into an error that I do not succeed to fix by myself.
I tried to build avr-rust on docker since a Dockerfile is provided, I supposed it was the easiest way to get it.

Here are command I run:
```
git clone https://github.com/avr-rust/rust.git
cd rust 
docker image build -t rust-avr-build .
```

And here is the result i've got:
```
Sending build context to Docker daemon  590.6MB
Step 1/12 : FROM ubuntu:18.04
18.04: Pulling from library/ubuntu
Digest: sha256:04d48df82c938587820d7b6006f5071dbbffceb7ca01d2814f81857c631d44df
Status: Downloaded newer image for ubuntu:18.04
 ---> 72300a873c2c
Step 2/12 : USER root
 ---> Running in 31aabc2ab164
Removing intermediate container 31aabc2ab164
 ---> 56b4ad5629d3
Step 3/12 : RUN apt-get update && apt-get install -y build-essential cmake curl git python
 ---> Running in 3c3f3d6b39d8
Get:1 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Get:3 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:5 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [31.0 kB]
Get:6 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [7640 B]
Get:7 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [826 kB]
Get:8 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [836 kB]
Get:9 http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages [1344 kB]
Get:10 http://archive.ubuntu.com/ubuntu bionic/restricted amd64 Packages [13.5 kB]
Get:11 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [11.3 MB]
Get:12 http://archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [186 kB]
Get:13 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [44.7 kB]
Get:14 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1355 kB]
Get:15 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [11.7 kB]
Get:16 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [1128 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [2496 B]
Get:18 http://archive.ubuntu.com/ubuntu bionic-backports/universe amd64 Packages [4247 B]
Fetched 17.6 MB in 7s (2588 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu ca-certificates
  cmake-data cpp cpp-7 dirmngr dpkg-dev fakeroot file g++ g++-7 gcc gcc-7
  gcc-7-base gcc-8-base git-man gnupg gnupg-l10n gnupg-utils gpg gpg-agent
  gpg-wks-client gpg-wks-server gpgconf gpgsm krb5-locales less
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libarchive13 libasan4 libasn1-8-heimdal libassuan0 libatomic1 libbinutils
  libbsd0 libc-dev-bin libc6-dev libcc1-0 libcilkrts5 libcurl3-gnutls libcurl4
  libdpkg-perl libedit2 liberror-perl libexpat1 libfakeroot
  libfile-fcntllock-perl libgcc-7-dev libgcc1 libgdbm-compat4 libgdbm5
  libgomp1 libgssapi-krb5-2 libgssapi3-heimdal libhcrypto4-heimdal
  libheimbase1-heimdal libheimntlm0-heimdal libhx509-5-heimdal libicu60
  libisl19 libitm1 libjsoncpp1 libk5crypto3 libkeyutils1 libkrb5-26-heimdal
  libkrb5-3 libkrb5support0 libksba8 libldap-2.4-2 libldap-common
  liblocale-gettext-perl liblsan0 liblzo2-2 libmagic-mgc libmagic1 libmpc3
  libmpfr6 libmpx2 libnghttp2-14 libnpth0 libperl5.26 libpsl5 libpython-stdlib
  libpython2.7-minimal libpython2.7-stdlib libquadmath0 libreadline7 librhash0
  libroken18-heimdal librtmp1 libsasl2-2 libsasl2-modules libsasl2-modules-db
  libsqlite3-0 libssl1.0.0 libssl1.1 libstdc++-7-dev libstdc++6 libtsan0
  libubsan0 libuv1 libwind0-heimdal libx11-6 libx11-data libxau6 libxcb1
  libxdmcp6 libxext6 libxml2 libxmuu1 linux-libc-dev make manpages
[... 1471 lines cut ...]
   Compiling syn v0.15.22
   Compiling rls-data v0.19.0
   Compiling tempfile v3.0.5
   Compiling parking_lot v0.7.1
   Compiling rustc_data_structures v0.0.0 (/code/src/librustc_data_structures)
   Compiling arena v0.0.0 (/code/src/libarena)
   Compiling synstructure v0.10.2
   Compiling rustc_macros v0.1.0 (/code/src/librustc_macros)
   Compiling syntax_pos v0.0.0 (/code/src/libsyntax_pos)
   Compiling rustc_errors v0.0.0 (/code/src/librustc_errors)
   Compiling syntax_ext v0.0.0 (/code/src/libsyntax_ext)
error: Could not compile `rustc`.

Caused by:
  process didn't exit successfully: `/build/bootstrap/debug/rustc --edition=2018 --crate-name rustc src/librustc/lib.rs --color never --error-format json --crate-type dylib --emit=dep-info,link -C prefer-dynamic -C opt-level=2 -C metadata=c3b86f1aefb9449e -C extra-filename=-c3b86f1aefb9449e --out-dir /build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps --target x86_64-unknown-linux-gnu -L dependency=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps -L dependency=/build/x86_64-unknown-linux-gnu/stage0-rustc/release/deps --extern arena=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libarena-59b6aa170dd18238.so --extern backtrace=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libbacktrace-a0e4c87a9562fd79.rlib --extern bitflags=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libbitflags-c8326f205062942c.rlib --extern byteorder=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libbyteorder-1ff1b53c4083de0b.rlib --extern chalk_engine=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libchalk_engine-47389a0d66fdb172.rlib --extern flate2=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libflate2-a336721d6cdb31d7.rlib --extern fmt_macros=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libfmt_macros-a1c850595bc2a1b5.so --extern graphviz=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libgraphviz-5802bba1e3dd27f2.so --extern jobserver=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libjobserver-00fd123b1176fa69.rlib --extern lazy_static=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/liblazy_static-530e97b37714b7bd.rlib --extern log=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/liblog-49282ed3d5ba531a.rlib --extern measureme=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libmeasureme-f6591458a3297557.rlib --extern num_cpus=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libnum_cpus-dc001c0cc6f0117b.rlib --extern parking_lot=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libparking_lot-2f3a4934f3ebcdb1.rlib --extern polonius_engine=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libpolonius_engine-982b4c206045df37.rlib --extern rustc_rayon=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_rayon-d86f8a37471ec550.rlib --extern rustc_rayon_core=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_rayon_core-6234250da64c560e.rlib --extern rustc_apfloat=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_apfloat-ee22d0f721c1e436.rlib --extern rustc_data_structures=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_data_structures-4d186a5a2a9b2d12.so --extern errors=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_errors-5632f2539e9ae150.so --extern rustc_fs_util=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_fs_util-5f0122864e858157.so --extern rustc_macros=/build/x86_64-unknown-linux-gnu/stage0-rustc/release/deps/librustc_macros-9a4673f13c40a44d.so --extern rustc_target=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/librustc_target-e2397e11a8e1e1e0.so --extern scoped_tls=/build/x86_64-unknown-linux-gnu/stage0-rustc/x86_64-unknown-linux-gnu/release/deps/libscoped_tls-1732f3e2b8d8cff1.r [... cut ...]
command did not execute successfully: "/build/x86_64-unknown-linux-gnu/stage0/bin/cargo" "build" "--target" "x86_64-unknown-linux-gnu" "-j" "2" "--release" "--features" "" "--manifest-path" "/code/src/rustc/Cargo.toml" "--message-format" "json"
expected success, got: exit code: 101
failed to run: /build/bootstrap/debug/bootstrap build
Build completed unsuccessfully in 0:25:43
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 1
The command '/bin/sh -c make' returned a non-zero code: 2
```

Thanks for your help, I'ld love to develop in Rust for AVR arduino.

shepmaster, 2020-03-15:
> I supposed it was the easiest way to get it.

I would not assume that this is the case. IIRC, the Docker container was mostly intended to be used for some amount of CI.

I don't know if there is an easy way to get it *right* now. There's an open PR to [enable AVR as a tier-3 platform in Rust itself](https://github.com/rust-lang/rust/pull/69478), but it has some issues that need to be addressed.

dylanmckay, 2020-03-16:
I set up the dockerfiles so that ideally people would be able to use them like you are now, but it's not really documented.

>  (signal: 9, SIGKILL: kill)
> command did not execute successfully: "/build/x86_64-unknown-linux-gnu/stage0/bin/cargo" "build" "--target" "x86_64-unknown-linux-gnu" "-j" "2" "--release" "--features" "" "--manifest-path" "/code/src/rustc/Cargo.toml" "--message-format" "json"
expected success, got: exit code: 101

This sounds like the error I get when I run out of memory during compiling (usually the link stage). Perhaps `rustc` got killed by the OOM killer?

dylanmckay, 2020-05-17:
I'm going to mark this as closed, feel free to reopen if seems to happen again.

closed 2020-05-17 by dylanmckay