avr-rust/rust-legacy-fork #161: A Hello from the Rust embedded WG

therealprof, 2020-03-17:
Hi AVR-rustians,

the Rust Embedded WG has noticed with excitement that you're making big strides towards becoming an officially supported architecture in Rust. Congratulations and thanks for your hard work!

Our community is very interested in your work and asked whether you would like to join our efforts and become a part of our WG as a new team.

We would be thrilled to have you on board and so I would wholeheartedly love to invite you to join forces.

CC @dylanmckay and @shepmaster

shepmaster, 2020-03-18:
Howdy 👋 

Do you have any links or further information for what the expectations, requirements, privileges, etc. are as part of the team?

For example, are there meetings we'd be expected to attend?

therealprof, 2020-03-18:
> Do you have any links or further information 

Sure, you can find out more about the WG here: https://github.com/rust-embedded/wg

> for what the expectations, requirements, 

Our expectaction is that we can better organize resources relevant for all embedded matters. We support some architectures already (Cortex-M, Cortex-A, MSP430, RISC-V, some MIPS) and provide not only architecture specific but also some architecture independent crates, documentation, blogs and meetings. By bringing a lot of parties together, we believe that we can provide a much better Rust experience for everyone without having to point interested people away from us in other directions.

We don't have any strict requirements other than the CoC.

> privileges, etc. are as part of the team?

Since we are a affiliated with the Rust project we get to use various resources of the Rust project and a voice in embedded matters. And of course members of the organisation get a vote on general WG decisions and you can also become part of further teams collaborating on other WG properties.

We can also discuss taking over maintainership of some AVR crates and integrating that into the WG portfolio. We're not generally interested in too specific repositories (e.g. we don't provide drivers or manufacturer specific repositories) but foundational and tooling crates as well as documentation might very well live unter the orga if that's what you want.

> For example, are there meetings we'd be expected to attend?

We do have a meeting every Tuesday 8:00-9:00 PM Berlin time on our public Matrix channel: https://matrix.to/#/#rust-embedded:matrix.org . Attendance is optional but recommended.

dylanmckay, 2020-04-01:
It's taken a while to gather my thoughts on this, bear with me whilst I dump them all into text form.

My main thoughts around the topic: I'm personally much more focused on getting the compiler working, as this is by far the largest blocker for the ecosystem, compared to a lot of the higher level issues the WG deals with like API design and library implementation and maintenance. I would personally prefer an AVR library ecosystem that develops without any benevolent dictation from me.

I think that merging with the embedded-wg is generally a good thing, and avr-rust users seem to agree. I also like the idea of codifying some list of "team members" so that others can feel some agency and have legitimate ownership over merging/approving things. Joining the embedded-wg would catalyze that.

I think my main reservation is that there may be distinction between AVR-Rust joining the embedded-wg and Dylan McKay joining the embedded-wg organization. I really like the idea of expanding the community, sharing tooling, gaining from community maintenance, HAL libraries, et cetera, but I worry about the potential of signing up for obligations in my spare time (timezones also suck). I'm an on/off maintainer most of the time - couple weeks on, couple weeks off - not intentional but my GitHub patterns always seem to be seasonal.

I just wanna work on the compiler, I don't want to own (or create) the AVR ecosystem. There seem to be plenty of people capable and willing to do that (there are a lot more avr-rust users than avr-rust and especially LLVM contributors), but even today the biggest blocker is compiler bugs. I'm the monkey that twists the cogs to make LLVM speak AVR, but that kind of super-specific technical bugfix doesn't really make sense in the embedded-wg (plus, with the backend being merged upstream, we can use Rust's issue tracker). Once the backend is merged, the avr-rust organization will be basically devoid of repositories to manage.

I would like to join the embedded-wg because I think it would definitely be a positive to the AVR Rust ecosystem, so long as there are no obligations on me personally - this is open source. I don't think there would be a difference in AVR-Rust development initially, as I can only speak for myself and I only work on the compiler, but I think moving the project there would help seed the ecosystem. We would benefit from adopting the WG's established processes and documentation with respect to contributors, team members, etc. Shared tooling is a big plus. It would also form a clear(ish) pathway for contributors to obtain write/approve access, sharing the ~big stick~ approve button which would otherwise require quite a lot of thought - most contributors wouldn't be comfortable enough to overtly ask for a maintainership role.

The only counter argument I've heard about merging with the embedded-wg,  is the possible risk of ossifying the ecosystem, in a kind of "One True Way" argument. After some thought, I think that this doesn't really hold because... there is no officialy-sanctioned AVR Rust ecosystem. We don't really put that kind of stuff in the avr-rust repository, it just gets written by people that need it and are interested in it and posted on GitHub - this wouldn't change under the WG.

That's a lot of writing from me, to sum up, I approve of merging the AVR Rust ecosystem into the embedded-wg ecosystem so long as I don't have to do things. It would make sense to solicit avr-rust contributors to see if they wanted to join the embedded-wg/take up a maintenance role.

Thanks for reaching out @therealprof